espressif/ESP8266_RTOS_SDK #765: ssl_ticket.c  implicit declaration of function   mbedtls_cipher_auth_encrypt 搞不定啊！

panguoxian, 2019-11-23:
编译的时候
ssl_ticket.c  implicit declaration of function   mbedtls_cipher_auth_encrypt

components/mbedtls/mbedtls/library/ssl_ticket.c:337:17: error: implicit declaration of function 'mbedtls_cipher_auth_encrypt' [-Werror=implicit-function-declaration]
     if( ( ret = mbedtls_cipher_auth_encrypt( &key->ctx,

kmalloc8, 2019-11-28:
这个问题有更新没？

kmalloc8, 2019-12-03:
mbedTLS  --->
    Symmetric Ciphers  --->
        [*] CCM (Counter with CBC-MAC) block cipher modes  
        [*] GCM (Galois/Counter) block cipher modes
    [*] Elliptic Curve Ciphers  --->

k4r4c, 2019-12-07:
Adding `#define MBEDTLS_CIPHER_MODE_AEAD` to `config.h` solved the problem for me (mbedtls_cipher_auth_encrypt is defined only for aead mode, see cipher.c).