espressif/ESP8266_RTOS_SDK #767: problem in chinese string processing

EaonL, 2019-11-26:
Description：
When I used the printf() function to print a Chinese string, I found that the string was output in the format of GB2312, and the source file encoding format of the string was utf-8. This means that the execution character set when the compiler generates an executable image is GB2312, but I want the compiler's execution character set to be utf-8. 

I tried adding a u8 declaration before the string, such as printf (u8"zh-hans testing: 中文测试"), but it didn't work, it output this: "zh-hans testing: 涓枃娴嬭瘯";  the string "涓枃娴嬭瘯" is the GB2312 encoding of the string "中文测试".

How can I make the compiler set the string's execution character set to utf-8 when generating an executable image?

EaonL, 2019-11-26:
In addition, I used the mkspiffs tool to make a spiffs file system image. I found that the generated file system image spiffs.bin also encodes Chinese strings in GB2312 format.

What can I do to make the resulting executable image encode the Chinese character string with UTF-8 character set?

EaonL, 2019-11-26:
I try to change the execution character set setting of xtensa-lx106-elf-gcc at compile time.
step like this:
1.  vim $IDF_PATH/make/project.mk

2.  Modify
COMMON_FLAGS = \
        -ffunction-sections -fdata-sections \
        -fstrict-volatile-bitfields \
        -mlongcalls \
        -nostdlib \
To
COMMON_FLAGS = \
        -ffunction-sections -fdata-sections \
        -fstrict-volatile-bitfields \
        -mlongcalls \
        -nostdlib \
        **-fexec-charset=UTF-8 \
        -finput-charset=UTF-8**

3. Compile the program and run it

4. However, it still outputs chinese string as GB2312 format.

Note:
I compiled the program with and without the execution of the character set parameters. After comparing the executable images, I found that the generated images were consistent. This means that the parameters **"-fexec-charset=UTF-8 -finput-charset=UTF-8"** have not taken effect

EaonL, 2019-11-27:
I compile the esp8266 image on another computer, and then can successfully output Chinese string...
This is a confusing question......

thisdream, 2019-11-28:
you can try diffrent CommAssistant to show printf()