espressif/ESP8266_RTOS_SDK #769: Problem with MQTT SSL mutual authentication

OskOnu, 2019-11-27:
I have problem connecting with mqtt broker over ssl with mutual auth. 
I have 3 certificates in PEM format: 
ca_certificate.pem:
-----BEGIN CERTIFICATE-----
...
-----END CERTIFICATE-----
, client_certificate.pem:
-----BEGIN CERTIFICATE-----
...
-----END CERTIFICATE-----
, private_key.pem:
-----BEGIN RSA PRIVATE KEY-----
-----END RSA PRIVATE KEY-----

I configured them like this:
const esp_mqtt_client_config_t mqtt_cfg = {
        .uri = "mqtts://server_adres",
	.port = 8883,
	.username = "device_nr",
	.password = "device_pass",
        .event_handle = mqtt_event_handler,
        .cert_pem = (const char *)ca_certificate_pem_start,
	.client_cert_pem = (const char *)client_certificate_pem,
        .client_key_pem = (const char *)private_key_pem_start,
    };
I'm using 'ssl_mutual_auth' example from ESP8266_RTOS_SDK. 
The output is:
I (601) MQTTS_EXAMPLE: [APP] Startup..
I (601) MQTTS_EXAMPLE: [APP] Free memory: 80380 bytes
I (611) MQTTS_EXAMPLE: [APP] IDF version:
I (631) MQTTS_EXAMPLE: start the WIFI SSID:[ssid]
I (641) MQTTS_EXAMPLE: Waiting for wifi
E (821) MQTTS_EXAMPLE: Disconnect reason : 3
E (951) MQTTS_EXAMPLE: Disconnect reason : 202
I (2111) event: sta ip: 192.168.43.173, mask: 255.255.255.0, gw: 192.168.43.1
I (2121) MQTTS_EXAMPLE: [APP] Free memory: 72268 bytes
I (2131) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
E (2151) MQTT_CLIENT: Client has not connected
E (12151) MQTT_CLIENT: Client has not connected
E (14661) esp-tls: mbedtls_ssl_handshake returned -0x7280
I (14661) esp-tls: Certificate verified.
E (14701) esp-tls: Failed to open new connection
E (14701) TRANS_SSL: Failed to open a new connection
E (14711) MQTT_CLIENT: Error transport connect
I (14721) MQTT_CLIENT: Reconnect after 10000 ms
I (14721) MQTTS_EXAMPLE: MQTT_EVENT_DISCONNECTED
E (22151) MQTT_CLIENT: Client has not connected
E (32151) MQTT_CLIENT: Client has not connected
E (41861) esp-tls: mbedtls_ssl_handshake returned -0x7280

What is the problem? What I'm doing wrong?

bbinet, 2020-04-18:
@OskOnu Have you been able to fix this issue? 

I'm having the exactly same error:
```
esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x50
```
Any clue to fix it?

OskOnu, 2020-04-18:
What broker do you use?

OskOnu, 2020-04-18:
> @OskOnu Have you been able to fix this issue?
> 
> I'm having the exactly same error:
> 
> ```
> esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x50
> ```
> 
> Any clue to fix it?

What broker do you use?

bbinet, 2020-04-18:
@OskOnu I'm using vernemq broker.

OskOnu, 2020-04-18:
@bbinet First of all I would check my SSL certificate configuration. Namely CN or Common Name should be the address of your server, according to this: https://support.dnsimple.com/articles/what-is-common-name .  
Also make sure, that broker have correct permission to access the certification files.

OskOnu, 2020-04-18:
@bbinet First of all I would check my SSL certificate configuration. Namely CN or Common Name should be the address to your server, according to this: https://support.dnsimple.com/articles/what-is-common-name .
Also make sure, that broker has correct permissions to access the certification files.

bbinet, 2020-04-18:
@OskOnu I've already checked my SSL certificate configuration, as I'm able to connect successfully with the same SSL certificates using mosquitto_pub/mosquitto_sub utilities, so the issue comes from the ESP side.

FYI, you can find more details of my setup here: https://github.com/espressif/esp-mqtt/issues/125#issuecomment-615916384

bbinet, 2020-04-20:
@OskOnu This is now working: the "issue" seems to occur only when the client key is 4096-bit.
I've generated a new client key (2048-bit) and everything works fine!

oskonu, 2020-04-22:
@bbinet I still have problem sometimes. How do you generate your certificate?

bbinet, 2020-04-22:
I was using a home-made script based on openssl, but i plan to move to https://github.com/OpenVPN/easy-rsa

But on my side, I'm using an esp32 which is more powerful than the esp8266, so this may explain you still have problem sometimes...

SERIDJ, 2020-07-06:
hi @orakso  @bbinet  please in config what mean cert_pem in esp_mqtt_client_config_t   its  authority certificat ca.crt ? 

i am using https://github.com/espressif/esp-idf/blob/master/examples/protocols/mqtt/ssl_mutual_auth/main/app_main.c and i dont know why i cant use ca.crt  ?

orakso, 2020-07-06:
> hi @orakso @bbinet please in config what mean cert_pem in esp_mqtt_client_config_t its authority certificat ca.crt ?
> 
> i am using https://github.com/espressif/esp-idf/blob/master/examples/protocols/mqtt/ssl_mutual_auth/main/app_main.c and i dont know why i cant use ca.crt ? and do u have idea how to use use_global_ca_store ?

I'm not sure whether I correctly understand your question. I think you have to use certificates in PEM format. You just need to extract PEMs from your keystore

seguidor777, 2020-12-30:
Same issue here using an esp32 controller and already tested with mosquitto client and the mTLS works fine.
On the esp32 side I'm getting
```
TRANS_SSL: Failed to open a new connection
MQTT_CLIENT: Error transport connect
```
It seems that the esp32 is not setting the CA properly