espressif/ESP8266_RTOS_SDK #770: 活脱脱【

LiThirteen, 2019-11-29:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at bbs.espressif.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](http://esp-idf.readthedocs.io/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    //bd6ea4393c7d2f059fc4decc70f1ec3eb3597268
- Development Env:      [Arduino IDE|Make|Eclipse|other]
- Operating System:     [Windows|Ubuntu|MacOS]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to repropduce

1. step1
2. ...

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

LiThirteen, 2019-11-29:
调试信息
[image]

LiThirteen, 2019-11-29:
[0;32mI (231723) At_Uart: -------------------- heap1:27932 --------------------------[0m
[0;32mI (231723) Http_Clinet_PostData: hoststr = 192.168.10.10:8080 

[0;32mI (231753) Http_Clinet_PostData: HTTP_EVENT_ON_CONNECTED[0m

[0;32mI (232063) response: HTTP_EVENT_ON_HEADER, key=Transfer-Encoding, value=chunked[0m
[0;32mI (232073) response: HTTP_EVENT_ON_HEADER, key=Date, value=Fri, 29 Nov 2019 03:52:43 
[0;32mI (232083) Http_Clinet_PostData: HTTP_EVENT_ON_DATA, len=57[0m
[0;32mI (232093) response: {"code":200,"msg":"success","response":[],"success":true}[0m
[0;32mI (232113) Http_Clinet_PostData: HTTP_EVENT_ON_FINISH[0m
[0;32mI (232113) Http_Clinet_PostData: POST url-1 Status = 200, content_length = -1[0m
repLen:58, repbuf:{"code":200,"msg":"success","response":[],"success":true}
[0;32mI (232143) Http_Clinet_PostData: HTTP_EVENT_DISCONNECTED[0m
---------------238--------------
[0;32mI (232153) At_Uart: -------------------- heap2:27772 --------------------------[0m
停止发送post请求，大约1分钟
[0;32mI (315483) At_Uart: -------------------- heap1:46132 --------------------------[0m
[0;32mI (315983) Http_Clinet_PostData: HTTP_EVENT_ON_CONNECTED[0m
[0;32mI (317223) response: HTTP_EVENT_ON_HEADER, key=Content-Type, value=application/json;charset=UTF-8[0m
[0;32mI (317233) response: HTTP_EVENT_ON_HEADER, key=Transfer-Encoding, value=chunked[0m
[0;32mI (317243) response: HTTP_EVENT_ON_HEADER, key=Date, value=Fri, 29 Nov 2019 03:54:08 GMT[0m
[0;32mI (317253) Http_Clinet_PostData: HTTP_EVENT_ON_DATA, len=57[0m
[0;32mI (317263) response: {"code":200,"msg":"success","response":[],"success":true}[0m
[0;32mI (317283) Http_Clinet_PostData: HTTP_EVENT_ON_FINISH[0m
[0;32mI (317283) Http_Clinet_PostData: POST url-1 Status = 200, content_length = -1[0m

[0;32mI (317313) Http_Clinet_PostData: HTTP_EVENT_DISCONNECTED[0m
---------------239--------------
[0;32mI (317323) At_Uart: -------------------- heap2:46372 --------------------------

lhespress, 2019-12-05:
`因为进行了多次短连接`：每次连接断开时，处于 TIME_WAIT 状态的连接需要等待 2MSL 时间，内存才会完全回收。可以通过 setsockopt 函数，启用 SO_LINGER 选项，参考 code 如下：

```
struct linger so_linger;
so_linger.l_onoff = 1;
so_linger.l_linger = 0;
setsockopt(sock, SOL_SOCKET, SO_LINGER, &so_linger, sizeof(so_linger));
```
当调用 close 函数时，套接字 sock 会立即中止，内存会被立即回收。

LiThirteen, 2019-12-05:
感谢您的回复，以及您的解决方案。

closed 2019-12-05 by LiThirteen