espressif/ESP8266_RTOS_SDK #774: dhcp problem in AP mode

bartekkr, 2019-12-04:
Hi, I am using branch release/v3.2.

 When i configure device (ESP8266) in AP mode and i want to run dhcp server on it then all dhcp messages are rejected - below function doesn't maches incomming datagram:

file:line `${IDF_PATH}/components/lwip/lwip/src/core/udp.c:152`
function `udp_input_local_match()`

After investigation, i discovered that following line should be added:

`ip_set_option(pcb_dhcps, SOF_BROADCAST);`
file:line `${IDF_PATH}/components/lwip/apps/dhcpserver/dhcpserver.c:1156`
function `dhcps_start()`

git hashtag: `6f9dec8`

now dhcp in ap mode works fine

best regards, Bartek

lhespress, 2019-12-06:
Can you share me how to reproduce the issues? eg: What device work on DHCP client? It's a special case from the option describe.

bartekkr, 2019-12-06:
I'll do it after tomorrow, currently i am out of home

bartekkr, 2019-12-08:
Hi
attached simple example.

[dhcp_bug_reproduction.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/3936511/dhcp_bug_reproduction.zip)

after running `make menuconfig` i changed following:
`Serial flasher config->Default serial port = com9`
`Serial flasher config->'make monitor' baud rate = 115200bps`
`Serial flasher config->Flash size = 1MB`
`Component config->ESP8266-specific->UART console baud rate = 115200`
`Component config->Log output->Default log verbosity = Debug`
`Component config->LWIP->Enable broadcast filter per pcb on udp and raw send operation = true`
`Component config->LWIP->Enable the broadcast filter on recv operations = true`

using module ESP-01
configuraiotn wifi: Access Point
tested with following stations (the same behavior):
- sony xperia 1 (do not know which wifi module is placed there)
- quallcom atheros ar9485

compile example without any changes in sdk and flash esp module
run `make monitor`
try connect to ssid `accesspt`. password `accesspt`
on console, last logs you see are:

`D (7271) event: SYSTEM_EVENT_AP_STACONNECTED, mac:12:75:ae:0a:05:3b, aid:1`
`I (7281) testapp: station:12:75:ae:0a:05:3b join, AID=1`

and on connecting devices i still see "waiting for IP..."

then apply following:
`ip_set_option(pcb_dhcps, SOF_BROADCAST);`
just before:
`udp_bind(pcb_dhcps, &netif->ip_addr, DHCPS_SERVER_PORT);`
in file:
`${IDF_PATH}/components/lwip/apps/dhcpserver/dhcpserver.c`

rebuild project, flash and run `make monitor`
try connect to `accesspt` ssid again using password `accesspt` and you see:
`D (34361) event: SYSTEM_EVENT_AP_STACONNECTED, mac:12:75:ae:0a:05:3b, aid:1`
`I (34371) testapp: station:12:75:ae:0a:05:3b join, AID=1`
`D (34481) udp_sync: UDP sync cache udp 0x40106f84`
`D (34491) udp_sync: UDP sync clear udp 0x40106f84`
`D (34511) udp_sync: UDP sync cache udp 0x40106f84`
`D (34511) udp_sync: UDP sync clear udp 0x40106f84`
**`I (34511) tcpip_adapter: softAP assign IP to station,IP is: 192.168.4.2`**
**`D (34521) event: SYSTEM_EVENT_AP_STAIPASSIGNED`**
**`I (34531) testapp: assigned ip to the connected station`**

and on connecting devices i see "assingned IP: 192.168.4.X"
so now it works fine

regards, bartek

closed 2020-01-07 by FayeY