espressif/ESP8266_RTOS_SDK #775: ESP8266_RTOS_SDK-3.2程序烧录求助

hscxg, 2019-12-05:
我用的开发板为: ESP8266DevKitC-02D-F  (2M Flash)

所使用的ESP8266 SDK版本为: ESP8266_RTOS_SDK-3.2, 

直接编译它提供的例程: project_template, 

编译OK后, 生成三个bin文件:project_template.bin, partitions_singleapp.bin, bootloader.bin

现在我用官方的烧录工具flash_download_tools_v3.6.7, 想将这三个bin文件烧录到ESP8266芯片上去, 但不知每个bin文件其对应的烧录地址是什么, 网上找了很长时间, 也没找到相关资料说明, 请指点, 谢谢!

lhespress, 2019-12-06:
--flash_mode dio --flash_freq 40m --flash_size 2MB 0x0000 ~/ESP8266_RTOS_SDK/examples/get-started/project_template/build/bootloader/bootloader.bin 0x10000 ~/ESP8266_RTOS_SDK/examples/get-started/project_template/build/project_template.bin 0x8000 ~/ESP8266_RTOS_SDK/examples/get-started/project_template/build/partitions_singleapp.bin

hscxg, 2019-12-07:
我按这地址将编译后的程序烧入到ESP8266后, 再上电运行, 程序却一直在报Fatal exception信息, 无法正常运行, 不知是怎么回事? 
[image]

上电时LOG 信息如下:
ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x40100000, len 2592, room 16 
tail 0
chksum 0xf3
load 0x3ffe8000, len 764, room 8 
tail 4
chksum 0x92
load 0x3ffe82fc, len 676, room 4 
tail 0
chksum 0x22
csum 0x22

2nd boot version : 1.7(5d6f877)
SPI Speed : 40MHz
SPI Mode : QIO
SPI Flash Size & Map: 16Mbit(512KB+512KB)
jump to run user2 @ 81000

Fatal exception (29): 
epc1=0x4010fd0b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (29): 
epc1=0x4010fd0b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (29): 
epc1=0x4010fd0b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (29): 
epc1=0x4010fd0b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (29): 
epc1=0x4010fd0b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (29): 
epc1=0x4010fd0b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (29): 
epc1=0x4010fd0b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (29): 
epc1=0x4010fd0b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (29): 
epc1=0x4010fd0b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (29): 
epc1=0x4010fd0b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (29): 
epc1=0x4010fd0b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
...
...

hscxg, 2019-12-09:
LOG信息如下:
ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x000000ee, len 0, room 16 
tail 0
chksum 0xef
load 0x00000000, len 16777216, room 8 

然后就程序好像就停止运行了.

烧录配置界面如下, 不知是否有错?

[image]

lhespress, 2019-12-10:
需要匹配你编译时的配置信息，我给出的是默认配置。

closed 2020-01-07 by FayeY