espressif/ESP8266_RTOS_SDK #779: OTA broken in v3.3

DKrepsky, 2019-12-15:
## Problem Description

OTA is broken after commit 6d428b28510d149e7606c5a4146265588e54b30e in branch release/v3.3.

Log:
I (5572) simple_ota_example: Connect to Wifi ! Start to Connect to Server....
I (5599) esp_https_ota: Starting OTA...
I (5601) esp_https_ota: Writing to partition subtype 17 at offset 0x110000
I (12608) esp_https_ota: esp_ota_begin succeeded
I (12610) esp_https_ota: Please Wait. This may take time
I (22663) esp_https_ota: Connection closed,all data received
I (22670) esp_image: segment 0: paddr=0x00110010 vaddr=0x40210010 size=0x943f0 (607216) map
0x40210010: _stext at ??:?

E (22819) esp_image: unaligned segment length 0xffffffff
E (22821) esp_https_ota: Error: esp_ota_end failed! err=0x5379. Image is invalid
E (22825) simple_ota_example: Firmware Upgrades Failed

### Steps to reproduce
Use ```examples/system/ota/simple_ota_example``` to reproduce.

NOTE1: I'm using an HTTP server (not HTTPS) with HTTP OTA enabled.
NOTE2: Same code works up to commit b64b51f091d40ddf5dd445c921ad7f53c58885f4

DKrepsky, 2019-12-15:
Log with commit b64b51f

I (5288) simple_ota_example: Connect to Wifi ! Start to Connect to Server....
I (5318) esp_https_ota: Starting OTA...
I (5320) esp_https_ota: Writing to partition subtype 17 at offset 0x110000
I (12313) esp_https_ota: esp_ota_begin succeeded
I (12315) esp_https_ota: Please Wait. This may take time
I (33682) esp_https_ota: Connection closed,all data received
I (33690) esp_image: segment 0: paddr=0x00110010 vaddr=0x40210010 size=0x943f0 (607216) map
0x40210010: _stext at ??:?

I (33840) esp_image: segment 1: paddr=0x001a4408 vaddr=0x402a4400 size=0x15aec ( 88812) map
I (33866) esp_image: segment 2: paddr=0x001b9efc vaddr=0x3ffe8000 size=0x007dc (  2012) 
I (33870) esp_image: segment 3: paddr=0x001ba6e0 vaddr=0x40100000 size=0x041b0 ( 16816) 
I (33878) esp_image: segment 4: paddr=0x001be898 vaddr=0x401041b0 size=0x0593c ( 22844) 
0x401041b0: ppCalFrameTimes at ??:?

I (33888) esp_image: segment 0: paddr=0x00110010 vaddr=0x40210010 size=0x943f0 (607216) map
0x40210010: _stext at ??:?

I (34040) esp_image: segment 1: paddr=0x001a4408 vaddr=0x402a4400 size=0x15aec ( 88812) map
I (34066) esp_image: segment 2: paddr=0x001b9efc vaddr=0x3ffe8000 size=0x007dc (  2012) 
I (34070) esp_image: segment 3: paddr=0x001ba6e0 vaddr=0x40100000 size=0x041b0 ( 16816) 
I (34078) esp_image: segment 4: paddr=0x001be898 vaddr=0x401041b0 size=0x0593c ( 22844) 
0x401041b0: ppCalFrameTimes at ??:?

I (34118) esp_https_ota: esp_ota_set_boot_partition succeeded
I (34121) wifi: state: 5 -> 0 (0)
I (34123) wifi: pm stop
E (34127) simple_ota_example: Disconnect reason : 8
�I (42) boo: ESP-IDF v3.2-297-gb64b51f0 2d stage bootloader

DKrepsky, 2019-12-15:
Same thing on master branch:

I (42) boot: ESP-IDF v3.2-301-g3d394bd9 2nd stage bootloader
I (42) boot: compile time 18:12:32
I (43) qio_mode: Enabling default flash chip QIO
I (43) boot: SPI Speed      : 40MHz
I (43) boot: SPI Mode       : QIO
I (43) boot: SPI Flash Size : 4MB
I (43) boot: Partition Table:
I (43) boot: ## Label            Usage          Type ST Offset   Length
I (44) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (44) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (45) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (45) boot:  3 ota_0            OTA app          00 10 00010000 000f0000
I (46) boot:  4 ota_1            OTA app          00 11 00110000 000f0000
I (46) boot: End of partition table
I (46) boot: No factory image, trying OTA 0
I (47) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x69a20 (432672) map
0x40210010: _stext at ??:?

I (47) esp_image: segment 1: paddr=0x00079a38 vaddr=0x40279a30 size=0x0fe30 ( 65072) map
I (48) esp_image: segment 2: paddr=0x00089870 vaddr=0x3ffe8000 size=0x005ec (  1516) load
I (48) esp_image: segment 3: paddr=0x00089e64 vaddr=0x40100000 size=0x00a50 (  2640) load
I (49) esp_image: segment 4: paddr=0x0008a8bc vaddr=0x40100a50 size=0x05854 ( 22612) load
I (49) boot: Loaded app from partition at offset 0x10000
I (65) system_api: Base MAC address is not set, read default base MAC address from EFUSE
I (71) system_api: Base MAC address is not set, read default base MAC address from EFUSE
W (75) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
phy_version: 1155.0, 6cb3053, Nov 11 2019, 17:31:08, RTOS new
I (235) phy_init: phy ver: 1155_0
W (237) phy_init: saving new calibration data because of checksum failure, mode(2)
I (247) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2
I (293) simple_ota_example: Setting WiFi configuration SSID testnet...
I (300) simple_ota_example: Starting OTA example...
I (3208) wifi: state: 0 -> 2 (b0)
I (3222) wifi: state: 2 -> 3 (0)
I (3233) wifi: state: 3 -> 5 (10)
I (3236) wifi: pm start, type: 2
I (5912) event: sta ip: 192.168.0.60, mask: 255.255.255.0, gw: 192.168.0.1
I (5916) simple_ota_example: Connect to Wifi ! Start to Connect to Server....
I (5950) esp_https_ota: Starting OTA...
I (5953) esp_https_ota: Writing to partition subtype 17 at offset 0x110000
I (12945) esp_https_ota: esp_ota_begin succeeded
I (12947) esp_https_ota: Please Wait. This may take time
I (29604) esp_https_ota: Connection closed,all data received
I (29610) esp_image: segment 0: paddr=0x00110010 vaddr=0x40210010 size=0x943f0 (607216) map
0x40210010: _stext at ??:?

E (29763) esp_image: unaligned segment length 0xffffffff
E (29766) esp_https_ota: Error: esp_ota_end failed! err=0x5379. Image is invalid
E (29769) simple_ota_example: Firmware Upgrades Failed

d51x, 2019-12-25:
the same problem with ota on sdk v.3.3 on esp8266
i think, in esp_https_ota function
this code
```
while (1) {
        int data_read = esp_http_client_read(client, upgrade_data_buf, OTA_BUF_SIZE);
        if (data_read == 0) { 
```

esp_http_client_read return 0 randomly, and binary file doesn't load fully from server, that's way we have the problem with ota on v.3.3

but the way, i tried to change loop condition to (comment BREAK)
```
        if (data_read == 0) {
            ESP_LOGI(TAG, "Connection closed,all data received");
            ///break;    commeted BREAK
        }
```

and ota works, but esp_http_client_read working unstable
if it returns 0, i read again (call esp_http_client_read) and again, and again, if esp_http_client_read return 0
sometimes this function returns 0 constantly... and watchdog and reboot

I think the problem with function **esp_http_client_read** on v.3.3

maierkomor, 2020-01-09:
I have also have issues with OTA not working in 3.3. Looking at the created images, they all are build for the factory partition address instead of the OTA partition addresses. Additionally, I have seen that on 3.2 I get the same issue when I switch from the 1MB flash size setting to the 4MB flash size setting. I did a test in which I really only changed that setting...

svenbieg, 2020-01-31:
I download the image manually and the update-process is working, but the device fails to boot the image. On the master-branch there is an exception 0 calling cpu_start() and the device reboots. On v3.3-rc1 there is a message "user code done".

Dipendra77, 2020-02-10:
> Same thing on master branch:
>
> I (42) boot: ESP-IDF v3.2-301-g3d394bd9 2nd stage bootloader
> I (42) boot: compile time 18:12:32
> I (43) qio_mode: Enabling default flash chip QIO
> I (43) boot: SPI Speed [...]

@FayeY  @donghengqaz , I  am facing the same issue. OTA was working for me few days back but after the last update of the SDK, esp_http_client_read( ) function fails to download the entire binary image and OTA upgrade is failing consistently.  Please provide a fix for this error in the latest SDK.

DKrepsky, 2020-05-05:
Hi,

It seems that the latest commits on both master and release/v3.3 branches fixes this issue.
Did some trials (1000 updates for both TLS and no TLS) with 100% success rate.

If someone else still have fota problems I don't think they are related to this, so I'm gonna close this issue.

Thanks.

closed 2020-05-05 by DKrepsky

maierkomor, 2020-05-05:
OTA still seems to be broken for me on IDF version: v3.3-rc1-80-gb74255d0.

The addresses of the binaries are for both app1 and app2 identical. This can be reproduced with my software package called atrium (www.github.com/maierkomor/atrium). Run setupenv.sh and update the IDF tree to the relevant release. Then build the s20 package with "make PROJECT=s20 ota".

This yields firmware files with following addresses:
thomas@azalin:~/src/atrium.git% esptool.py image_info build/firmware.app1.bin
esptool.py v2.5.0
WARNING: Suspicious segment 0x40210010, length 356924
WARNING: Suspicious segment 0x4026724c, length 53252
Image version: 1
Entry point: 402125c4
5 segments
Segment 1: len 0x5723c load 0x40210010 file_offs 0x00000008
Segment 2: len 0x0d004 load 0x4026724c file_offs 0x0005724c
Segment 3: len 0x00ae8 load 0x3ffe8000 file_offs 0x00064258
Segment 4: len 0x00a30 load 0x40100000 file_offs 0x00064d48
Segment 5: len 0x05858 load 0x40100a30 file_offs 0x00065780
Checksum: 0f (valid)
thomas@azalin:~/src/atrium.git% esptool.py image_info build/firmware.app2.bin
esptool.py v2.5.0
WARNING: Suspicious segment 0x40210010, length 356924
WARNING: Suspicious segment 0x4026724c, length 53252
Image version: 1
Entry point: 402125c4
5 segments
Segment 1: len 0x5723c load 0x40210010 file_offs 0x00000008
Segment 2: len 0x0d004 load 0x4026724c file_offs 0x0005724c
Segment 3: len 0x00ae8 load 0x3ffe8000 file_offs 0x00064258
Segment 4: len 0x00a30 load 0x40100000 file_offs 0x00064d48
Segment 5: len 0x05858 load 0x40100a30 file_offs 0x00065780
Checksum: 0f (valid)

Segment 1 of app2 should be at 0x40280010...

Updating to the app2 firmware image fails with address errors, and after changing the boot partition to app2 booting fails with a loop on: 
Fatal exception (0):
epc1=0x402125c4, epc2=0x00000000, epc3=0x40274c6c, excvaddr=0x4010648c, depc=0x00000000
0x402125c4: call_start_cpu at /opt/idf-esp8266/components/esp8266/source/startup.c:120

In contrast the identical setup with release/v3.2 works without a problem. For v3.2 the addresses of apps are:
homas@azalin:~/src/atrium.git% esptool.py image_info build/firmware.app2.bin
esptool.py v2.5.0
WARNING: Suspicious segment 0x40280010, length 413928
Image version: 1
Entry point: 4028e078
4 segments
Segment 1: len 0x650e8 load 0x40280010 file_offs 0x00000008
Segment 2: len 0x00ab8 load 0x3ffe8000 file_offs 0x000650f8
Segment 3: len 0x001e0 load 0x3ffe8ab8 file_offs 0x00065bb8
Segment 4: len 0x067a4 load 0x40100000 file_offs 0x00065da0
Checksum: 73 (valid)

The Atrium build relies on the IDF make build and uses directly its infrastructure for creating the OTA binaries.