espressif/ESP8266_RTOS_SDK #784: Writing data to flash permanently

Dipendra77, 2019-12-19:
Hi, 
I am configuring the wi-fi credentials from an android app, allowing user to connect to the wifi network they desire to connect. I am sending the wifi credentials through a udp. But the wifi credentials are lost on next boot and expects us to configure again. Can anyone tell me if we can store these credentials to some flash sector permanently so it remembers the previously configured network and connects to it even after reboot, unless user wants to change the credentials to other network. 
Thank you in advance.

JackieLi1011, 2019-12-20:
Please explore NVS API,we ll find out the solution

Dipendra77, 2019-12-30:
How do I connect to the previously configured wifi credentials after the device boots up? Is it stored in a structure which we can read and connect pour device  using the saved ssid and password?

FayeY, 2020-01-07:
Could this example help？https://github.com/espressif/ESP8266_RTOS_SDK/tree/master/examples/wifi/simple_wifi