espressif/ESP8266_RTOS_SDK #785: make: xtensa-lx106-elf-gcc: Command not found

GarMingLi, 2019-12-19:
## Environment
- IDF version : a215fe3b43d10cef2475437861d30dfbc3a74e0f
- Development Env:      Make
- Operating System:     win10 wsl（Ubuntu18.04）
- toolchain: xtensa-lx106-elf-linux32-1.22.0-100-ge567ec7-5.2.0.tar

## Problem Description
I built the development environment of ESP8266 RTOS SDK in the Ubuntu subsystem of win10, but there was an error when compiling:
```
make: xtensa-lx106-elf-gcc: Command not found
expr: syntax error
/bin/sh: 1: xtensa-lx106-elf-gcc: not found
/bin/sh: 1: xtensa-lx106-elf-gcc: not found
Toolchain path: 
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
Python requirements from /home/ljm/esp/ESP8266_RTOS_SDK/requirements.txt are satisfied.
make[1]: xtensa-lx106-elf-gcc: Command not found
expr: syntax error
/bin/sh: 1: xtensa-lx106-elf-gcc: not found
/bin/sh: 1: xtensa-lx106-elf-gcc: not found
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
CC build/bootloader/main/bootloader_start.o
make[2]: xtensa-lx106-elf-gcc: Command not found
/home/ljm/esp/ESP8266_RTOS_SDK/make/component_wrapper.mk:291: recipe for target 'bootloader_start.o' failed
make[2]: *** [bootloader_start.o] Error 127
/home/ljm/esp/ESP8266_RTOS_SDK/make/project.mk:571: recipe for target 'component-main-build' failed
make[1]: *** [component-main-build] Error 2
```
echo $PATH
`/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:~/esp/xtensa-lx106-elf/bin`

xtensa-lx106-elf-gcc -v
```
Using built-in specs.
COLLECT_GCC=/home/ljm/esp/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc
COLLECT_LTO_WRAPPER=/home/ljm/esp/xtensa-lx106-elf/bin/../libexec/gcc/xtensa-lx106-elf/5.2.0/lto-wrapper
Target: xtensa-lx106-elf
Configured with: /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/configure --build=i686-build_pc-linux-gnu --host=i686-build_pc-linux-gnu --target=xtensa-lx106-elf --prefix=/builds/idf/crosstool-NG/builds/xtensa-lx106-elf --with-local-prefix=/builds/idf/crosstool-NG/builds/xtensa-lx106-elf/xtensa-lx106-elf/sysroot --with-sysroot=/builds/idf/crosstool-NG/builds/xtensa-lx106-elf/xtensa-lx106-elf/sysroot --with-newlib --enable-threads=no --disable-shared --with-pkgversion='crosstool-NG crosstool-ng-1.22.0-100-ge567ec7' --disable-__cxa_atexit --enable-cxx-flags='-fno-exceptions -fno-rtti' --with-gmp=/builds/idf/crosstool-NG/.build/xtensa-lx106-elf/buildtools --with-mpfr=/builds/idf/crosstool-NG/.build/xtensa-lx106-elf/buildtools --with-mpc=/builds/idf/crosstool-NG/.build/xtensa-lx106-elf/buildtools --with-isl=/builds/idf/crosstool-NG/.build/xtensa-lx106-elf/buildtools --with-cloog=/builds/idf/crosstool-NG/.build/xtensa-lx106-elf/buildtools --with-libelf=/builds/idf/crosstool-NG/.build/xtensa-lx106-elf/buildtools --enable-lto --enable-target-optspace --without-long-double-128 --disable-libgomp --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libquadmath-support --disable-nls --disable-multilib --enable-languages=c,c++
Thread model: single
gcc version 5.2.0 (crosstool-NG crosstool-ng-1.22.0-100-ge567ec7) 
```
The environment variable is normal, but I'm not sure why it was compiled incorrectly.

JackieLi1011, 2019-12-20:
Type**source ~/.profile** in your terminal to execute your environment variable

GarMingLi, 2019-12-20:
I would have done this earlier, otherwise the print path wouldn't have seen the message.

GarMingLi, 2019-12-20:
But now that I've switched to root, I'm ready to compile.

donghengqaz, 2020-01-03:
It is amazing, I see the toolchain is `xtensa-lx106-elf-linux32-1.22.0-100-ge567ec7-5.2.0.tar`, and it is used at 32-bit ubuntu system, so do you ubuntu WSL is 32-bit ?

GarMingLi, 2020-07-04:
64-bit.

Ordinateur-Hack, 2020-12-20:
@GarMingLi Did you manage to get this to work? I'm not using WSL(2) since there is no USB passthrough yet, thus I can't flash to my ESP8266 board.

I'm now trying MSYS2 and downloaded the "all-in-one toolchain & MSYS2 zip file" from [here](https://docs.espressif.com/projects/esp8266-rtos-sdk/en/latest/get-started/windows-setup.html). Then, I also downloaded the toolchain for ESP8266 from the same site and put it in my `~/esp` folder. I've updated the PATH (in `.bash_profile`) to include `xtensa-lx106-elf/bin`, however when running `make` the command can't be found (same error message).

Ordinateur-Hack, 2020-12-21:
I've just moved the `xtensa-lx106-elf-gcc` folder to `/opt` (where you can already see the folder `xtensa-esp32-elf`. Then, I've added these two lines at the end of `.bash_profile`:

_Note that `C:/msys32/home/domin/esp/ESP8266_RTOS_SDK` is the path I've cloned the RTOS_SDK into._
```
export PATH=$PATH:/opt/xtensa-lx106-elf/bin
export IDF_PATH="C:/msys32/home/domin/esp/ESP8266_RTOS_SDK"
```

I don't know why, but now it's working 🎉
**The fact, that we should put the extracted xtensa toolchain (in my case for Windows 10 using MSYS2) into the `/opt` folder, should definitely be added to the documentation [here](https://docs.espressif.com/projects/esp8266-rtos-sdk/en/latest/get-started/windows-setup.html).**

GiaMat90, 2024-01-08:
>  I solved this by extracting the xtensa-lx106-elf-gcc8_4_0-esp-2020r3-win32.zip file into msys32 home directory
> so now my Xtensa path looks like this: C:\msys32\home\PC\xtensa-lx106-elf\bin (where instead of PC it will be your username)
> 
>  Once this is done, open mingw32.exe and run the following commands:
> PATH="${HOME}/xtensa-lx106-elf/bin:${PATH}"
> export IDF_PATH=~/esp/ESP8266_RTOS_SDK
> 
>  Now run echo $PATH and echo $IDF_PATH to check if the newly included paths are there.

This works for me too. I agree with @nabeelkirmani, the [https://docs.espressif.com/projects/esp8266-rtos-sdk/en/latest/get-started/windows-setup.html](guide) should be improved.