espressif/ESP8266_RTOS_SDK #790: unable to sniff eapol/authentication frames in promiscuous mode

huzaifah0x00, 2019-12-21:
## Environment

- Development Kit:
- IDF version : ``a215fe3b``
- Development Env:      [Make]
- Operating System:     [Ubuntu]
- Power Supply:         [USB]

## Problem Description

i want to sniff 4-way handshakes using the esp8266 and save them to sdcard

it is able to capture beacon/deauth/probe/association and other frames but it 
looks like Authentication/eapol frames (the frames containing the 4 way handshake) are being ignored 
and not passed to the callback function... 

i made sure that an authentication frame was "in the air" 
using other tools (aircrack) but the esp8266 couldn't see it .

if anyone wants to analyze this issue  
	the project is at my repo: `https://github.com/huzaifah0x00/esp8266_pcap_uart`

	i've simplified the app in my repo to debug this issue ,
	instead	of writing packets to a pcap file over uart, the 
	code checks and prints "EAPOL PACKET DETECTED.." if the 
	recieved packet has `0x88 as the 32nd byte and  0x8e as 33rd byte`
	of the frame (i believe this is how eapol frames are identified) 

### Expected Behavior
Authentication Frames should be captured  by the esp8266

### Actual Behavior
Authentication Frames aren't being captured (or atleast not being passed to the callback function)

### Steps to repropduce
#assuming you have your dev environment setup..

1. git clone https://github.com/huzaifah0x00/esp8266_pcap_uart
2. cd esp8266_pcap_uart
3. make menuconfig # change the channel in sniffer config here if you need to 
3. make flash monitor
4. use another device to authenticate to any AP using wpa2 (same channel as the esp8266 is listening on of course)
5. on the `make monitor` screen wait for the string "EAPOL PACKET DETECTED.." to *not* appear

### Code to reproduce this issue 
i've included a simplified promiscuous callback function here:

```cpp

void sniffer_handler(void* buff, wifi_promiscuous_pkt_type_t type)
{

    wifi_promiscuous_pkt_t *ppkt = (wifi_promiscuous_pkt_t *)buff;
    uint32_t length = ppkt->rx_ctrl.sig_mode ? ppkt->rx_ctrl.HT_length : ppkt->rx_ctrl.legacy_length;
    
    if(type == WIFI_PKT_MGMT) length -= 4; // known bugfix

    // check if we have a authentication frame(eapol)

    //this if came from : https://github.com/G4lile0/ESP32-WiFi-Hash-Monster/blob/0ef6d2403f89b66e89447b31541784cda8ee9b85/ESP32-WiFi-Hash-Monster/ESP32-WiFi-Hash-Monster.ino#L457
    if (( (ppkt->payload[30] == 0x88 && ppkt->payload[31] == 0x8e) ||
            (ppkt->payload[31] == 0x88 && ppkt->payload[32] == 0x8e) || //i see eapol has bytes 32-33 for 0x88-0x8e in wireshark
            (ppkt->payload[32] == 0x88 && ppkt->payload[33] == 0x8e) 
       ))
    {
     
            printf("EAPOL PACKET DETECTED..\n"); // <- THIS STATEMENT IS NEVER REACHED  

    }
}

```

## Other items if possible

- [x] sdkconfig file (https://github.com/huzaifah0x00/esp8266_pcap_uart/blob/master/sdkconfig)

- [x] elf file in the ``build`` folder (https://github.com/huzaifah0x00/esp8266_pcap_uart/blob/master/build/esp8266_pcap_uart.elf)

donghengqaz, 2020-01-03:
I doubt that how do you make sure the type/subtype element is at offset 30 - 33 ?

I mean that before the Wi-Fi control frame head element, there is a hardare wifi packet attribute frame head element. But at different platforms, they are not same although bettown ESP32 and ESP8266, so you may check if is is so.

huzaifah0x00, 2020-01-06:
you may be right (i don't know much about 802.11 frames structure ) 
but i don't think that offset is the problem because even if i omit the statements that check for the frame's subtype at offset 30-33 and send all received [packets over uart...](https://github.com/huzaifah0x00/esp8266_pcap_uart/blob/master/main/sniffer_main.c#L45) 
i still never [see](https://github.com/huzaifah0x00/esp8266_pcap_uart/blob/master/SerialShark.py) any Authentication frames but i can see all other MGMT frames.

huzaifah0x00, 2020-01-14:
can anyone confirm if this behavior is by design or a bug ?

empyfi, 2022-10-02:
you are correct, eapol takes place in a QOS Data Frame (0x88) containing llc header followed by the snap header. Last part at byte 32 and 33 is the ethertype value 0x88 0x8e. eapol then is directly following. For me also the values seem to be empty/not captured.

do you have news on that? bug or by design?

empyfi, 2022-10-02:
you are correct, eapol takes place in a QOS Data Frame (0x28) containing llc header followed by the snap header. Last part at byte 32 and 33 is the ethertype value 0x88 0x8e. eapol then is directly following. For me also the values seem to be empty/now captured.

but or by design?