espressif/ESP8266_RTOS_SDK #791: SPI transmit data on HSPI_HOST is wrong

Bresenham, 2019-12-25:
## Environment

- Development Kit: Wemos D1 mini
- IDF version: a3537856
- Development Env:      Make
- Operating System:     Ubuntu 18.04
- Power Supply:         USB

## Problem Description

The ESP8266 is configured as an SPI-Master and transmits 2 messages to an SPI-Slave. After these two message are transmitted, one SPI-read of 8-bits is performed.
The first message, called `writeMsg` in code, consists of five individual bytes. The second message, called `readMsg` in code, consists of four individual bytes. writeMsg[1:3] and readMsg[1:3] are equal (as you can see in the example code) but for whatever reason, readMsg[1:3] show up as `0` on my LogicAnalyzer even though `writeMsg` is transmitted perfectly fine.
However, if I don't transmit`writeMsg`, `readMsg` is sent correctly.

### Expected Behavior
Bytes readMsg[1:3] should be equal to [0x00, 0x16, 0xC5].

### Actual Behavior
Bytes readMsg[1:3] are [0x00, 0x00, 0x00].

### Steps to repropduce

1. Copy linked code
2. Compile and flash
3. Watch SPI-Pins of HSPI_HOST via LogicAnalyzer / digital oscilloscope

### Code to reproduce this issue
https://gist.github.com/Bresenham/f5c016361e21b9a3711ef5e7565f7874

https://pasteboard.co/IMVfzCV.png

https://pasteboard.co/IMVfHs1.png

## Debug Logs

```
<27>[0;33mW (102718) spi: Using unaligned data may reduce transmission efficiency<27>[0m
```

## Other items if possible

- sdkconfig: https://pastebin.com/R2j5hFLP

donghengqaz, 2020-01-03:
It is hard to using the SPI driver directly, but you can refer to the new SPI [example](https://github.com/espressif/ESP8266_RTOS_SDK/tree/master/examples/peripherals/spi) and the new [SPI high level driver](https://github.com/espressif/ESP8266_RTOS_SDK/blob/master/components/esp8266/driver/hspi_logic_layer.c).

vargavik, 2020-12-14:
Hi, this problem is related with a small error in the spi driver. When the data is unaligned, only the first byte is sent (or read), all the other bytes will be zero. Since your first byte is already zero, you will end up with 3 zeros.  Check out #723.