espressif/ESP8266_RTOS_SDK #793: Unknown CMake command "idf_build_set_property".

emichel11, 2019-12-27:
in the root directory I run cmake. 
I get 
_Unknown CMake command "idf_build_set_property"._
cmake trace
./ESP8266/ESP8266_RTOS_SDK/CMakeLists.txt(4):  unset(compile_options )
./ESP8266/ESP8266_RTOS_SDK/CMakeLists.txt(5):  unset(c_compile_options )
./ESP8266/ESP8266_RTOS_SDK/CMakeLists.txt(6):  unset(cxx_compile_options )
./ESP8266/ESP8266_RTOS_SDK/CMakeLists.txt(7):  unset(compile_definitions )
./ESP8266/ESP8266_RTOS_SDK/CMakeLists.txt(12):  if(CONFIG_COMPILER_OPTIMIZATION_LEVEL_RELEASE )
./ESP8266/ESP8266_RTOS_SDK/CMakeLists.txt(14):  else()
./ESP8266/ESP8266_RTOS_SDK/CMakeLists.txt(15):  list(APPEND compile_options -Og )
./ESP8266/ESP8266_RTOS_SDK/CMakeLists.txt(18):  if(CONFIG_COMPILER_CXX_EXCEPTIONS )
./ESP8266/ESP8266_RTOS_SDK/CMakeLists.txt(20):  else()
./ESP8266/ESP8266_RTOS_SDK/CMakeLists.txt(21):  list(APPEND cxx_compile_options -fno-exceptions )
./ESP8266/ESP8266_RTOS_SDK/CMakeLists.txt(24):  if(CONFIG_COMPILER_DISABLE_GCC8_WARNINGS )
./ESP8266/ESP8266_RTOS_SDK/CMakeLists.txt(44):  if(CONFIG_COMPILER_OPTIMIZATION_ASSERTIONS_DISABLE )
./ESP8266/ESP8266_RTOS_SDK/CMakeLists.txt(48):  if(CONFIG_COMPILER_STACK_CHECK_MODE_NORM )
./ESP8266/ESP8266_RTOS_SDK/CMakeLists.txt(50):  elseif(CONFIG_COMPILER_STACK_CHECK_MODE_STRONG )
./ESP8266/ESP8266_RTOS_SDK/CMakeLists.txt(52):  elseif(CONFIG_COMPILER_STACK_CHECK_MODE_ALL )
CMake Error at CMakeLists.txt:57 (idf_build_set_property):
  Unknown CMake command "idf_build_set_property".

donghengqaz, 2020-01-03:
We can't just input command like "cmake" to build the cmake project, the command "$IDF_PATH/tools/idf.py build" will compile the cmake project.

emichel11, 2020-01-27:
Hi,
I run this command:

EMis-Mac:ESP8266_RTOS_SDK edgar$ $IDF_PATH/tools/idf.py build
Checking Python dependencies...
Python requirements from /Users/edgar/ESP8266/ESP8266_RTOS_SDK/requirements.txt are satisfied.
Executing action: all (aliases: build)
Running cmake in directory /Users/edgar/ESP8266/ESP8266_RTOS_SDK/build
Executing "cmake -G 'Unix Makefiles' -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=0 /Users/edgar/ESP8266/ESP8266_RTOS_SDK"...
Warn about uninitialized values.
CMake Error at CMakeLists.txt:57 (idf_build_set_property):
  Unknown CMake command "idf_build_set_property".

-- Configuring incomplete, errors occurred!
See also "/Users/edgar/ESP8266/ESP8266_RTOS_SDK/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1