espressif/ESP8266_RTOS_SDK #795: C++ WIFI_INIT_CONFIG_DEFAULT failed because of wrong order

THKDev, 2019-12-29:
## Environment

- Development Kit: ESP8266
- IDF version: d18148e6 (branch v3.3)
- Development Env:  make
- Operating System:  Linux
- Power Supply:  external 3.3V

## Problem Description
```
ESP8266_RTOS_SDK/components/esp8266/include/esp_wifi.h:172:1: sorry, unimplemented: non-trivial designated initializers not supported                    
 };                                                                                                                                                                                 
 ^                                                                                                                                                                                  
main/include/wifi.hpp:91:38: note: in expansion of macro 'WIFI_INIT_CONFIG_DEFAULT'                                                               
             wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
```
### Expected Behavior
should compile also in C++ .

### Actual Behavior
failed with compile error.

### Steps to repropduce
compile any program as C++ .

Compile error comes from wrong order of member initialization in macro [WIFI_INIT_CONFIG_DEFAULT.](https://github.com/espressif/ESP8266_RTOS_SDK/blob/release/v3.3/components/esp8266/include/esp_wifi.h#L154) Order in macro does not match struct definition of [wifi_init_config_t.](https://github.com/espressif/ESP8266_RTOS_SDK/blob/release/v3.3/components/esp8266/include/esp_wifi.h#L99)

donghengqaz, 2020-01-03:
I will fix the issues and add a c++ project to test so that the issues will not happen agian.

FayeY, 2020-01-13:
Please have a try with the latest release/v3.3 branch, and feel free to let us know if this issue is still unsolved. Thanks.

THKDev, 2020-01-13:
Looks good to me. Thanks.

closed 2020-01-13 by THKDev