espressif/ESP8266_RTOS_SDK #796: ESP_MQTT API definition missmatch

THKDev, 2019-12-29:
**mqtt_client.h**  differs to  **mqtt_client.c**

[esp_err_t esp_mqtt_client_subscribe(esp_mqtt_client_handle_t client, const char *topic, int qos)](https://github.com/espressif/ESP8266_RTOS_SDK/blob/release/v3.3/components/mqtt/esp-mqtt/include/mqtt_client.h#L112) differs to [int esp_mqtt_client_subscribe(esp_mqtt_client_handle_t client, const char *topic, int qos)](https://github.com/espressif/ESP8266_RTOS_SDK/blob/release/v3.3/components/mqtt/esp-mqtt/mqtt_client.c#L813)

[esp_err_t esp_mqtt_client_unsubscribe(esp_mqtt_client_handle_t client, const char *topic)](https://github.com/espressif/ESP8266_RTOS_SDK/blob/release/v3.3/components/mqtt/esp-mqtt/include/mqtt_client.h#L113) differs to 
[int esp_mqtt_client_unsubscribe(esp_mqtt_client_handle_t client, const char *topic)](https://github.com/espressif/ESP8266_RTOS_SDK/blob/release/v3.3/components/mqtt/esp-mqtt/mqtt_client.c#L836)

donghengqaz, 2020-01-03:
I check the newer esp-mqtt of esp-idf, now the `esp_err_t` of `esp_err_t esp_mqtt_client_unsubscribe(esp_mqtt_client_handle_t client, const char *topic) ` and `esp_err_t esp_mqtt_client_subscribe(esp_mqtt_client_handle_t client, const char *topic, int qos)` has been modified to `int`, and I will do same modification it as soon as possible.

FayeY, 2020-01-13:
Could you have a try with the latest release/v3.3 branch? it should be solved now.

THKDev, 2020-01-13:
Looks good for me. Thanks!

closed 2020-01-13 by THKDev