espressif/ESP8266_RTOS_SDK #797: Serial port error : Framing error

AHafsy, 2019-12-30:
#100  Environment

- Development Kit: ESP-LAUNCHER
- Development Env:      Make
- Operating System:     Windows|Ubuntu
- Power Supply:         USB|external 5V

I'm working on ESP8266 in order to have wifi signal, I followed the steps described on the get started document: downloading RTOS SDK, compiling the program and executing it in order to generate the binary file. Once, it was generated I flash it into the board using ESP8266 Download Tool, and the next step was to send some command to the Development board via Coolterm. Here, an error occurs which disable the connection between the board and CoolTerm.
The error is:  A serial Port Error Occured  104: Framing Error.

I already search in some forums but it is said that this error could come from baud rate mismatch, so I tried all the baud rates and I still got the same error.

Do anyone know how to solve this type of error please?

donghengqaz, 2020-01-03:
What's the OS platfrom you are using, ubuntu, windows  or windows ubuntu subsystem? After you flashing all binaries, the next step is open UART tools to look the outputting log from ESP8266, I don't know your step that "send some command to the Development board via Coolterm".

AHafsy, 2020-01-06:
Thank you for your response.
I'm using Ubuntu to generate binary files and then windows to flash those files into the board using Flash_download_tool.
Coolterm is the serial port debug tool that I use, it's all new for me and I don't know how can I check the outputting log from ESP8266 using it. I have already configure its settings but there are no outputs. Can you please specify that for me?

AHafsy, 2020-01-06:
I have some updates, after several research I find out that an extra binary file should be downloaded and flashed into the board. So, I did it and it works, I don't really understand why but it seems to be working. 
here is the binary file that I flashed into the board: File: At firmware bin1.54.zip
from: https://www.electrodragon.com/w/ESP8266_AT-Command_firmware

Do you have any idea of this binary file? Because I want to understand why does it work instead of the other binary files.
Thank you