espressif/ESP8266_RTOS_SDK #798: Can't connect wifi ,wake up from light_sleep

thisdream, 2020-01-02:
## Environment

- Development Kit:esp8266v3.3
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    a3537856
- Development Env:      [vscode]
- Operating System:     [Windows]
- Power Supply:         [Battery]

## Problem Description

I need the device to go into lightsleep periodically.I refer to esp32 v3.3's deom put the device to sleep.

### Expected Behavior
 the device to go into lightsleep periodically,and auto canconnect wifi .

### Actual Behavior
the device canconnect wifi

### Code to reproduce this issue

```cpp
/* Light sleep example
   This example code is in the Public Domain (or CC0 licensed, at your option.)
   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/

#include "driver/gpio.h"
#include "esp_event_loop.h"
#include "esp_log.h"
#include "esp_sleep.h"
#include "esp_timer.h"
#include "esp_wifi.h"
#include "freertos/FreeRTOS.h"
#include "freertos/event_groups.h"
#include "freertos/task.h"
#include "nvs_flash.h"
#include "rom/ets_sys.h"
#include <string.h>
/*set the ssid and password via "idf.py menuconfig"*/
#define EXAMPLE_ESP_WIFI_SSID "8888d"
#define EXAMPLE_ESP_WIFI_PASS "88888"
/* FreeRTOS event group to signal when we are connected*/
static EventGroupHandle_t wifi_event_group;
static const char *TAG = "sleep_wifi";
/* Most development boards have "boot" button attached to GPIO0.
 * You can also change this to another pin.
 */
#define BUTTON_GPIO_NUM_DEFAULT 5

/* "Boot" button on GPIO0 is active low */
#define BUTTON_WAKEUP_LEVEL_DEFAULT 0
//extern uint32_t esp_sleep_get_wakeup_cause(void);
static esp_err_t event_handler(void *ctx, system_event_t *event)
{
    /* For accessing reason codes in case of disconnection */
    system_event_info_t *info = &event->event_info;

    switch (event->event_id)
    {
    case SYSTEM_EVENT_STA_START:
        esp_wifi_connect();
        break;
    case SYSTEM_EVENT_STA_GOT_IP:
        ESP_LOGI(TAG, "got ip:%s",
                 ip4addr_ntoa(&event->event_info.got_ip.ip_info.ip));
        break;
    case SYSTEM_EVENT_AP_STACONNECTED:
        ESP_LOGI(TAG, "station:" MACSTR " join, AID=%d",
                 MAC2STR(event->event_info.sta_connected.mac),
                 event->event_info.sta_connected.aid);
        break;
    case SYSTEM_EVENT_AP_STADISCONNECTED:
        ESP_LOGI(TAG, "station:" MACSTR "leave, AID=%d",
                 MAC2STR(event->event_info.sta_disconnected.mac),
                 event->event_info.sta_disconnected.aid);
        break;
    case SYSTEM_EVENT_STA_DISCONNECTED:
        ESP_LOGE(TAG, "Disconnect reason : %d", info->disconnected.reason);
        if (info->disconnected.reason == WIFI_REASON_BASIC_RATE_NOT_SUPPORT)
        {
            /*Switch to 802.11 bgn mode */
            esp_wifi_set_protocol(ESP_IF_WIFI_STA, WIFI_PROTOCAL_11B | WIFI_PROTOCAL_11G | WIFI_PROTOCAL_11N);
        }
        esp_wifi_connect();
        break;
    default:
        break;
    }
    return ESP_OK;
}
void app_main(void)
{
    wifi_event_group = xEventGroupCreate();

    tcpip_adapter_init();
    ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL));

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));
    wifi_config_t wifi_config = {
        .sta = {
            .ssid = EXAMPLE_ESP_WIFI_SSID,
            .password = EXAMPLE_ESP_WIFI_PASS},
    };

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config));
    ESP_ERROR_CHECK(esp_wifi_start());
    /* Configure the button GPIO as input, enable wakeup */
    const int button_gpio_num = BUTTON_GPIO_NUM_DEFAULT;
    const int wakeup_level = BUTTON_WAKEUP_LEVEL_DEFAULT;
    gpio_config_t config = {
        .pin_bit_mask = (1ULL << button_gpio_num),
        .mode = GPIO_MODE_INPUT};
    ESP_ERROR_CHECK(gpio_config(&config));
    gpio_wakeup_enable(button_gpio_num,
                       wakeup_level == 0 ? GPIO_INTR_LOW_LEVEL : GPIO_INTR_HIGH_LEVEL);
    config.pin_bit_mask = (1ULL << 0);
    config.mode = GPIO_MODE_OUTPUT;
    ESP_ERROR_CHECK(gpio_config(&config));
    gpio_set_level((1ULL << 0), 0);
    while (true)
    {
        /* Wake up in 2 seconds, or when button is pressed */
        esp_sleep_enable_timer_wakeup(10000000);
        esp_sleep_enable_gpio_wakeup();

        /* Wait until GPIO goes high */
        if (gpio_get_level(button_gpio_num) == wakeup_level)
        {
            printf("Waiting for GPIO%d to go high...\n", button_gpio_num);
            do
            {
                vTaskDelay(pdMS_TO_TICKS(10));
            } while (gpio_get_level(button_gpio_num) == wakeup_level);
        }

        printf("Entering light sleep\r\n");
        /* To make sure the complete line is printed before entering sleep mode,
         * need to wait until UART TX FIFO is empty:
         */

        //uart_wait_tx_idle_polling(CONFIG_ESP_CONSOLE_UART_NUM);

        /* Get timestamp before entering sleep */
        vTaskDelay(pdMS_TO_TICKS(5000));
        int64_t t_before_us = esp_timer_get_time();

        /* Enter sleep mode */
        printf("will esp_light_sleep_start\r\n");
        printf("t_before_us=%lld ms\r\n", t_before_us);
        fflush(stdout);
        esp_light_sleep_start();
        /* Execution continues here after wakeup */

        /* Get timestamp after waking up from sleep */
        int64_t t_after_us = esp_timer_get_time();

        /* Determine wake up reason */
        const char *wakeup_reason;
        wakeup_reason = "timer";

        printf("Returned from light sleep, reason: %s, t=%lld ms, slept for %lld ms\r\n",
               wakeup_reason, t_after_us / 1000, (t_after_us - t_before_us) / 1000);
    }
}
```

## Debug Logs

```
I (2311) sleep_wifi: got ip:192.168.1.62
will esp_light_s*}х5
t_before_us=ld ms
Returned from light sleep, reason: timer, t=ld ms, slept for ld ms
Entering light sleep
will esp_light_sleep_start
t_before_us=ld ms
Returned from light sleep, reason: timer, t=ld ms, slept for ld ms
Entering light sleep
I (11403) wifi: bcn_timout,ap_probe_send_start
I (13906) wifi: ap_probe_send_over, rest wifi status to disassoc
I (13908) wifi: state: 5 -> 0 (100)
I (13911) wifi: pm stop
E (13913) sleep_wifi: Disconnect reason : 200
E (14281) sleep_wifi: Disconnect reason : 201
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
will esp_light_sleep_start
t_before_us=ld ms
Returned from light sleep, reason: timer, t=ld ms, slept for ld ms
Entering light sleep
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
E (16948) sleep_wifi: Disconnect reason : 201
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
E (19614) sleep_wifi: Disconnect reason : 201
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
will esp_light_sleep_start
t_before_us=ld ms
```

Thank your hlep

donghengqaz, 2020-01-03:
I had tested your code and found it is bug from light sleep process, the I tried my best to fix the issues. I upload the patch file which I tested and it worked well, could you please help me to test it ? If it is OK, I will merge it to the master and backport to branch release/v3.3.

[0001-fix-esp8266-fix-light-sleep-time-error-when-waking-u.txt](https://github.com/espressif/ESP8266_RTOS_SDK/files/4018392/0001-fix-esp8266-fix-light-sleep-time-error-when-waking-u.txt)

thisdream, 2020-01-06:
Thank you for your great help,I will test and give you a response as soon as possible

thisdream, 2020-01-07:
Thank you for your reply.The probability of such bugs has been greatly reduced.However, when I ping the IP of the device, information displays that the request has timed out or cannot access the target host

thisdream, 2020-01-12:
Is there any progress?

thisdream, 2020-01-13:
Sorry for not reading the email in time.The probability of such bugs has been greatly reduced.However, when I  ping the IP of the device, information displays that the request has  timed out or cannot access the target host.
Thank espressif and all the engineers for the excellent products and high quality service.

Junhao-Espressif, 2020-01-17:
Hi @thisdream 

Could you try call  ```esp_wifi_set_ps(WIFI_PS_NONE)``` after wifi start and test again!

Thanks!

thisdream, 2020-01-22:
I call esp_wifi_set_ps(WIFI_PS_NONE) ,no work .
I have provided all the code and the corresponding development environment.
You can send me your revised code,if it is work in you drive

ShrinathN, 2020-05-15:
I have the same issue but interestingly I am not using any of the sleep functions anywhere in my code.

ShrinathN, 2020-05-22:
> I have the same issue but interestingly I am not using any of the sleep functions anywhere in my code.

I fixed this by writing the following line in the first few lines, before connecting to a station.

    esp_wifi_set_ps(WIFI_PS_NONE);

BlackMoonBefall, 2020-11-24:
Meeting the same problem,  request has  timed out or cannot access the target host. after wake up from light sleep. It seems doesn't start the wifi_connect event .Compare with before light sleep, missing the following logs :

BlackMoonBefall, 2020-11-24:
[0;32mI (19973) wifi: state: 0 -> 2 (b0)[0m
[0;32mI (19984) wifi: state: 2 -> 3 (0)[0m
[0;32mI (19998) wifi: state: 3 -> 5 (10)[0m
[0;32mI (20001) wifi: pm start, type: 2[0m
[0;32mI (20981) tcpip_adapter: sta ip: 192.168.13.58, mask: 255.255.255.0, gw: 192.168.43.1[0m

twt2008, 2020-12-10:
I have the same problem,This problem has bothered me for a long time,I didn't activate hibernation mode,Just timing 10 seconds, use ping function to get delay time and VCC voltage, WiFi signal strength, stack space to upload to my server.

twt2008, 2020-12-10:
I (46097007) DQ_SYS: heap_size = 34464

[22:18:53.575]收←◆I (46102868) wifi: bcn_timout,ap_probe_send_start

[22:18:56.081]收←◆I (46105367) wifi: ap_probe_send_over, rest wifi status to disassoc
I (46105369) wifi: state: 5 -> 0 (100)
I (46105371) wifi: pm stop
E (46105382) DQ_WIFI: Disconnect reason : 200
E (46105430) TRANS_TCP: tcp_poll_write select error 113, errno = Software caused connection abort, fd = 54
E (46105539) MQTT_CLIENT: Error write data or timeout, written len = -1
E (46105616) MQTT_CLIENT: Error sending ping
E (46105664) MQTT_CLIENT: Can't send ping, disconnected
E (46105723) DQ_WIFI: Disconnect reason : 201
I (46105774) MQTT_CLIENT: Reconnect after 10000 ms
I (46105826) DQ_MQTT: MQTT_EVENT_DISCONNECTED
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!

[22:18:57.745]收←◆error: pll_cal exceeds 2ms!!!
I (46107068) DQ_SYS: heap_size = 37092
error: pll_cal exceeds 2ms!!!

Yahuma, 2023-01-29:
Hello @donghengqaz, @Junhao-Espressif 

Can you please write if Espressif still maintenance ESP8266_RTOS_SDK?

Best Regards