espressif/ESP8266_RTOS_SDK #806: mbedtls_ssl_handshake return -0x5180

Vbansal1996, 2020-01-07:
Hardware description - nodemcu 12E module

Problem:
I am trying to make **GET** request to the https url. I am connected to WPA2 Enterprise network.
When ever I tried to make the connection with the host url I am getting this error.

[image]

I am not able to Identify the error reason, or meaning of this error.
This error value is not even listed in ssl.h file.

Can I get any help in this.

francescolavra, 2020-01-07:
The -0x5180 error code corresponds to MBEDTLS_ERR_MD_ALLOC_FAILED, i.e. you are running out of heap space.
If you are using the master or release/v3.3 branches of the SDK code, please make sure to pull the latest commits, since there was a bug (which got fixed in commit https://github.com/espressif/ESP8266_RTOS_SDK/commit/e0b73761b947449b2cce192e8b1dce14ac2d61ca) that in my case was causing the same MBEDTLS_ERR_MD_ALLOC_FAILED error even when there actually was enough free space in the heap. If you are not using the master or release/v3.3 branches, or if you still get the same error after pulling the latest commits, then you are really running out of heap space.

Vbansal1996, 2020-01-07:
I am using master and using the latest commited SDK in master, and I am getting this issue.

There is any way to resolve this issue?

francescolavra, 2020-01-07:
You could try to decrease the TLS maximum incoming fragment length option value with `make menuconfig`: go to `> Component config > mbedTLS > TLS maximum incoming fragment length` and change the value from the default 16384 to, say, 8192. This of course works only if your server doesn't send TLS fragments larger than 8192 bytes.

Vbansal1996, 2020-01-07:
ok Thanks I will check and update.

Vbansal1996, 2020-01-09:
I make the changes and not getting the error but now It is giving me the 502 bad gateway issue

Vbansal1996, 2020-01-10:
Thanks for your help I am able to resolve the issue.

closed 2020-01-10 by Vbansal1996

XingYu2535, 2024-02-03:
> You could try to decrease the TLS maximum incoming fragment length option value with `make menuconfig`: go to `> Component config > mbedTLS > TLS maximum incoming fragment length` and change the [...]

Thanks!!