espressif/ESP8266_RTOS_SDK #810: Unable to load program image if it stored after 0x200000 (Does my solution makes sense?)

peleshenko, 2020-01-12:
Hi! 
In SDK v3.2 (and 3.3 I think) we can't use flash space after 0x200000 (~2Mb) for program. The bootloader at boot loader_utility.c:858 throws error in case if you image is stored after 0x200000. However I found that we can use 2nd parameter of Cache_Read_Enable to access space after 0x200000, I've adde following code:

```
...

#define ESP_CACHE3_ADDR_MAX 0x300000

...

 else if ( image_data->start_addr >= ESP_CACHE2_ADDR_MAX // Map 2m-3m space
               && image_data->start_addr + image_data->image_len < ESP_CACHE3_ADDR_MAX ) {
        map = 0;
        bank = 1;
        bank = 0;
    } 

... 

Cache_Read_Enable(map, bank, SOC_CACHE_SIZE);  
```

I need to use more flash for my OTA update scenario. Modified code seems to be working good. Does this modification makes sense? May I have some issue with it? Why it was not designed in this way from the very begging? Do you want me to create a pool request?

bschwind, 2020-03-02:
I just ran into this issue as well, I similarly have an OTA scenario where I'd like to use more than 2 MB of flash.

bschwind, 2020-03-03:
I did finally get a working solution, but it's a little disappointing because I still can't use all 4MB of my flash memory:

```bash
$ make partition_table

*******************************************************************************
# Espressif ESP32 Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,16K,
otadata,data,ota,0xd000,8K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,400K,
ota_0,app,ota_0,0x110000,400K,
ota_1,app,ota_1,0x174000,400K,
*******************************************************************************
```

Each app image is limited to 400K. This way the two OTA partitions both fit cleanly in a 1MB block, and it stays below 2MB.

Aditionally, I originally had `ota_0` at `0x100000` but it would have boot errors after the OTA update:

```
I (63) boot: ESP-IDF v3.2-354-g4dee04c0-dirty 2nd stage bootloader
I (64) boot: compile time 00:22:40
I (74) qio_mode: Enabling default flash chip QIO
I (74) boot: SPI Speed      : 40MHz
I (79) boot: SPI Mode       : QIO
I (85) boot: SPI Flash Size : 4MB
I (91) boot: Partition Table:
I (97) boot: ## Label            Usage          Type ST Offset   Length
I (108) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (119) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (131) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (143) boot:  3 factory          factory app      00 00 00010000 00064000
I (154) boot:  4 ota_0            OTA app          00 10 00100000 00064000
I (166) boot:  5 ota_1            OTA app          00 11 00164000 00064000
I (178) boot: End of partition table
I (184) esp_image: segment 0: paddr=0x00100010 vaddr=0x40210010 size=0x412b4 (266932) map
0x40210010: _flash_cache_start at ??:?

I (287) esp_image: segment 1: paddr=0x001412cc vaddr=0x402512c4 size=0x08174 ( 33140) map
I (299) esp_image: segment 2: paddr=0x00149448 vaddr=0x3ffe8000 size=0x00678 (  1656) load
I (300) esp_image: segment 3: paddr=0x00149ac8 vaddr=0x40100000 size=0x00a30 (  2608) load
I (312) esp_image: segment 4: paddr=0x0014a500 vaddr=0x40100a30 size=0x0584c ( 22604) load
I (332) boot: Loaded app from partition at offset 0x100000
Fatal exception (28): 
epc1=0x40210f08, epc2=0x00000000, epc3=0x4023d410, excvaddr=0x0000000c, depc=0x00000000
0x40210f08: _flash_cache_start at ??:?

0x4023d410: esp_light_sleep_internal at /Users/brian/projects/gc-mod/ESP8266_RTOS_SDK/components/esp8266/source/esp_sleep.c:396

Fatal exception (28): 
epc1=0x40210f08, epc2=0x00000000, epc3=0x4023d410, excvaddr=0x0000000c, depc=0x00000000
0x40210f08: _flash_cache_start at ??:?

0x4023d410: esp_light_sleep_internal at /Users/brian/projects/gc-mod/ESP8266_RTOS_SDK/components/esp8266/source/esp_sleep.c:396

Fatal exception (28): 
epc1=0x40210f08, epc2=0x00000000, epc3=0x4023d410, excvaddr=0x0000000c, depc=0x00000000
0x40210f08: _flash_cache_start at ??:?

0x4023d410: esp_light_sleep_internal at /Users/brian/projects/gc-mod/ESP8266_RTOS_SDK/components/esp8266/source/esp_sleep.c:396

Fatal exception (28): 
epc1=0x40210f08, epc2=0x00000000, epc3=0x4023d410, excvaddr=0x0000000c, depc=0x00000000
0x40210f08: _flash_cache_start at ??:?

0x4023d410: esp_light_sleep_internal at /Users/brian/projects/gc-mod/ESP8266_RTOS_SDK/components/esp8266/source/esp_sleep.c:396
```

Finally, I'm on an older commit at the moment (4dee04c03b4d6f3f437d327cf09efac14326c13f), because `esp_ota_begin()` was failing when trying to erase the flash memory for the new OTA app image.

Seems like there's a bit of work to be done getting OTA stable and running on flash sizes greater than 4MB.

bschwind, 2020-03-03:
Seems like this commit may have just fixed it? I'll test it later

3f9eccddf657db0e83a6604c5bee61b0b0831d82