espressif/ESP8266_RTOS_SDK #812: Secure Boot and other Security Options Missing

Gungeoneer, 2020-01-14:
Using the latest version of the SDK, I wanted to ask why Secure Boot and other Security Options are missing when the target platform is selected as ESP8266? Is there a way to manually perform these operations? Or do I have to enable something in `make menuconfig` to be able to see these options?

nuclearcat, 2020-01-14:
AFAIK there is no secure boot and options on ESP8266 hardware.
It is available only on ESP32.

Gungeoneer, 2020-01-14:
Thanks for the reply.
So, is there any way to add security to the bootloader?

nuclearcat, 2020-01-14:
Secure boot need support in hardware. You can add security by obscurity, some kind of encryptor to OTA, but still it is possible (and even trivial) to download firmware and reverse engineer it.
ESP32 is much more secure for this matters, because it keeps decryption key in specially designated (and not readable) chunk of memory.