espressif/ESP8266_RTOS_SDK #814: esp8266 工作一晚上后第二天wifi一直 2 -> 0 (200) 和 2 -> 0 (200)，请帮忙看一下，谢谢

luweiqi, 2020-01-16:
1.esp8266 mqtt工程 间隔10s左右上报一条消息，工作一晚后第二天WiFi连接不上了，一直在连接断开。剩余堆内存10k不到点，我又给它换了个ssid，一会儿后堆内存耗尽了，WiFi没有日志打出来了，只剩下堆内存打印日志。具体请看日志。
2.以前用esp8266_rtos_sdk 3.2 运行也出现过类似问题，目前刚刚移植到3.3
3.工程用esp-mqtt/tcp为demo修改，接收串口命令上报mqtt server。
4.运行着HEAP_TRACE_LEAKS检测，10s间隔打印1次残留内存

wifi无法连接时打印情况：
I (55823913) wifi: state: 0 -> 2 (b0)
D (55824913) esp_alink_data_transport: total free heap size: 9808
I (55824915) wifi: state: 2 -> 0 (200)
E (55824916) MQTT_EXAMPLE: Disconnect reason : 2
D (55824921) esp_alink_data_transport: post_handle free heap size: 1024B
D (55824933) esp_alink_data_transport: read_handle free heap size: 852B

堆内存中出现许多未释放的：
I (56323340) heap_trace: mem @0x3fffd7d0 size 92 caller file esf_buf.c line 433
I (56323352) heap_trace: mem @0x3fffd82c size 84 caller file esf_buf.c line 400
I (56323364) heap_trace: mem @0x3fffd880 size 44 caller file esf_buf.c line 417

出现问题后配置新的ssid，内存逐渐消耗完：
        Line 2474: D (55932103) esp_alink_data_transport: total free heap size: 9748
	Line 2505: D (55933124) esp_alink_data_transport: total free heap size: 8728
	Line 2510: D (55935076) esp_alink_data_transport: total free heap size: 8728
	Line 2516: D (55936097) esp_alink_data_transport: total free heap size: 8728
	Line 2521: D (55937118) esp_alink_data_transport: total free heap size: 8224
	Line 2527: D (55938138) esp_alink_data_transport: total free heap size: 8224
	Line 2532: D (55939151) esp_alink_data_transport: total free heap size: 7224
	Line 2538: D (55940172) esp_alink_data_transport: total free heap size: 6976
	Line 2543: D (55941184) esp_alink_data_transport: total free heap size: 6480
	Line 2550: D (55942205) esp_alink_data_transport: total free heap size: 6232
	Line 2790: D (55946071) esp_alink_data_transport: total free heap size: 6168
	Line 2795: D (55947091) esp_alink_data_transport: total free heap size: 5668
	Line 2801: D (55948119) esp_alink_data_transport: total free heap size: 4924
	Line 2805: D (55949151) esp_alink_data_transport: total free heap size: 3188
	Line 2809: D (55950162) esp_alink_data_transport: total free heap size: 2940
	Line 2814: D (55951172) esp_alink_data_transport: total free heap size: 2940
	Line 2822: D (55952197) esp_alink_data_transport: total free heap size: 1948
	Line 2828: D (55953218) esp_alink_data_transport: total free heap size: 1948
	Line 2833: D (55954239) esp_alink_data_transport: total free heap size: 1452
	Line 2839: D (55955260) esp_alink_data_transport: total free heap size: 1204
	Line 3172: D (55960263) esp_alink_data_transport: total free heap size: 460

配置wifi  ssid 相关代码：
        ESP_LOGD(TAG, "get Bluefi confirm ");
        ESP_ERROR_CHECK( esp_wifi_get_config(ESP_IF_WIFI_STA, &wifi_config) );
        rw_len = sizeof(wifi_config.sta.ssid);
        ESP_ERROR_CHECK(get_nvs_str("ssid", (char*)(wifi_config.sta.ssid), &rw_len));
        rw_len = sizeof(wifi_config.sta.password);
        ESP_ERROR_CHECK(get_nvs_str("passwd", (char*)(wifi_config.sta.password), &rw_len));
        ESP_ERROR_CHECK( esp_wifi_disconnect() );
        ESP_ERROR_CHECK( esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config) );
        ESP_ERROR_CHECK( esp_wifi_connect() );

[Serial-COM10 serial-com10 2020 01 16 09-35-47.log](https://github.com/espressif/ESP8266_RTOS_SDK/files/4068980/Serial-COM10.serial-com10.2020.01.16.09-35-47.log)

Junhao-Espressif, 2020-01-17:
hi @luweiqi 

看上去像是应用内存泄漏。建议用下面文件中的函数debug一下：
https://github.com/espressif/ESP8266_RTOS_SDK/blob/master/components/heap/include/esp_heap_trace.h

luweiqi, 2020-01-17:
hi@Junhao-Espressif
我这个是开了内存泄漏检测的，附件中有残留内存情况打印输出
esf_buf.c这个文件我找不到，应该是不对外开放的lib中的文件吧，把内存都消耗完了
当时一直是连不上wifi的状况内存还有10k，但是那个wifi是正常的，我其他设备都能连上那个wifi

LiThirteen, 2020-02-04:
@luweiqi  请问这个问题解决了嘛，我出现的问题是重启后wifi连接不上，SSID与PASSWORD都正确，但就是一直连接断开这样的循环。

luweiqi, 2020-02-04:
@LiThirteen 官方没有给我明确答复,问题没有解决呢，我怀疑是我开着内存泄漏检测可用堆空间不够了导致WiFi那边出问题。
你的问题有可能是WiFi初始化的时候没有把ssid和password配置上去吧，我这边暂时没有遇到这种问题

LiThirteen, 2020-02-04:
@luweiqi  都有初始化wifi配置的，也是wifi一直 2 -> 0 (200) 和 2 -> 0 (200)，在多次连接不上的时候，我强制重新配置wifi信息一下就又好了