espressif/ESP8266_RTOS_SDK #815: wake up from light_sleep，still not connected to the wifi

thisdream, 2020-01-16:
## Environment

- Development Kit:
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    da769106
- Development Env: [vscode]
- Operating System: [Windows]
- Power Supply: [Battery]
## Problem Description

The drive wake up from lightsleep,even if I call esp_wifi_connect(),can't connnect wifi anymore!
I ping IP of the drive,can't find drive .

### Expected Behavior
      the device to go into lightsleep periodically,and can canconnect wifi .
### Actual Behavior
       the device can't connect wifi anymore 

### Code to reproduce this issue

```cpp

//#include "driver/rtc_io.h"
//#include "driver/uart.h"
#include "driver/gpio.h"
#include "esp_event_loop.h"
#include "esp_log.h"
#include "esp_sleep.h"
#include "esp_timer.h"
#include "esp_wifi.h"
#include "freertos/FreeRTOS.h"
#include "freertos/event_groups.h"
#include "freertos/task.h"
#include "nvs_flash.h"
#include "rom/ets_sys.h"
#include <string.h>
// #include <stdio.h>
// #include <stdlib.h>
// #include <string.h>
// #include <sys/time.h>
// #include <time.h>
/*set the ssid and password via "idf.py menuconfig"*/
#define EXAMPLE_ESP_WIFI_SSID "***********"
#define EXAMPLE_ESP_WIFI_PASS "************"
/* FreeRTOS event group to signal when we are connected*/
static EventGroupHandle_t wifi_event_group;
static const char *TAG = "sleep_wifi";
/* Most development boards have "boot" button attached to GPIO0.
 * You can also change this to another pin.
 */
#define BUTTON_GPIO_NUM_DEFAULT 5

/* "Boot" button on GPIO0 is active low */
#define BUTTON_WAKEUP_LEVEL_DEFAULT 0
//extern uint32_t esp_sleep_get_wakeup_cause(void);
static esp_err_t event_handler(void *ctx, system_event_t *event)
{
    /* For accessing reason codes in case of disconnection */
    system_event_info_t *info = &event->event_info;

    switch (event->event_id)
    {
    case SYSTEM_EVENT_STA_START:
        esp_wifi_connect();
        break;
    case SYSTEM_EVENT_STA_GOT_IP:
        ESP_LOGI(TAG, "got ip:%s",
                 ip4addr_ntoa(&event->event_info.got_ip.ip_info.ip));
        break;
    case SYSTEM_EVENT_AP_STACONNECTED:
        ESP_LOGI(TAG, "station:" MACSTR " join, AID=%d",
                 MAC2STR(event->event_info.sta_connected.mac),
                 event->event_info.sta_connected.aid);
        break;
    case SYSTEM_EVENT_AP_STADISCONNECTED:
        ESP_LOGI(TAG, "station:" MACSTR "leave, AID=%d",
                 MAC2STR(event->event_info.sta_disconnected.mac),
                 event->event_info.sta_disconnected.aid);
        break;
    case SYSTEM_EVENT_STA_DISCONNECTED:
        ESP_LOGE(TAG, "Disconnect reason : %d", info->disconnected.reason);
        if (info->disconnected.reason == WIFI_REASON_BASIC_RATE_NOT_SUPPORT)
        {
            /*Switch to 802.11 bgn mode */
            esp_wifi_set_protocol(ESP_IF_WIFI_STA, WIFI_PROTOCAL_11B | WIFI_PROTOCAL_11G | WIFI_PROTOCAL_11N);
        }
        esp_wifi_connect();
        break;
    default:
        break;
    }
    return ESP_OK;
}
void app_main(void)
{
    int sleep_time = 0;
    wifi_event_group = xEventGroupCreate();

    tcpip_adapter_init();
    ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL));

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));
    wifi_config_t wifi_config = {
        .sta = {
            .ssid = EXAMPLE_ESP_WIFI_SSID,
            .password = EXAMPLE_ESP_WIFI_PASS},
    };

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config));
    ESP_ERROR_CHECK(esp_wifi_start());
    /* Configure the button GPIO as input, enable wakeup */
    const int button_gpio_num = BUTTON_GPIO_NUM_DEFAULT;
    const int wakeup_level = BUTTON_WAKEUP_LEVEL_DEFAULT;
    gpio_config_t config = {
        .pin_bit_mask = (1ULL << button_gpio_num),
        .mode = GPIO_MODE_INPUT};
    ESP_ERROR_CHECK(gpio_config(&config));
    gpio_wakeup_enable(button_gpio_num,
                       wakeup_level == 0 ? GPIO_INTR_LOW_LEVEL : GPIO_INTR_HIGH_LEVEL);
    config.pin_bit_mask = (1ULL << 0);
    config.mode = GPIO_MODE_OUTPUT;
    ESP_ERROR_CHECK(gpio_config(&config));
    gpio_set_level((1ULL << 0), 0);
    vTaskDelay(pdMS_TO_TICKS(5000));
    while (true)
    {
        /* Wake up in 2 seconds, or when button is pressed */
        //esp_wifi_fpm_do_sleep(10000000);
        //esp_wifi_fpm_set_sleep_type(WIFI_LIGHT_SLEEP_T);
        esp_sleep_enable_timer_wakeup(1*1000000);
        esp_sleep_disable_wakeup_source(ESP_SLEEP_WAKEUP_GPIO);
        //esp_sleep_enable_gpio_wakeup();

        /* Wait until GPIO goes high */
        if (gpio_get_level(button_gpio_num) == wakeup_level)
        {
            printf("Waiting for GPIO%d to go high...\n", button_gpio_num);
            do
            {
                vTaskDelay(pdMS_TO_TICKS(10));
            } while (gpio_get_level(button_gpio_num) == wakeup_level);
        }

        printf("Entering light sleep\r\n");
        /* To make sure the complete line is printed before entering sleep mode,
         * need to wait until UART TX FIFO is empty:
         */

        //uart_wait_tx_idle_polling(CONFIG_ESP_CONSOLE_UART_NUM);

        /* Get timestamp before entering sleep */
        
        int64_t t_before_us = esp_timer_get_time();

        /* Enter sleep mode */
        //printf("will esp_light_sleep_start\r\n");
        fflush(stdout);
        sleep_time++;
        printf("will SLEEP=%d\r\n", sleep_time);
        esp_light_sleep_start();
        printf("wake up=%d\r\n", sleep_time);
        esp_wifi_set_mode(WIFI_MODE_STA); //	set	station	mode
        esp_wifi_connect();               //	connect	to	AP
        //printf("t_before_us=%lld ms\r\n", t_before_us);
        /* Execution continues here after wakeup */
        vTaskDelay(pdMS_TO_TICKS(10000));
        /* Get timestamp after waking up from sleep */
        int64_t t_after_us = esp_timer_get_time();

        /* Determine wake up reason */
        const char *wakeup_reason;
        wakeup_reason = "timer";
        // switch (esp_sleep_get_wakeup_cause())
        // {
        // case ESP_SLEEP_WAKEUP_TIMER:
        //     wakeup_reason = "timer";
        //     break;
        // case ESP_SLEEP_WAKEUP_GPIO:
        //     wakeup_reason = "pin";
        //     break;
        // default:
        //     wakeup_reason = "other";
        //     break;
        // }

        // printf("Returned from light sleep, reason: %s, t=%lld ms, slept for %lld ms\r\n",
        //        wakeup_reason, t_after_us / 1000, (t_after_us - t_before_us) / 1000);
    }
}
## Debug Logs

```
```

thank your for help

Junhao-Espressif, 2020-01-17:
Hi @thisdream 

Please use this example https://github.com/espressif/ESP8266_RTOS_SDK/tree/master/examples/wifi/power_save

Thanks!

BlackMoonBefall, 2020-11-18:
Met the same issue, have you solved it ?  after wake up from light_sleep , i try "esp_wifi_connect()", it just print ".... socket connect failed errno=118"

BlackMoonBefall, 2020-11-18:
> Hi @Junhao-Espressif
Is that you mean to use this example to replace the plan "using light sleep to save power" ? 
But I wonder if any way to reconnect on wifi after wake up from light_sleep.  Is it possible ? 
urgent issue , waiting for your reply.
thanks a lot.