espressif/ESP8266_RTOS_SDK #819: Send Data From Client to server in Chunk

Vbansal1996, 2020-01-22:
## Environment

- Development Kit: ESP8266 CHip
- IDF version Master
- Development Env:      Make
- Operating System:     Ubuntu
- Power Supply:         external 3.3V

## Problem Description

Hi I need to send image data to http server. I am reading this image data from external flash in size of 23 KB each and send it to server. total image size if 250KB. 

But I am getting some error.

Steps I had followed:

```cpp
    esp_http_client_set_url(client,custom_url ? custom_url_server : "http: test.com");
    printf("sent image header\r\n");
    esp_http_client_open(client, image_t.image1_size + image_t.image2_size + 32);
    printf("sent image header\r\n");
    esp_http_client_write(client, (const char *)image_t.image_header, 32);
```

First I am sending 32 Bytes image header then I send the image data. But every time I am getting this error in logs

## Debug Logs

```
I (56181) wifi: state: 0 -> 2 (b0)
I (56189) wifi: state: 2 -> 3 (0)
I (56195) wifi: state: 3 -> 5 (10)
I (56197) wifi: pm start, type: 2
I (57224) event: sta ip: 192.168.2.140, mask: 255.255.255.0, gw: 192.168.2.1
I (57228) Vista-wifi: got ip:192.168.2.140
sent image header
I (69096) wifi: bcn_timout,ap_probe_send_start
I (71600) wifi: ap_probe_send_over, rest wifi status to disassoc
I (71602) wifi: state: 5 -> 0 (100)
I (71604) wifi: pm stop
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4010157e  PS      : 0x00000033  A0      : 0x40223558  A1      : 0x3fff5550  
0x4010157e: uxListRemove at /home/dcl33/Desktop/Vaibhav/esp/ESP8266_RTOS_SDK/components/freertos/freertos/list.c:176

0x40223558: xTaskRemoveFromEventList at /home/dcl33/Desktop/Vaibhav/esp/ESP8266_RTOS_SDK/components/freertos/freertos/tasks.c:4471

A2      : 0x3fff0b5c  A3      : 0x40239568  A4      : 0x00000000  A5      : 0x00000000  
0x40239568: _lock_release_recursive at /home/dcl33/Desktop/Vaibhav/esp/ESP8266_RTOS_SDK/components/newlib/newlib/port/locks.c:214

A6      : 0xc87ecaba  A7      : 0x00000000  A8      : 0x3fff55e0  A9      : 0x00000090  
A10     : 0x402955a1  A11     : 0x00000070  A12     : 0x3fff0b44  A13     : 0x3fff0b5c  
A14     : 0x00000000  A15     : 0x3fff55b0  SAR     : 0x0000001a  EXCCAUSE: 0x0000001d  

Backtrace: 0x4010157e:0x3fff5550 0x40223558:0x3fff5550 0x4022268e:0x3fff5560 0x40259dc7:0x3fff55a0 0x4021b37f:0x3fff55b0 0x4021c751:0x3fff5610 0x4021247b:0x3fff5620 0x402124ec:0x3fff5630 
0x4010157e: uxListRemove at /home/dcl33/Desktop/Vaibhav/esp/ESP8266_RTOS_SDK/components/freertos/freertos/list.c:176

0x40223558: xTaskRemoveFromEventList at /home/dcl33/Desktop/Vaibhav/esp/ESP8266_RTOS_SDK/components/freertos/freertos/tasks.c:4471

0x4022268e: xQueueGenericSend at /home/dcl33/Desktop/Vaibhav/esp/ESP8266_RTOS_SDK/components/freertos/freertos/queue.c:2332

0x40259dc7: esp_event_send at /home/dcl33/Desktop/Vaibhav/esp/ESP8266_RTOS_SDK/components/esp8266/source/event_loop.c:92

0x4021b37f: ieee80211_sta_new_state at ??:?

0x4021c751: chm_check_same_channel at ??:?

0x4021247b: ets_timer_handler_isr at ??:?

0x402124ec: ets_timer_handler_isr at ??:?

```

Is there any another method to post Images to server in chunk or any other format?

copercini, 2020-05-08:
I'm also hitting this `ap_probe_send_over, rest wifi status to disassoc` issue during OTA, have you made any progress on this?

Vbansal1996, 2020-05-08:
Yeah I do able to resolve this, but I had changed the method to push my image data to server. Now I am not using this logic.