espressif/ESP8266_RTOS_SDK #821: RTC-Memory does not keep content after esp_restart()

tinu11, 2020-01-27:
## Environment

- Development Kit:
- IDF version (``git rev-parse --short HEAD`` to get the commit id.):   456efce

- Development Env:      [VSC]
- Operating System:     [Ubuntu]
- Power Supply:         [external 5V]

## Problem Description
RTC memory variable does not keep content after esp_restart()

### Expected Behavior
RTC memory should keep content after esp_restart() ( and deep_sleep() - which works under some sircumstances)

### Actual Behavior
RTC-memory variable is always reinitialized after esp_restart()

### Steps to repropduce

1. just run below app_main
2. ...

### Code to reproduce this issue

#include "freertos/FreeRTOS.h"
#include "esp_log.h"

RTC_DATA_ATTR uint32_t nr_reports_in_rtcmem = 0;

void app_main(){
    ESP_LOGI(TAG, "[APP] Startup..");
    ESP_LOGI(TAG, "[APP] Free memory: %d bytes", esp_get_free_heap_size());
    ESP_LOGI(TAG, "[APP] IDF version: %s", esp_get_idf_version());

    esp_log_level_set("*", ESP_LOG_VERBOSE);

    ESP_LOGI(TAG, "The current report count is: %d", nr_reports_in_rtcmem);
    nr_reports_in_rtcmem+= 1;

    vTaskDelay(5000 / portTICK_PERIOD_MS);
    esp_restart();
}

## Debug Logs
rll▒▒|▒l▒|

▒▒I (49) boot: ESP-IDF v3.2-352-gda769106 2nd stage bootloader
I (49) boot: compile time 16:56:49
I (49) boot: SPI Speed      : 40MHz
I (49) boot: SPI Mode       : DIO
I (53) boot: SPI Flash Size : 4MB
I (56) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (78) boot:  2 factory          factory app      00 00 00010000 000f0000
I (84) boot: End of partition table
I (87) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x2324c (143948) map
I (155) esp_image: segment 1: paddr=0x00033264 vaddr=0x4023325c size=0x049e8 ( 18920) map
I (164) esp_image: segment 2: paddr=0x00037c54 vaddr=0x60001200 size=0x00004 (     4) load
I (164) esp_image: segment 3: paddr=0x00037c60 vaddr=0x3ffe8000 size=0x0093c (  2364) load
I (170) esp_image: segment 4: paddr=0x000385a4 vaddr=0x40100000 size=0x00a30 (  2608) load
I (178) esp_image: segment 5: paddr=0x00038fdc vaddr=0x40100a30 size=0x05434 ( 21556) load
I (194) boot: Loaded app from partition at offset 0x10000
I (215) system_api: Base MAC address is not set, read default base MAC address from EFUSE
I (219) system_api: Base MAC address is not set, read default base MAC address from EFUSE
phy_version: 1155.0, 6cb3053, Nov 11 2019, 17:31:08, RTOS new
I (273) phy_init: phy ver: 1155_0
I (276) reset_reason: RTC reset 2 wakeup 0 store 2360220716, reason is 2
I (279) TEMP_SENSOR: [APP] Startup..
I (280) TEMP_SENSOR: [APP] Free memory: 101168 bytes
I (285) TEMP_SENSOR: [APP] IDF version: v3.2-352-gda769106
I (290) TEMP_SENSOR: The current report count is: 0
rll▒▒|▒l▒|

▒▒I (49) boot: ESP-IDF v3.2-352-gda769106 2nd stage bootloader
I (49) boot: compile time 16:56:49
I (49) boot: SPI Speed      : 40MHz
I (49) boot: SPI Mode       : DIO
I (53) boot: SPI Flash Size : 4MB
I (56) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (78) boot:  2 factory          factory app      00 00 00010000 000f0000
I (84) boot: End of partition table
I (87) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x2324c (143948) map
I (155) esp_image: segment 1: paddr=0x00033264 vaddr=0x4023325c size=0x049e8 ( 18920) map
I (164) esp_image: segment 2: paddr=0x00037c54 vaddr=0x60001200 size=0x00004 (     4) load
I (164) esp_image: segment 3: paddr=0x00037c60 vaddr=0x3ffe8000 size=0x0093c (  2364) load
I (170) esp_image: segment 4: paddr=0x000385a4 vaddr=0x40100000 size=0x00a30 (  2608) load
I (178) esp_image: segment 5: paddr=0x00038fdc vaddr=0x40100a30 size=0x05434 ( 21556) load
I (194) boot: Loaded app from partition at offset 0x10000
I (215) system_api: Base MAC address is not set, read default base MAC address from EFUSE
I (219) system_api: Base MAC address is not set, read default base MAC address from EFUSE
phy_version: 1155.0, 6cb3053, Nov 11 2019, 17:31:08, RTOS new
I (271) phy_init: phy ver: 1155_0
I (273) reset_reason: RTC reset 2 wakeup 0 store 3, reason is 2
I (276) TEMP_SENSOR: [APP] Startup..
I (278) TEMP_SENSOR: [APP] Free memory: 101168 bytes
I (282) TEMP_SENSOR: [APP] IDF version: v3.2-352-gda769106
I (287) TEMP_SENSOR: The current report count is: 0

## Other items if possible

dsptech, 2020-02-02:
Hi,
It work as intended. The rtc data are preserved in a sleep cycle only. 
See this note in esp_attr.h:

`// Forces data into RTC memory.
// Any variable marked with this attribute will keep its value
// during a deep sleep / wake cycle.
#define RTC_DATA_ATTR _SECTION_ATTR_IMPL(".rtc.data", __COUNTER__)`

regards.

tinu11, 2020-02-04:
Ohh! 
I thought I saw a table which stated that RTC-Memory gets only initialized after power up and hard reset.
BTW; the note does not tell anything about restart and timeout. If it would say  ONLY during a deep sleep...

Thanks for the clarification.
Regards

closed 2020-02-04 by tinu11

dsptech, 2020-02-25:
Hi,
I noticed now the last commit of the sdk. 
I think that the new attribute RTC_NOINIT_ATTR (and __NOINIT_ATTR too) in esp_attr.h could be useful to you.
Regards.