espressif/ESP8266_RTOS_SDK #824: Can it do WPS_TYPE_PIN but from the esp side without the need to enter anything on the router ?

minanagehsalalma, 2020-02-05:
Just like a normal wps pin connect app

FireWinter, 2020-02-11:
The sdk supports wps pbc and wps pin, if you are using wps pin, you should first get the pin code from Espressif's device and then enter it into the router just link normal wps pin process.

minanagehsalalma, 2020-02-11:
> Espressif's device and then enter it into the router just link normal wps pin process

@FireWinter  nob ... that's not how a normal wps connection work ... i want it to connect with the pin without me entring anything on the router !

FireWinter, 2020-02-11:
wps pin method can't do that, you can use wps pbc, the user simply has to push a button, either an actual or a virtual one.

minanagehsalalma, 2020-02-11:
> wps pin method can't do that
@FireWinter  why ? it's the typical use of it !

FireWinter, 2020-02-11:
The wps pin method requires that the pin code on the device and the router match each other. If you do not want to do anything with the router, the premise is that you need to obtain the pin code of the router and then configure it on the device.

minanagehsalalma, 2020-02-11:
"the premise is that you need to obtain the pin code of the router and then configure it on the device." 

yes this what i do on android the pin is mostly on the back of the router .
@FireWinter

FireWinter, 2020-02-11:
Currently, the SDK wps sample code does not support wps pin configuration, it only supports generating a pin code randomly.

minanagehsalalma, 2020-02-11:
> Currently, the SDK wps sample code does not support wps pin configuration, it only supports generating a pin code randomly.

@FireWinter  will it support it in the future ?

FireWinter, 2020-02-11:
We will consider whether to support in future versions, please wait.

minanagehsalalma, 2020-02-11:
@FireWinter  ok thanks keep me updated :)