espressif/ESP8266_RTOS_SDK #827: Make all include path relative to component folder.

sake402, 2020-02-12:
I have a stack of libraries that would not play well with idf. So I have been compiling by using eclipse to build my project into a library file generating libProject.a. In postbuild step, I copy the libProject.a over libmain.a in the build/main folder. Then I run make in msys2 again to generate the elf and upload to the esp8266.

This approach works quite satisfactorily for me. Only that in my eclipse project, If I intend to use say freertos, it should have been enough to simply do

#include "freertos/port/esp8266/include/freertos/FreeRTOSConfig.h"
#include "freertos/include/freertos/FreeRTOS.h"
#include "freertos/include/freertos/task.h"

while I add the component folder to include path

But these included files pull in other header file not relative to the component folder. for example in "portable.h" there is "#include "esp_heap_caps.h". But this file is inside "heap/include" and I now have to add that path to my include paths. At the end of the day I have tons of includes in my paths and the annoying part is one has to keep hunting for where these files are located once the compiler complains about it.

My suggestion is to make all #include relative to the component folder so that #include "esp_heap_caps.h" becomes #include "heap/include/esp_heap_caps.h". A single Include path to the component folder will then be neccessary to pull in all dependencies.

dsptech, 2020-02-17:
Hi,
it happens because "heap" is another component and the (indirect) use of its internal makefile is required. For details, refer to the component structure described in the Esp32 ESP-IDF docs, ver. 3.x. (It is not the latest).

In the case of a already written library and you have to compile it every time, the easiest way I found is to write a "component.mk" file (to allow the library to be used as a component) and add the path component to the project's makefile .

I use this way (and with Eclipse IDE too) with my libraries.

dsptech, 2020-02-17:
https://docs.espressif.com/projects/esp-idf/en/v3.3.1/api-guides/build-system.html#using-the-build-system

sake402, 2020-02-17:
@dsptech Thanks for your response. But I still think having a fully qualified relative path to an header file seems better. For example there are places in the component libraries where you see #include "platform.h". Now searching for plarform.h in the component folder reveals there are two hits. One in mbedtls and another in esp-mqtt. It is then hard to tell which of these in particular is being referenced from a compilation unit. And worse is the fact that the one the compiler sees now depends on the order of the include paths. So the compilation of one of these module might fail if passed the same order of -I when compiling.
Having these as "mbedtls/include/platform.h" or "esp-mqtt/include/platform.h" will eliminate such ambiguity. 

My libraries span across a lot of folders with structure
MyLibrary
--Component1
----Component1.c
----Component1.h
--Component2
----Component2.c
----Component2.h
 And I find it easier to just drag and drop the root folder to these libraries to eclipse. And simply add path to MyLibrary to include path.

If I want to use component.mk as you suggested I will have create tons of it as I currently almost 1000 folders in MyLibrary folder. 

All references withing the library therefore starts from MyLibrary Path. If Component2 depends on Component1 then I have #include "MyLibrary/Component1/Component1.h".

Like this I find the code easier to read as the exact dependency being imported is obvious from the #include relative path

dsptech, 2020-02-17:
Hi,
I already written dozens of component.mk files, mostly the copy and paste of the same file (fortunately).
Anyway, I agree with you:  the fully qualified relative path should be better.

However, I think your question is very difficult for Espressif guys due to the resulting work and (maybe) portability issues versus the ESP-IDF framework.

I will wait for their answer.
Regards.