espressif/ESP8266_RTOS_SDK #832: v3.2 分支， IOS12上使用ESPtouch, smart config 无反应

YoYo0304, 2020-02-25:
v3.2 分支， IOS12上使用ESPtouch, smart config 无反应。如下打印：
[image]

请帮忙定位一下

YoYo0304, 2020-02-25:
图片没显示出来，直接贴一下打印
        I (62) boot: ESP-IDF v3.2-46-g7179056f 2nd stage bootloader                                          p{lrp
I (62) boot: compile time 14:54:35
I (72) qio_mode: Enabling default flash chip QIO
I (72) boot: SPI Speed      : 40MHz
I (73) boot: SPI Mode       : QIO
I (75) boot: SPI Flash Size : 4MB
I (79) boot: Partition Table:
I (83) boot: ## Label            Usage          Type ST Offset   Length
I (90) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (98) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (105) boot:  2 factory          factory app      00 00 00010000 000f0000
I (113) boot: End of partition table
I (117) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x4b488 (308360) map
I (229) esp_image: segment 1: paddr=0x0005b4a0 vaddr=0x3ffe8000 size=0x0054c (  1356) load
I (230) esp_image: segment 2: paddr=0x0005b9f4 vaddr=0x3ffe854c size=0x0019c (   412) load
I (237) esp_image: segment 3: paddr=0x0005bb98 vaddr=0x40100000 size=0x06ba4 ( 27556) load
I (254) boot: Loaded app from partition at offset 0x10000
I (295) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (295) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
,ծ;32mI (355) phy_init: phy ver: 1058_13
I (355) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2
SC version: V2.8.1
I (3865) sc: SC_STATUS_FINDING_CHANNEL

FireWinter, 2020-02-25:
可以在 smartconfig example里将 log 打开看能否收到一些配网包：修改如图：
[image]
 或者用最新的SDK master 版本试一下看是否有同样的问题
@YoYo0304

FireWinter, 2020-02-25:
图片无法显示，开log是在example中修改这一行：
ESP_ERROR_CHECK( esp_smartconfig_start(sc_callback, 1) );

YoYo0304, 2020-02-26:
今天又试了一下 V3.2分支，能配上，当时会出错，另外又试了下android app, 感觉安卓的能配上的概率更高。ios比较慢。报错是203  或者 2， 这个原因是什么意思？

[image]

FireWinter, 2020-02-26:
203是连接路由器失败，2是认证有问题（密码错误会打印这个），可以测试下配置好ssid和pwd直接去连这台路由器看是否有同样的问题

YoYo0304, 2020-02-26:
试了simple wifi这个例子，一样的反应。这个路由器配置 在V2.XX的时候是可以的。用esp32也可以连接

FireWinter, 2020-02-26:
路由器型号是什么？用最新的 master 或 v3.3 的版本试试看是否有这个问题 ？ @YoYo0304

YoYo0304, 2020-02-26:
master 也是一样，如下打印
[image]

FireWinter, 2020-02-26:
看log是连接过程由于路由器 Power Capability element 有问题导致的，问题必现的吗？测试的路由器具体设么型号 ？能否配置成 bg mode再试一下？ @YoYo0304

YoYo0304, 2020-02-26:
我问一下型号。但是esp32就是没事呢，如果是路由器的问题 应该都有问题啊

FireWinter, 2020-03-02:
hi，有问到型号吗？上面失败原因 802.11 spec上有解释的，从log 推测路由器回的association response的 Status Codes 应该是： Cannot support all requested capabilities in the Capability Information field, 你试下别的型号路由器，应该没有这个问题。 
这个问题没遇到过，我们需要知道这台路由器型号才能排查兼容性问题。@YoYo0304

YoYo0304, 2020-03-03:
恩  其他路由器确实可以连上。这个型号我再确认一下