espressif/ESP8266_RTOS_SDK #837: WiFi station example works but error: event: system event loop not initialized via esp_event_loop_init

hiperiondev, 2020-03-02:
Before connect have the message: event: system event loop not initialized via esp_event_loop_init

E (6371) event: system event loop not initialized via esp_event_loop_init
E (6373) wifi: send SYSTEM_EVENT_STA_START error
I (6374)example_wifi: wifi_init_sta finished.
I (7498) wifi: state: 0 -> 2 (b0)
I (7541) wifi: state: 2 -> 3 (0)
I (7546) wifi: state: 3 -> 5 (10)
I (7548) wifi: pm start, type: 2
E (8786) event: system event loop not initialized via esp_event_loop_init
E (8789) wifi: send SYSTEM_EVENT_STA_CONNECTED failed
I (11005) tcpip_adapter: sta ip: 192.168.100.106, mask: 255.255.255.0, gw: 192.168.100.1
I (11009) example_wifi: got ip:192.168.100.106
I (11011) example_wifi: connected to ap SSID:EXAMPLE

FireWinter, 2020-03-02:
Could you try with the latest master branch to see if there are still problems?

hiperiondev, 2020-03-05:
With the lastest branch works ok

closed 2020-03-05 by hiperiondev