espressif/ESP8266_RTOS_SDK #839: OTA Example Fails on esp_ota_begin()

bschwind, 2020-03-03:
## Environment

- Development Kit: ESP8266 NodeMCU Board (4MB flash)
- IDF version: 3f9eccddf657db0e83a6604c5bee61b0b0831d82 (currently `master` at the time of filing this issue)
- Development Env: Make
- Operating System: MacOS
- Power Supply: USB

## Problem Description

When running the OTA example, as `esp_ota_begin()` is called, it delays for awhile and then the device is reset, printing garbage in the process (probably just the bootloader at a different baud rate). Point is, `esp_ota_begin()` doesn't return properly.

With some printf tracing I followed it at least until `spi_flash_erase_sector()`, which originates up the call stack from `esp_partition_erase_range()` inside of `esp_ota_begin()`

### Expected Behavior

`esp_ota_begin()` should succeed.

### Actual Behavior

`esp_ota_begin()` causes a crash.

### Code to reproduce this issue

Try the example `examples/system/ota/native_ota/2+MB_flash/new_to_new_no_old`

A `git bisect` seems to indicate 0003ef6572b1cffa737d857b40530315d426ed5b as the commit which introduced the breakage.

walmis, 2020-03-08:
Can confirm this too

mptei, 2020-03-09:
Can confirm this too with a Wemos D1 (ESP8266EX, 4MB flash).
Also found that `spi_flash_erase_sector()` is the problem. I changed the example to directly call `spi_flash_erase_sector()` for some sectors. Sometimes it runs 1 or 2 times, but mostly gets stuck in the first call.

Found also that, when WiFi is not connected `spi_flash_erase_sector()` runs without problems. Means, before the connect method is called in the example `spi_flash_erase_sector()` runs OK. After the connect method is called it gets stucked.

mptei, 2020-03-09:
Now it works again on master. I guess it has something to do with commit 6da560a4.

bschwind, 2020-03-09:
@mptei thanks for the updates, I'll try it out again.

Do you by chance happen to know what's being referenced in the commit message:

```
See merge request sdk/ESP8266_RTOS_SDK!1328
```

I'm not quite sure how development on this repo is structured.

mptei, 2020-03-09:
Sorry I also don't have more knowledge about that.

bschwind, 2020-03-09:
Verified as fixed with commit 6da560a41c99776f9f4ccef83c37780643367beb

closed 2020-03-09 by bschwind