espressif/ESP8266_RTOS_SDK #840: Linkage errors in SDK

ds21h, 2020-03-03:
#Environment 

- Development Kit:
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): c4f549db
    //bd6ea4393c7d2f059fc4decc70f1ec3eb3597268
- Development Env:      Make and Eclipse
- Operating System:     Window 10
- Power Supply:         NA

## Problem Description

Since a couple of days I cannot compile my project anymore because there are a few unresolved references within the SDK. I tried refreshing the SDK but no luck. 
Does anyone have an idea on this?

### Expected Behavior

No unresolved references in the SDK.

### Actual Behavior

Unresolved references in SDK.

### Steps to reproduce

Clean and build ea project
de to reproduce this issue

## Debug Logs

$ make
Toolchain path: D:/Source/ESP_IDF/msys32/opt/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc
Toolchain version: crosstool-ng-1.22.0-100-ge567ec7b
Compiler version: 5.2.0
Python requirements from D:/Source/ESP_IDF/msys32/home/Jan/esp/ESP8266_RTOS_SDK/requirements.txt are satisfied.
LD build/Switch.elf
D:/Source/ESP_IDF/msys32/home/Jan/esp/switch/build/lwip\liblwip.a(sys_arch.o):(.literal.sys_init+0x14): undefined reference to `pthread_key_create'
D:/Source/ESP_IDF/msys32/home/Jan/esp/switch/build/lwip\liblwip.a(sys_arch.o):(.literal.sys_thread_sem_init+0x4): undefined reference to `pthread_setspecific'
D:/Source/ESP_IDF/msys32/home/Jan/esp/switch/build/lwip\liblwip.a(sys_arch.o):(.literal.sys_thread_sem_get+0x0): undefined reference to `pthread_getspecific'
D:/Source/ESP_IDF/msys32/home/Jan/esp/switch/build/lwip\liblwip.a(sys_arch.o): In function `sys_init':
D:/Source/ESP_IDF/msys32/home/Jan/esp/ESP8266_RTOS_SDK/components/lwip/port/esp8266/freertos/sys_arch.c:543: undefined reference to `pthread_key_create'
D:/Source/ESP_IDF/msys32/home/Jan/esp/switch/build/lwip\liblwip.a(sys_arch.o): In function `sys_thread_sem_init':
D:/Source/ESP_IDF/msys32/home/Jan/esp/ESP8266_RTOS_SDK/components/lwip/port/esp8266/freertos/sys_arch.c:543: undefined reference to `pthread_setspecific'
D:/Source/ESP_IDF/msys32/home/Jan/esp/switch/build/lwip\liblwip.a(sys_arch.o): In function `sys_thread_sem_get':
D:/Source/ESP_IDF/msys32/home/Jan/esp/ESP8266_RTOS_SDK/components/lwip/port/esp8266/freertos/sys_arch.c:543: undefined reference to `pthread_getspecific'
D:/Source/ESP_IDF/msys32/home/Jan/esp/switch/build/lwip\liblwip.a(vfs_lwip.o):(.literal.esp_vfs_lwip_sockets_register+0x2c): undefined reference to `esp_vfs_register_fd_range'
D:/Source/ESP_IDF/msys32/home/Jan/esp/switch/build/lwip\liblwip.a(vfs_lwip.o): In function `esp_vfs_lwip_sockets_register':
D:/Source/ESP_IDF/msys32/home/Jan/esp/ESP8266_RTOS_SDK/components/lwip/port/esp8266/vfs_lwip.c:73: undefined reference to `esp_vfs_register_fd_range'
D:/Source/ESP_IDF/msys32/home/Jan/esp/switch/build/esp_http_server\libesp_http_server.a(httpd_main.o):(.literal.httpd_server+0xc): undefined reference to `select'
D:/Source/ESP_IDF/msys32/home/Jan/esp/switch/build/esp_http_server\libesp_http_server.a(httpd_main.o): In function `httpd_server':
D:/Source/ESP_IDF/msys32/home/Jan/esp/ESP8266_RTOS_SDK/components/esp_http_server/src/httpd_main.c:113: undefined reference to `select'
collect2.exe: error: ld returned 1 exit status
make: *** [/home/Jan/esp/ESP8266_RTOS_SDK/make/project.mk:510: /home/Jan/esp/switch/build/Switch.elf] Fout 1

ds21h, 2020-03-08:
Problem is solved after completely recreating the build environment. I have no idea what went wrong.

closed 2020-03-08 by ds21h

dsptech, 2020-05-16:
Hi @ds21h ,
I experienced the same same problem by updating an old project. It seems that VSF and PTHREAD components are mandatory now. Their exclusion by the menuconfig cause the undefined references you reported (from files: sys_arch.c and vfs_lwip.c).
Likely, by reinstalling the environment you also enabled the two components above.
This behavior is likely unwanted. It may be advised to reopen the issue, so we may receive feedback from the developers.
Regards.