espressif/ESP8266_RTOS_SDK #844: Guru Meditation Error: Core 0 panic'ed (IllegalInstruction)  ESP-IDF v3.3-rc1-dirty 2nd stage bootloader

llcxyz, 2020-03-14:
ESP8266
IDF 版本: v3.3-rc1
当 CONFIG_CONSOLE_UART_NUM = 1 
也就是调试输出口到uart1时. bootloader 载入后会崩溃.
Guru Meditation Error: Core 0 panic'ed (IllegalInstruction)
发现是esp2866/source/rom.c 
uart_tx_wait_idle 方法,在启动时第一次调用时.
clk_div.val 为0.    uart1还没初始化就调用了.才导致崩溃.

临时修补如下:

`void uart_tx_wait_idle(uint8_t uart_no)
{
    uint32_t tx_bytes;
    uint32_t baudrate, byte_delay_us;
    uart_dev_t *const UART[2] = {&uart0, &uart1};
    uart_dev_t *const uart = UART[uart_no];

    if (uart->clk_div.val==0) {
        baudrate = 115200;
    }
    else {
        baudrate = (UART_CLK_FREQ / (uart->clk_div.val & 0xFFFFF));
    }
    
    byte_delay_us = (uint32_t)(10000000 / baudrate);

    do {
        tx_bytes = uart->status.txfifo_cnt;
        /* either tx count or state is non-zero */
    } while (tx_bytes);

    ets_delay_us(byte_delay_us);
}`

llcxyz, 2020-03-14:
`
    /**
     * The API "register_chipv6_phy" will modify the APB frequency to 80MHz,
     * so UARTs must be flush here, then reconfigurate the UART frequency dividor
     */
    uart_tx_wait_idle(0); // 可能是这个地方. 没仔细测试.
    uart_div_modify(0, UART_CLK_FREQ / uart_baudrate);

    uart_tx_wait_idle(1);
    uart_div_modify(1, UART_CLK_FREQ / uart_baudrate);

    rtc_init_clk(local_init_data);`

Yahuma, 2022-12-16:
ESP8266_RTOS_SDK\components\esp8266\source\phy_init.c:

```

esp_err_t esp_phy_rf_init(...) {
     ...

       /**
     * The API "register_chipv6_phy" will modify the APB frequency to 80MHz,
     * so UARTs must be flush here, then reconfigurate the UART frequency dividor
     */
    uart_tx_wait_idle(0);
    uart_tx_wait_idle(1);

    int ret = register_chipv6_phy(local_init_data);
    
    #ifdef CONFIG_CONSOLE_UART_BAUDRATE
    const uint32_t uart_baudrate = CONFIG_CONSOLE_UART_BAUDRATE;
    #else
    const uint32_t uart_baudrate = 74880; // ROM default baudrate
    #endif

    uart_div_modify(0, UART_CLK_FREQ / uart_baudrate);
    uart_div_modify(1, UART_CLK_FREQ / uart_baudrate);

    if (ret) {
        ESP_LOGI(TAG, "phy register error, ret:%d", ret);
    }
    
    ...
 
```