espressif/ESP8266_RTOS_SDK #848: wifi stops working 15 minutes after the start of the program

kormilicinkostia, 2020-03-19:
- Development Kit:
- IDF version b0e5afe3
- Development Env:    idf.py
- Operating System:     Ubuntu
- Power Supply:         external 3.3V

## Problem Description
I am using Esp8266 in WIFI_MODE_APSTA mode. My device successfully connects to the selected network and creates an access point. After 15-20 minutes, I see that the module turns off Wi-Fi (the access point I created is lost, and esp8266 disconnects from the selected Wi-Fi station). In the logs, I see the following message:

```
D (1774806) tcpip_adapter: if0 dhcpc cb
D (1774809) tcpip_adapter: if0 start ip lost tmr: enter
D (1774813) tcpip_adapter: if0 start ip lost tmr: interval=120
D (1774818) tcpip_adapter: if0 start ip lost tmr: enter
D (1774823) tcpip_adapter: if0 start ip lost tmr: already started
D (1774913) system_event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:dreamkas8, ssid_len:9, bssid:00:00:00:00:00:00, reason:201 (no ap found)
D (1774923) event: running post WIFI_EVENT:5 with handler 0x40225694 on loop 0x40107fa0
0x40225694: event_handler at /home/pavel/git/viki-wifi/application/main/wifi.c:229
```
After that, the message in an endless loop displays the following text:

```
D (1774806) tcpip_adapter: if0 dhcpc cb
D (1774809) tcpip_adapter: if0 start ip lost tmr: enter
D (1774813) tcpip_adapter: if0 start ip lost tmr: interval=120
D (1774818) tcpip_adapter: if0 start ip lost tmr: enter
D (1774823) tcpip_adapter: if0 start ip lost tmr: already started
D (1774913) system_event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:dreamkas8, ssid_len:9, bssid:00:00:00:00:00:00, reason:201 (no ap found)
D (1774923) event: running post WIFI_EVENT:5 with handler 0x40225694 on loop 0x40107fa0
0x40225694: event_handler at /home/pavel/git/viki-wifi/application/main/wifi.c:229

E (1774932) WIFI: connect to the AP fail
D (1774935) event: running post WIFI_EVENT:5 with handler 0x40262560 on loop 0x40107fa0
0x40262560: handle_sta_disconnected at /home/pavel/esp/ESP8266_RTOS_SDK/components/tcpip_adapter/event_handlers.c:123
```
You can restore normal system behavior only by rebooting the power board

### Expected Behavior
I want esp8266 to work stably in WIFI_MODE_APSTA mode
### Actual Behavior
Turn off the wi-fi module 15-20 minutes after the start of work
### Steps to repropduce

1) Turn on the wifi module in WIFI_MODE_APSTA mode
2) Connect wi-fi to the selected network
3) Wait 15-20 minutes

### Code to reproduce this issue

```cpp
static void event_handler(void* arg, esp_event_base_t event_base,
                          int32_t event_id, void* event_data)
{
    if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_START)
    {
        ESP_LOGI(TAG, "SYSTEM_EVENT_STA_START");
        esp_wifi_connect();
    }
    else if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_DISCONNECTED)
    {
        ip_addres[0]='\0';
        esp_wifi_connect();
        ESP_LOGE(TAG,"connect to the AP fail");
    }
    else if (event_base == IP_EVENT && event_id == IP_EVENT_STA_GOT_IP)
    {
        ip_event_got_ip_t* event = (ip_event_got_ip_t*) event_data;
        ESP_LOGI(TAG, "got ip:" IPSTR, IP2STR(&event->ip_info.ip));
        sprintf(ip_addres, "%s", ip4addr_ntoa(&event->ip_info.ip));
    }else if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_AP_STACONNECTED)
[... 282 lines cut ...]
D (1812308) system_event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:dreamkas8, ssid_len:9, bssid:00:00:00:00:00:00, reason:201 (no ap found)
D (1812314) event: running post WIFI_EVENT:5 with handler 0x40225694 on loop 0x40107fa0
0x40225694: event_handler at /home/pavel/git/viki-wifi/application/main/wifi.c:229

E (1812323) WIFI: connect to the AP fail
D (1812326) event: running post WIFI_EVENT:5 with handler 0x40262560 on loop 0x40107fa0
0x40262560: handle_sta_disconnected at /home/pavel/esp/ESP8266_RTOS_SDK/components/tcpip_adapter/event_handlers.c:123

D (1812333) tcpip_adapter: if0 start ip lost tmr: enter
D (1812338) tcpip_adapter: if0 start ip lost tmr: already started
D (1815184) system_event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:dreamkas8, ssid_len:9, bssid:00:00:00:00:00:00, reason:201 (no ap found)
D (1815190) event: running post WIFI_EVENT:5 with handler 0x40225694 on loop 0x40107fa0
0x40225694: event_handler at /home/pavel/git/viki-wifi/application/main/wifi.c:229

E (1815199) WIFI: connect to the AP fail
D (1815202) event: running post WIFI_EVENT:5 with handler 0x40262560 on loop 0x40107fa0
0x40262560: handle_sta_disconnected at /home/pavel/esp/ESP8266_RTOS_SDK/components/tcpip_adapter/event_handlers.c:123
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
[sdkconfig.txt](https://github.com/espressif/ESP8266_RTOS_SDK/files/4354813/sdkconfig.txt)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

mriksman, 2020-04-19:
Confirm I am having similar issues. Length of time though is variable.

My code starts in STA mode and connects to the router fine. If the connection is lost (or I use my configuration button), it will start in APSTA mode. Normally I'll be able to see the soft AP. 

But when this issue kicks up, I can't see the softAP nor will it connect to the router. Even though the router is fine.

JonathanWitthoeft, 2020-06-24:
I am also experiencing this issue. I have been debugging for weeks. In my case it uses STA mode only and the time to failure is random. It has taken 1 hour to 48 hours to fail and is variable every time. When I get the 201 (no ap found) error, I have tried multiple variations of stopping wifi, deinit wifi, re-configuring wifi, re-init wifi, and restarting wifi. It never resolves the issue and the AP is never found again. I am convinced that the WiFi component is not scanning at all and reporting AP not found. As soon as the ESP8266 is power cycled, it reconnects to the AP.