espressif/ESP8266_RTOS_SDK #852: IDF_PATH common across ESP32 and ESP8266 IDF

vortex314, 2020-03-27:
## Environment - Linux

Both this repo and the esp-idf repo supposes that IDF_PATH environment variable points to the location of the repo on your local system. 
Since I'm trying to write code for both platforms, I cannot set that fixed in the .profile login. 
Can I handle this in another way ? I'm using Codelite for development and invoke a Makefile for the build process.

Thanks for any suggestion you could provide.

setkeh, 2020-10-19:
I am also in this position it would be nice to not have to use a script to set these environment variables and make them configurable in the SDK.

setkeh, 2021-07-24:
Nearly 12 Months onward and this is still a Problem. Using Arduino IDE / scripts for ESP01's so i dont have to mess up my ESP32's is getting beyond a Joke.

yhsb2k, 2021-11-27:
Please, support the following pull request: https://github.com/espressif/ESP8266_RTOS_SDK/pull/1085

It allows to set IDF_PATH automatically (if you are using CMake build) and pass it to child processes to avoid build error. So, you don't need to set IDF_PATH explicitly.