espressif/ESP8266_RTOS_SDK #854: Cannot Make OTA bin for ota_1 Using CMake - 1MB Flash Issue?

mriksman, 2020-03-28:
I'm certainly no expert, but shouldn't the OTA of an ESP8266 be similar to ESP32 now? Using ESP8266 RTOS SDK, shouldn't the process be;

- Create a partition with `ota_0` and `ota_1`
- Load (via serial) an application into `ota_0`
- Create the `hello_world` example, and simply `idf build` to create the `bin` file
- Use the application in `ota_0` to load the `hello_world.bin` into `ota_1`

But it doesn't work. `eps_ota_end()` complains;

```
D (48942) esp_image: reading image header @ 0x70000
D (48949) esp_image: image header: 0xe9 0x05 0x02 0x01 40210a9c
I (48957) esp_image: segment 0: paddr=0x00070010 vaddr=0x40210010 size=0x2493c (149820) map
0x40210010: _flash_cache_start at ??:?

I (48997) esp_image: segment 1: paddr=0x00094954 vaddr=0x4023494c size=0x04bc0 ( 19392) map
0x4023494c: ieee80211_send_mgmt at ??:?

I (49002) esp_image: segment 2: paddr=0x0009951c vaddr=0x3ffe8000 size=0x00640 (  1600) 
I (49010) esp_image: segment 3: paddr=0x00099b64 vaddr=0x40100000 size=0x00a50 (  2640) 
I (49023) esp_image: segment 4: paddr=0x0009a5bc vaddr=0x40100a50 size=0x05394 ( 21396) 
D (49040) esp_ota_ops: OTA binary start entry 0x10a8c, partition start from 0x70000 to 0x100000

E (49050) esp_ota_ops: **Important**: The OTA binary link data is error, please refer to document <<ESP8266_RTOS_SDK/examples/system/ota/README.md>> for how to generate OTA binaries
E (49076) httpd: Image validation failed, image is corrupted
E (49085) httpd: esp_ota_end failed (ERROR)!
```

Looking at the function `esp_ota_verify_binary` in `esp_ota_ops`, what is this `entry` value? The address of `app_main()` right? Why is this only being checked for 1MB flash?

I'm assuming the `entry` should be something like `0x70a8c` (as my `ota_1` starts at `0x70000` instead of `0x10a8c` where `ota_0` starts at `0x10000`).

So has eboot/bootloader not been updated? Or something else..??

I'm also using the `CMake` system - so can't run `make ota` that is mentioned in the docs. Seems like there is a disconnect between old and new...

mriksman, 2020-03-28:
https://github.com/espressif/ESP8266_RTOS_SDK/commit/3d63a6ca528cf49b18568a28a7754b8835c20f86
Looks like there _were_ some configuration options to manage this start address? These appear to have been removed?

marmurek, 2020-04-25:
Hello, i have similar problem with ESP8266 1MB; I tried to run native_ota/new_to_new_no_old example and i faced a problem...

Step i do:

1. Flash the chip directly with compiled program (make ota erase_flash flash monitor)
2. Recompile previously changing eg. "HELLO WORLD" to "HELLO WORLD2", using "make ota" command
3. Let the chip update with recompiled program from OTA0 partition to OTA1 partition.
4. Recompile again, changing to "HELLO WORLD3"
5. Let the chip update with recompiled soft from OTA1 to OTA0
6. Error - famous message "**Important**: The OTA binary link data is error, please refer to document <<ESP8266_RTOS_SDK/examples/syst
em/ota/README.md>> for how to generate OTA binaries"

I see the message "OTA binary start entry 0x81c98, partition start from 0x10000 to 0x80000" which looks like verify chooses OTA1 entry not recently updated OTA0 (which was selected while updating as log info says) and due to this fails, but why?

When i do not recompile and let continously update from ota0 to ota1 and then back to ota0 it just works. When i change just one letter in code and recompile -> error happens.  Any ideas what I'm doing wrong?

opcli, 2020-10-07:
I would like to report that I run into similar problem.
It seems there is some error in the build system for doing incremental build.

Always use "make clean" then follow up with "make ota" works for me will produce the right ota bin image.
I suspect there is some make dependency error on the ota build process.

The problem is easy to reproduce-able. Just "make ota" follow by touch one of the C source file and do "make ota"
again. It will produce the incorrect ota binary.

It is very annoying I have to make clean and run the full build process every time.