espressif/ESP8266_RTOS_SDK #855: New Commit Doesn't Work (esp_app_desc is not stored in rodata_desc)

mriksman, 2020-03-29:
https://github.com/espressif/ESP8266_RTOS_SDK/commit/525c34b4b625643b430996b5fdc2066b4e4d861b

Downloaded new `master`, did a `fullclean` and compiled using `CMake`. 
```
    const esp_partition_t *next = esp_ota_get_running_partition();
    esp_app_desc_t app_desc;
    esp_ota_get_partition_description(next, &app_desc);
    ESP_LOGI(TAG, "Magic word 0x%8x, App version %s, Proj Name %s, Time %s",
         app_desc.magic_word, app_desc.version, app_desc.project_name, app_desc.time);
```
Output
```
I (605) main: Magic word 0x4026747c, App version , Proj Name `W@|@P��?4a@��, Time 
```

Looking at the HEX dump of the bin file, I can see `esp_image_header_t` and `esp_image_segment_header_t`, but no `esp_app_desc_t`. On an ESP-IDF project I can see the magic bytes at the correct location.

What am I missing?

mriksman, 2020-04-06:
https://github.com/mriksman/esp8266-tests/tree/master/examples/app_desc

QB4-dev, 2024-04-30:
This issue still persists

The reason is that function `esp_ota_get_partition_description` is directly copied from ESP-IDF and will work for ESP32 images where `esp_app_desc_t` is located always in the first flash segment. On current (v3.4) ESP8266 RTOS SDK `esp_app_desc_t` is located in the second one.

Here is my workaround for this without ld script modifications
https://github.com/QB4-dev/esp_http_server_utils/blob/035414049d7c7a6fed011177597de2c29afe73b2/esp_http_server_fota.c#L40