espressif/ESP8266_RTOS_SDK #876: Please Update LWIP

mriksman, 2020-04-19:
As per
https://github.com/espressif/esp-idf/issues/3842
I am having the `assertion "netconn marked closed" failed:` error randomly occur. 

Could we get the lwip submodule updated in this SDK? 
Commit `076270e7324c3d2f36404601038618317f8f0e74`

nsfilho, 2020-05-23:
When this is happen with you? What flags are you using in the sdkconfig for LWIP? I’m asking that, because I have some projects running for a long time, in this release and this error not happen.

If you want, I can share my sdkconfig flags for lwip :)

mriksman, 2020-05-23:
Default LWiP config. 
How do you know you haven’t had this issue? Do you panic-halt, or panic-reset?

nsfilho, 2020-05-23:
> Default LWiP config. 
> 
> How do you know you haven’t had this issue? Do you panic-halt, or panic-reset?

All times when my board reset, send some mqtt additional packages to configure my system :)

Because of that, I believe on this. Another important thing, my uptime is monitored :)

You can see my code, if you would. I have some lwip configurations to handle many “small and important features”, like can halt httpd server.