espressif/ESP8266_RTOS_SDK #879: Not able to connect with AWS-IOT.

pradeep14581, 2020-04-22:
Hi All,

So i am able to successfully configure and flashed the project into the ESP8266 12E. So i followed the following step.
1) setup and configure the AWS iot-core, thingname, security, policy every thing. I enabled the policy also. I got one rootca one private key and one client certificate.
2) Now i took these certificate and private, change their name as mentioned in the readme file as following. Kindly help me to setup and configure the device. What is the issue, why i am not able to connect to the aws. Please help me
certificate.pem.crt
private.pem.key
paste the rootca certificate as aws-root-ca.pem
3) configure the host name and port as mentioned in the readme file document
4) after configuring and following all the steps i flashed the project into the ESP8266 12E
5) open the monitor and i got following log. device is able to connect with the internet but failing to connect with AWS iot core
log

MONITOR
--- idf_monitor on /dev/ttyUSB0 74880 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 7168, room 16 
0x40100000: _stext at ??:?

tail 0
chksum 0xa9
load 0x3ffe8408, len 24, room 8 
tail 0
chksum 0x40
load 0x3ffe8420, len 3544, room 8 
tail 0
chksum 0x75
I (45) boot: ESP-IDF v3.1.1-16-gc2ed2271 2nd stage bootloader
I (45) boot: compile time 12:02:37
I (46) qio_mode: Enabling default flash chip QIO
I (54) boot: SPI Speed      : 80MHz
I (60) boot: SPI Mode       : QIO
I (66) boot: SPI Flash Size : 4MB
I (72) boot: Partition Table:
I (78) boot: ## Label            Usage          Type ST Offset   Length
I (89) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (101) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (112) boot:  2 factory          factory app      00 00 00010000 000f0000
I (124) boot: End of partition table
I (130) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x66018 (417816) map
I (263) esp_image: segment 1: paddr=0x00076030 vaddr=0x3ffe8000 size=0x0075c (  1884) load
I (264) esp_image: segment 2: paddr=0x00076794 vaddr=0x3ffe875c size=0x0011c (   284) load
I (274) esp_image: segment 3: paddr=0x000768b8 vaddr=0x40100000 size=0x074b0 ( 29872) load
0x40100000: _stext at ??:?

I (295) boot: Loaded app from partition at offset 0x10000
I (316) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (326) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
,��I (396) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2
I (406) subpub: Setting WiFi configuration SSID lum_jio_2520...
mode : sta(84:f3:eb:ee:68:78)
add if0
I (416) subpub: AWS IoT SDK Version 2.2.1-
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
pm open phy_2,type:2 0 0
cnt 

connected with lum_jio_2520, channel 11
I (7736) event: sta ip: 192.168.225.88, mask: 255.255.255.0, gw: 192.168.225.1
I (7736) subpub: Connecting to AWS...
mbedtls: ssl_tls.c:6674 => handshake
mbedtls: ssl_cli.c:3336 client state: 0
[... 119 lines cut ...]
mbedtls: ssl_tls.c:2414 in_left: 0, nb_want: 5
mbedtls: ssl_tls.c:2438 in_left: 0, nb_want: 5
mbedtls: ssl_tls.c:2439 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)
mbedtls: ssl_tls.c:2451 <= fetch input
mbedtls: ssl_tls.c:3535 dumping 'input record header' (5 bytes)
mbedtls: ssl_tls.c:3535 0000:  16 03 03 13 93                                   .....
mbedtls: ssl_tls.c:3544 input record: msgtype = 22, version = [3:3], msglen = 5011
mbedtls: ssl_tls.c:3582 bad message length
mbedtls: ssl_tls.c:3793 mbedtls_ssl_read_record_layer() returned -29184 (-0x7200)
mbedtls: ssl_cli.c:1465 mbedtls_ssl_read_record() returned -29184 (-0x7200)
mbedtls: ssl_tls.c:6684 <= handshake
E (9186) esp-tls: mbedtls_ssl_handshake returned -0x7200
I (9196) esp-tls: Certificate verified.
E (9216) esp-tls: Failed to open new connection
E (9216) subpub: Error(-13) connecting to a21vacda74k99l-ats.iot.eu-west-1.amazonaws.com:8883
E (10226) subpub: Error(-15) connecting to a21vacda74k99l-ats.iot.eu-west-1.amazonaws.com:8883
E (11226) subpub: Error(-15) connecting to a21vacda74k99l-ats.iot.eu-west-1.amazonaws.com:8883
E (12226) subpub: Error(-15) connecting to a21vacda74k99l-ats.iot.eu-west-1.amazonaws.com:8883
E (13226) subpub: Error(-15) connecting to a21vacda74k99l-ats.iot.eu-west-1.amazonaws.com:8883
E (14226) subpub: Error(-15) connecting to a21vacda74k99l-ats.iot.eu-west-1.amazonaws.com:8883
E (15226) subpub: Error(-15) connecting to a21vacda74k99l-ats.iot.eu-west-1.amazonaws.com:8883
E (16226) subpub: Error(-15) connecting to a21vacda74k99l-ats.iot.eu-west-1.amazonaws.com:8883

thanks and regrads
pradeep

AnhTuDo1998, 2020-05-24:
Hey pradeep, may I check if you have also tried out the iot-shadow example in the esp-aws-iot package as well? Does it work?

Have you tried updating both the esp-aws-iot and ESP8266-RTOS-SDK to the newest version?

AnhTuDo1998, 2020-05-24:
Please refer to the Troubleshooting segment of the below README.md.

Double check that:
- You have created the necessary Thing, certificate, policy on AWS IoT
- You have attached the correct Thing and iot policy to the certificate.
[https://github.com/espressif/esp-aws-iot/tree/master/examples](url)