espressif/ESP8266_RTOS_SDK #880: something about the IP_EVENT_AP_STAIPASSIGNED

KannaNyaa, 2020-04-24:
i can get in the fork `else if (event_id == IP_EVENT_AP_STAIPASSIGNED){}`
but when i use the `ip4addr_ntoa(event->ip)`,  8266 will reboot until station exit.
this is the event handler:
`    if (event_id == WIFI_EVENT_AP_STACONNECTED) {
        wifi_event_ap_staconnected_t* event = (wifi_event_ap_staconnected_t*) event_data;
        ESP_LOGI(TAG, "station "MACSTR" join, AID=%d",
                 MAC2STR(event->mac), event->aid);
    } else if (event_id == WIFI_EVENT_AP_STADISCONNECTED) {
        wifi_event_ap_stadisconnected_t* event = (wifi_event_ap_stadisconnected_t*) event_data;
        ESP_LOGI(TAG, "station "MACSTR" leave, AID=%d",
                 MAC2STR(event->mac), event->aid);
    }else if (event_id == IP_EVENT_AP_STAIPASSIGNED){
        ip_event_ap_staipassigned_t* event =(ip_event_ap_staipassigned_t*) event_data;
        ESP_LOGI(TAG, "got ip:%s",ip4addr_ntoa(event->ip));
    }`
this is the handler register:
`  esp_event_handler_register(WIFI_EVENT, ESP_EVENT_ANY_ID, &wifi_event_handler, NULL);
    esp_event_handler_register(IP_EVENT, ESP_EVENT_ANY_ID, &wifi_event_handler, NULL);`
i will send all my codes in the files
[softap_example_main.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/4526446/softap_example_main.zip)

KannaNyaa, 2020-04-24:
i forgot showing the errors:

 I (12565) wifi softAP: station 34:e1:2d:a0:34:8a join, AID=1
I (12635) tcpip_adapter: softAP assign IP to station,IP is: 192.168.4.2
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40225cf9  PS      : 0x00000030  A0      : 0x40225dc7  A1      : 0x3ffedfe0  
0x40225cf9: ip4addr_ntoa_r at /home/kanna/esp_ws/ESP8266_RTOS_SDK/components/lwip/lwip/src/core/ipv4/ip4_addr.c:294

0x40225dc7: ip4addr_ntoa at /home/kanna/esp_ws/ESP8266_RTOS_SDK/components/lwip/lwip/src/core/ipv4/ip4_addr.c:271

A2      : 0x00000000  A3      : 0x40106638  A4      : 0x00000010  A5      : 0x00000000  
A6      : 0x40232700  A7      : 0x00000000  A8      : 0x3ffed6a8  A9      : 0x00000090  
0x40232700: wifi_event_handler at /home/kanna/esp_ws/ESP_ws/softAP/main/softap_example_main.c:34

A10     : 0x3ffedee3  A11     : 0x00000031  A12     : 0x401078a4  A13     : 0x4010788c  
A14     : 0x40107680  A15     : 0x40107704  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c  

Backtrace: 0x40225cf9:0x3ffedfe0 0x40225dc7:0x3ffee030 0x4023277a:0x3ffee040 0x4023f729:0x3ffee060 0x4023c2b4:0x3ffee080 0x4023c383:0x3ffee0c0 
0x40225cf9: ip4addr_ntoa_r at /home/kanna/esp_ws/ESP8266_RTOS_SDK/components/lwip/lwip/src/core/ipv4/ip4_addr.c:294

0x40225dc7: ip4addr_ntoa at /home/kanna/esp_ws/ESP8266_RTOS_SDK/components/lwip/lwip/src/core/ipv4/ip4_addr.c:271

0x4023277a: wifi_event_handler at /home/kanna/esp_ws/ESP_ws/softAP/main/softap_example_main.c:45

0x4023f729: handler_execute at /home/kanna/esp_ws/ESP8266_RTOS_SDK/components/esp_event/esp_event.c:160

0x4023c2b4: esp_event_loop_run at /home/kanna/esp_ws/ESP8266_RTOS_SDK/components/esp_event/esp_event.c:646

0x4023c383: esp_event_loop_run_task at /home/kanna/esp_ws/ESP8266_RTOS_SDK/components/esp_event/esp_event.c:646

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 7296, room 16 
tail 0
chksum 0xa3
load 0x3ffe8408, len 24, room 8 
tail 0
chksum 0x2a
load 0x3ffe8420, len 3540, room 8 
tail 12
chksum 0xec
csum 0xec�[0;32mI (237) boot: ESP-IDF v3.2-438-gb29ffce3-dirty 2nd stage bootloader
I (237) boot: compile time 10:47:39
I (246) qio_mode: Enabling default flash chip QIO
I (246) boot: SPI Speed      : 40MHz
I (253) boot: SPI Mode       : QIO
I (259) boot: SPI Flash Size : 2MB
I (265) boot: Partition Table:
I (271) boot: ## Label            Usage          Type ST Offset   Length
I (283) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (294) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (306) boot:  2 factory          factory app      00 00 00010000 000f0000
I (317) boot: End of partition table
I (324) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x3db90 (252816) map
0x40210010: _flash_cache_start at ??:?

I (422) esp_image: segment 1: paddr=0x0004dba8 vaddr=0x4024dba0 size=0x0bf68 ( 49000) map
I (439) esp_image: segment 2: paddr=0x00059b18 vaddr=0x3ffe8000 size=0x00640 (  1600) load
I (440) esp_image: segment 3: paddr=0x0005a160 vaddr=0x40100000 size=0x008e0 (  2272) load
I (450) esp_image: segment 4: paddr=0x0005aa48 vaddr=0x401008e0 size=0x0584c ( 22604) load
I (471) boot: Loaded app from partition at offset 0x10000
I (492) system_api: Base MAC address is not set, read default base MAC address from EFUSE
I (497) system_api: Base MAC address is not set, read default base MAC address from EFUSE
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40105398  PS      : 0x00000033  A0      : 0x40100478  A1      : 0x3ffea8e0  
0x40105398: wDev_ProcessFiq at ??:?

0x40100478: nmi_reentried at /home/kanna/esp_ws/ESP8266_RTOS_SDK/components/freertos/port/esp8266/xtensa_vectors.S:862

A2      : 0x3ffee1e0  A3      : 0x00c52ccc  A4      : 0x00000030  A5      : 0x4024f599  
A6      : 0x00000100  A7      : 0x00000015  A8      : 0x3ffeb979  A9      : 0x402553c6  
A10     : 0x0000001e  A11     : 0x000001e0  A12     : 0x3ffea384  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x40106e90  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c  

Backtrace: 0x40105398:0x3ffea8e0 0x40100478:0x3ffea950 0x4010624c:0x3ffea9a0 0x40100478:0x3ffea9a0 
0x40105398: wDev_ProcessFiq at ??:?

0x40100478: nmi_reentried at /home/kanna/esp_ws/ESP8266_RTOS_SDK/components/freertos/port/esp8266/xtensa_vectors.S:862

0x40100478: nmi_reentried at /home/kanna/esp_ws/ESP8266_RTOS_SDK/components/freertos/port/esp8266/xtensa_vectors.S:862

KannaNyaa, 2020-04-24:
i find out which code is wrong:
in`event_send_compat.c` 94 lines:
`HANDLE_SYS_EVENT(IP, AP_STAIPASSIGNED);`
However this kind of event has its returnable data,so we should modify it to:
`HANDLE_SYS_EVENT_ARG(IP, AP_STAIPASSIGNED, ap_staipassigned);`
Then this bug was solved.

mriksman, 2020-04-25:
You're mixing up the old and new event system.

This is new;
```
static void wifi_event_handler(void* arg, esp_event_base_t event_base,
                                    int32_t event_id, void* event_data)
```
These are old;
```
        wifi_event_ap_staconnected_t* event = (wifi_event_ap_staconnected_t*) event_data;
        wifi_event_ap_stadisconnected_t* event = (wifi_event_ap_stadisconnected_t*) event_data;
        ip_event_ap_staipassigned_t* event =(ip_event_ap_staipassigned_t*) event_data;
```

Look at example https://github.com/espressif/ESP8266_RTOS_SDK/blob/master/examples/wifi/getting_started/station/main/station_example_main.c

Use this;
```
    } else if (event_base == IP_EVENT && event_id == IP_EVENT_STA_GOT_IP) {
        ip_event_got_ip_t* event = (ip_event_got_ip_t*) event_data;
        ESP_LOGI(TAG, "got ip:%s",
                 ip4addr_ntoa(&event->ip_info.ip));
```

KannaNyaa, 2020-04-25:
ok. I'll have a try. I thought it just can be used in station mode and i wanted the ap to get its stations' ip addresses

 

 
 
     } else if (event_base == IP_EVENT &amp;&amp; event_id == IP_EVENT_STA_GOT_IP) {         ip_event_got_ip_t* event = (ip_event_got_ip_t*) event_data;         ESP_LOGI(TAG, "got ip:%s",                  ip4addr_ntoa(&amp;event-&gt;ip_info.ip));

mriksman, 2020-04-25:
Ahh. I understand what you are trying to do now. Get the IP address of a STA connecting to the ESP acting as a soft-AP.
I see function `tcpip_adapter_dhcps_cb` in `tcpip_adapter_lwip.c` sends the event with the client IP address in `event_info` via `esp_event_send(&evt)`, but the mapping between the legacy and new event system doesn't map the `event_info` - it's missing the ARGS `HANDLE_SYS_EVENT(IP, AP_STAIPASSIGNED)`
They have it working properly in ESP-IDF.

Perhaps fire up a legacy event handler as well? You'll need it if you are using mDNS anyway;
```
/* Legacy event loop still required for the old (ESP8266 RTOS SDK) mDNS event loop dispatch */
esp_err_t legacy_event_handler(void *ctx, system_event_t *event) {
    mdns_handle_system_event(ctx, event);
    return ESP_OK;
}
```
Otherwise, your Pull Request is a valid idea.

KannaNyaa, 2020-04-26:
Thx I got it. And i am still learning the examples and APIs. （在guyhub上习惯了用英文，感觉邮件上可以用中文。。。。勉强过6级的我，看还是看得懂，就是表述起来不是很清晰，留下了英语极差的泪QAQ）

 

 I see function tcpip_adapter_dhcps_cb in tcpip_adapter_lwip.c sends the event with the client IP address in event_info via esp_event_send(&amp;evt), but the mapping between the legacy and new event system doesn't map the event_info - it's missing the ARGS HANDLE_SYS_EVENT(IP, AP_STAIPASSIGNED)

MrJake222, 2021-07-08:
> i find out which code is wrong:
> in`event_send_compat.c` 94 lines:
> `HANDLE_SYS_EVENT(IP, AP_STAIPASSIGNED);`
> However this kind of event has its returnable data,so we should modify it to:
> `HANDLE_SYS_EVENT_ARG(IP, AP_STAIPASSIGNED, ap_staipassigned);`
> Then this bug was solved.

Can confirm, this fixed the issue for me.

hadif66, 2022-12-11:
You should do this inside WiFi_event_handler:
else if(event_base == IP_EVENT && event_id == IP_EVENT_AP_STAIPASSIGNED){
        ip_event_ap_staipassigned_t* event = (ip_event_ap_staipassigned_t*) event_data;
        ESP_LOGI(TAG, "New station IP:" IPSTR, IP2STR(&event->ip));                                        
    }

before that inside wifi_init you should register that event like this:

    esp_event_handler_instance_t ap_staipassigned_ip;
    ESP_ERROR_CHECK(esp_event_handler_instance_register(IP_EVENT, IP_EVENT_AP_STAIPASSIGNED,&wifi_event_handler,NULL,&ap_staipassigned_ip));  

 The most important part is using "ip_event_ap_staipassigned_t" as the variable

KannaNyaa, 2022-12-11:
> You should do this inside WiFi_event_handler:
>
> ```
>      else if(event_base == IP_EVENT && event_id == IP_EVENT_AP_STAIPASSIGNED){
>           ip_event_ap_staipassigned_t* event = [...]

In former release, the erro was cause by `inevent_send_compat.c`.
In the 94 lines: `HANDLE_SYS_EVENT(IP, AP_STAIPASSIGNED);` should be modified to `HANDLE_SYS_EVENT_ARG(IP, AP_STAIPASSIGNED, ap_staipassigned);`. The latest release sovle these problem.
I just forget close this issue, sry.

closed 2022-12-11 by KannaNyaa