espressif/ESP8266_RTOS_SDK #882: Error compiling "simple_ota_example" with 1MB flash size

decelect-gchauvelin, 2020-04-24:
## Environment

- Development Kit: ESP01
- IDF version:  b29ffce3
- Development Env:      Make
- Operating System:     Ubuntu
- Power Supply:         USB

## Problem Description

I try to compile example "example/system/ota/simple_ota_example" to test it before include some part in my project but I get error : 
section '.flash.text' will not fit in region 'iram0_2_seg'
section '.flash.rodata' will not fit in region 'iram0_2_seg'

I only modified /Serial flasher config/Flash size to 1MB after download latest version of SDK

Could you please tell me how to compile this example please (for 1MB memory)?

### Expected Behavior

### Actual Behavior

### Steps to repropduce

1. Download latest version SDK
2. Modify Flash size under menuconfig of simple_ota_example
3. make flash monitor

mriksman, 2020-04-25:
Total image size for this example is 582KB. So this is not going to work with a 1MB flash for OTA, as OTA will need to divide your total flash (1MB) by 2.

That is one of the problems with ESP8266 RTOS SDK becoming more like ESP-IDF; the bin sizes are getting larger and will be less likely to fit on a 1MB flash.

What I've done is create a 384KB and a 576KB partition, and created my own OTA routine. First you boot to OTA_0 which allows to to load your main program into OTA_1. On the next restart, OTA_1 boots.

app0 | app | ota_0 | 0x10000 | 576K
-- | -- | -- | -- | --
app1 | app | ota_1 | 0xA0000 | 384K

Or better still. Go buy an ESP32... Or an ESP8266 with more flash.

decelect-gchauvelin, 2020-04-27:
Ok, thank you for your help.

closed 2020-04-27 by decelect-gchauvelin