espressif/ESP8266_RTOS_SDK #883: 使用 LEDC 例子时候，发现无法 `LEDC_TIMER_0` 的通道1的占空比无法为0 。

xuhongv, 2020-04-28:
## Environment

- Development Kit:
- IDF version : b29ffce3
- Development Env:   Make
- Operating System:   Ubuntu
- Power Supply:         USB

## Problem Description

使用 LEDC 例子时候，发现无法 `LEDC_TIMER_0` 的通道1的占空比无法为0 。

### Steps to repropduce

1. 下载
2. GPIO13 接 LED0 的正极 ，GPIO14 接 LED1 的正极 ， LED0 和 LED1 公地；
3. 虽然我设置为 占空比为0 ， 发现 LED1 不会熄灭；

### Code to reproduce this issue

```cpp
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/ledc.h"
#include "esp_err.h"

#define LEDC_HS_TIMER          LEDC_TIMER_0
#define LEDC_HS_MODE           LEDC_HIGH_SPEED_MODE
#define LEDC_HS_CH0_GPIO       (13)
#define LEDC_HS_CH0_CHANNEL    LEDC_CHANNEL_0
#define LEDC_HS_CH1_GPIO       (14)
#define LEDC_HS_CH1_CHANNEL    LEDC_CHANNEL_1

#define LEDC_LS_TIMER          LEDC_TIMER_1
#define LEDC_LS_MODE           LEDC_HIGH_SPEED_MODE
#define LEDC_LS_CH2_GPIO       (4)
#define LEDC_LS_CH2_CHANNEL    LEDC_CHANNEL_2
#define LEDC_LS_CH3_GPIO       (15)
#define LEDC_LS_CH3_CHANNEL    LEDC_CHANNEL_3

#define LEDC_TEST_CH_NUM       (2)
#define LEDC_TEST_DUTY         (8192)
#define LEDC_TEST_FADE_TIME    (2000)

void app_main()
{
    int ch;
    ledc_timer_config_t ledc_timer = {
        .duty_resolution = LEDC_TIMER_13_BIT, // resolution of PWM duty
        .freq_hz = 5000,                      // frequency of PWM signal
        .speed_mode = LEDC_HS_MODE,           // timer mode
        .timer_num = LEDC_HS_TIMER            // timer index
    };
    // Set configuration of timer0 for high speed channels
    ledc_timer_config(&ledc_timer);

    ledc_channel_config_t ledc_channel[LEDC_TEST_CH_NUM] = {
        {
            .channel    = LEDC_HS_CH0_CHANNEL,
            .duty       = 0,
            .gpio_num   = LEDC_HS_CH0_GPIO,
            .speed_mode = LEDC_HS_MODE,
            .hpoint     = 0,
            .timer_sel  = LEDC_HS_TIMER
        },
        {
            .channel    = LEDC_HS_CH1_CHANNEL,
            .duty       = 0,
            .gpio_num   = LEDC_HS_CH1_GPIO,
            .speed_mode = LEDC_HS_MODE,
            .hpoint     = 0,
            .timer_sel  = LEDC_HS_TIMER
        },
        {
            .channel    = LEDC_LS_CH2_CHANNEL,
            .duty       = 0,
            .gpio_num   = LEDC_LS_CH2_GPIO,
            .speed_mode = LEDC_LS_MODE,
            .hpoint     = 0,
            .timer_sel  = LEDC_LS_TIMER
        },
        {
            .channel    = LEDC_LS_CH3_CHANNEL,
            .duty       = 0,
            .gpio_num   = LEDC_LS_CH3_GPIO,
            .speed_mode = LEDC_LS_MODE,
            .hpoint     = 0,
            .timer_sel  = LEDC_LS_TIMER
        },
    };

    // Set LED Controller with previously prepared configuration
    for (ch = 0; ch < LEDC_TEST_CH_NUM; ch++) {
        ledc_channel_config(&ledc_channel[ch]);
    }

    // Initialize fade service.
    ledc_fade_func_install(0);

   // while (1)
     {
        printf("1. LEDC fade up to duty = %d\n", LEDC_TEST_DUTY);
        for (ch = 0; ch < LEDC_TEST_CH_NUM; ch++) {
            ledc_set_fade_with_time(ledc_channel[ch].speed_mode,  ledc_channel[ch].channel, LEDC_TEST_DUTY, LEDC_TEST_FADE_TIME);
            ledc_fade_start(ledc_channel[ch].speed_mode,  ledc_channel[ch].channel, LEDC_FADE_NO_WAIT);
        }
        vTaskDelay(LEDC_TEST_FADE_TIME / portTICK_PERIOD_MS);

        printf("2. LEDC fade down to duty = 0\n");
        for (ch = 0; ch < LEDC_TEST_CH_NUM; ch++) {
            ledc_set_fade_with_time(ledc_channel[ch].speed_mode,ledc_channel[ch].channel, 0, LEDC_TEST_FADE_TIME);
            ledc_fade_start(ledc_channel[ch].speed_mode,    ledc_channel[ch].channel, LEDC_FADE_NO_WAIT);
        }
        vTaskDelay(LEDC_TEST_FADE_TIME / portTICK_PERIOD_MS);
    }
}

```

## Other items if possible

- [ ] 还麻烦复现下！

xuhongv, 2020-04-28:
- 神奇的是：试下看看日志，我在  `esp8266/driver/ledc.c`  的 方法 `ledc_fade_down() `加上  `ESP_LOGI(LEDC_TAG,"duty"); ` 竟然可以全灭了；

- 身为一个程序员，我也不信有这样的 bug 存在！

ChenZhengwei20190741, 2020-04-28:
好的，谢谢您提供的反馈，复现了发现了这个问题，初步排查可能是在 pwm 在最后一次设置的时候出现问题，正在积极解决。修复后会给您一个回复。

xuhongv, 2020-04-29:
- 哈哈。

ChenZhengwei20190741, 2020-05-07:
@xuhongv 
请在此处加一下pwm_start(),
```
static void ledc_task(void* pvParameters)
{
    ledc_channel_t channel;
    uint8_t i;
    uint8_t flag[LEDC_CHANNEL_MAX] = { 0 };

    while (1) {

        while (pdTRUE == xQueueReceive(channel_queue, &channel, 0)) {
            flag[channel] = LEDC_FLAG_ON;
        }
        vTaskSuspendAll();
        for (i = 0; i < ledc_usr_channel_max; i++) {
            if (flag[i] == LEDC_FLAG_ON) {
                if (p_ledc_obj[i]->duty_p < p_ledc_obj[i]->duty) {
                    ledc_fade_up(i, &flag[i]);
                } else {
                    ledc_fade_down(i, &flag[i]);
                }
            }
        }
        pwm_start();/*!< test */
        xTaskResumeAll();
        vTaskDelay(LEDC_STEP_TIME / portTICK_PERIOD_MS);
    }
    vTaskDelete(NULL);
}
```
ledc_fade_down 和 ledc_fade_up 中的`pwm_start();`注释掉。

ChenZhengwei20190741, 2020-05-07:
如下代码，在此处加入pwm_start();
```c
static void ledc_task(void* pvParameters)
{
    ledc_channel_t channel;
    uint8_t i;
    uint8_t flag[LEDC_CHANNEL_MAX] = { 0 };

    while (1) {

        while (pdTRUE == xQueueReceive(channel_queue, &channel, 0)) {
            flag[channel] = LEDC_FLAG_ON;
        }
        vTaskSuspendAll();
        for (i = 0; i < ledc_usr_channel_max; i++) {
            if (flag[i] == LEDC_FLAG_ON) {
                if (p_ledc_obj[i]->duty_p < p_ledc_obj[i]->duty) {
                    ledc_fade_up(i, &flag[i]);
                } else {
                    ledc_fade_down(i, &flag[i]);
                }
            }
        }
        xTaskResumeAll();
        pwm_start(); /*!< ----------------- test  -------------- */
        vTaskDelay(LEDC_STEP_TIME / portTICK_PERIOD_MS);
    }
    vTaskDelete(NULL);
}
```
在`ledc_fade_up()` ,`ledc_fade_down(),`中的 `pwm_start();`屏蔽掉。后续会更新到github