espressif/ESP8266_RTOS_SDK #884: OTA_Esp8266 using http

sowmimalu, 2020-04-28:
Hi,
i am using esp8266 development board...i am trying  to download a file from local server by using http..
    In my application i need to update ota in esp8266 by using http.... 
I got an  example in arduino (webupdater) when i tried with this example web page has opened and i am able to choose a bin file which i want ,from that i can able to update the bin file by OTA. 

Now i need to download a file in my source code by using http ...i tried basic example in ardunio (httpupdate) but its not work ...i got connection refused error ....when I tried to download a bin file from local server.

can you please suggest me a example ??

27-301, 2020-04-28:
Hi, if you use ESP8266_RTOS_SDK directly, you can refer to ESP8266_RTOS_SDK/examples/system/ota/simple_ota_example.

sowmimalu, 2020-04-28:
i am using arduino for esp8266
For this i am using following example 
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266httpUpdate/examples/httpUpdate/httpUpdate.ino

it is always showing error, tried with system IP as below command 
http://ip/new/hello_world.bin

sowmimalu, 2020-04-30:
can you please tell me on  how to do this..??

mriksman, 2020-05-03:
You're in the wrong project. Arduino ESP8266 is _totally_ different than ESP8266 RTOS SDK.

Try logging an issue here; https://github.com/esp8266/Arduino/issues. Or better still, because it's likely not a bug and you need help, try the forums like; https://www.esp8266.com/ or https://forum.arduino.cc/.

Please close this issue, and good luck!