espressif/ESP8266_RTOS_SDK #885: ERROR while compiling the subscribe_publish in ESP8266_RTOS_SDK v3.0

jkjayakumar, 2020-04-28:
Hi All
    When i run the subscribe_publish program . i made some modifications. In the publishing part I'm getting ERROR.

After flashing the example to your ESP32, it should connect to Amazon and start subscribing/publishing MQTT data.

The example code publishes MQTT data to the topic `test_topic/esp32`. Amazon provides a web interface to subscribe to MQTT topics for testing:

* On the AWS IoT console, click "MQTT Client" near the top-right.
* Click "Generate Client ID" to generate a random client ID.
* Click "Connect"

One connection succeeds, you can subscribe to the data published by the ESP32:

* Click "Subscribe to Topic"
* Enter "Subscription Topic" `test_topic/esp32`
* Click "Subscribe"

... you should see MQTT data published from the running example.

To publish data back to the device:

* Click "Publish to Topic"
* Enter "Publish Topic" `test_topic/esp32`
* Enter a message in the payload field
* Click Publish

ERROR:
    Toolchain path: /home/jayakumar/esp/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc
Toolchain version: crosstool-ng-1.22.0-92-g8facf4c
Compiler version: 5.2.0
Python requirements from /home/jayakumar/esp/ESP8266_RTOS_SDK/requirements.txt are satisfied.
MONITOR
--- idf_monitor on /dev/ttyUSB0 74880 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)
load 0x40100000, len 7204, room 16
tail 4
chksum 0x42
load 0x3ffe8408, len 24, room 4
tail 4
chksum 0xcb
load 0x3ffe8420, len 3548, room 4
tail 8
chksum 0xce
I (44) boot: ESP-IDF v3.2-59-g86808496-dirty 2nd stage bootloader
I (44) boot: compile time 18:20:41
I (45) qio_mode: Enabling default flash chip QIO
I (53) boot: SPI Speed      : 40MHz
I (59) boot: SPI Mode       : QIO
I (66) boot: SPI Flash Size : 2MB
I (72) boot: Partition Table:
I (77) boot: ## Label            Usage          Type ST Offset   Length
I (89) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (100) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (112) boot:  2 factory          factory app      00 00 00010000 000f0000
I (123) boot: End of partition table
I (130) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x391fc (233980) map
0x40210010: _flash_cache_start at ??:?

I (221) esp_image: segment 1: paddr=0x00049214 vaddr=0x4024920c size=0x06c34 ( 27700) map
I (231) esp_image: segment 2: paddr=0x0004fe50 vaddr=0x3ffe8000 size=0x00578 (  1400) load
I (233) esp_image: segment 3: paddr=0x000503d0 vaddr=0x40100000 size=0x00a30 (  2608) load
I (247) esp_image: segment 4: paddr=0x00050e08 vaddr=0x40100a30 size=0x04e50 ( 20048) load
I (266) boot: Loaded app from partition at offset 0x10000
I (315) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (325) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (495) phy_init: phy ver: 1055_12
I (495) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2
I (515) subpub: Setting WiFi configuration SSID Kre38labs...
E (2925) subpub: Disconnect reason : 201
E (5335) subpub: Disconnect reason : 201
E (7745) subpub: Disconnect reason : 201
E (10155) subpub: Disconnect reason : 201
E (12565) subpub: Disconnect reason : 201
E (14975) subpub: Disconnect reason : 201
E (17385) subpub: Disconnect reason : 201
I (27695) event: sta ip: 192.168.43.217, mask: 255.255.255.0, gw: 192.168.43.1

Thanks and regrads
Jayakumar

AnhTuDo1998, 2020-05-24:
Hi please attach a build log (compile log).

This looks like an output from make monitor ?

Have you tried checking the configuration for AWS certificate, thing name, policy, etc based on the examples README.md ?

AnhTuDo1998, 2020-05-25:
Hi OP, can you use make menuconfig > Component Config > Log output > Select Verbose or Debug mode and run make, make flash again?

Please copy the output of make monitor thereafter? This is to help others to see the hidden log level.

Thank you