espressif/ESP8266_RTOS_SDK #886: mbedtls_ssl_handshake returned -0x2700

clibing, 2020-04-30:
## Environment

* nodemcu esp8266 
* esp-mqtt
* esp8266_rtos_sdk 3.2
* mqtt broker is emqx 
* use mqtts

## Problem Description

````
D (124256) MQTT_CLIENT: Reconnecting...
D (129256) esp-tls: host:149.129.44.205: strlen 14
D (129576) esp-tls: handshake in progress...
mbedtls: ssl_tls.c:6674 => handshake
mbedtls: ssl_cli.c:3336 client state: 0
mbedtls: ssl_tls.c:2464 => flush output
mbedtls: ssl_tls.c:2476 <= flush output
mbedtls: ssl_cli.c:3336 client state: 1
mbedtls: ssl_tls.c:2464 => flush output
mbedtls: ssl_tls.c:2476 <= flush output
mbedtls: ssl_cli.c:730 => write client hello
mbedtls: ssl_cli.c:768 client hello, max version: [3:3]
mbedtls: ssl_cli.c:777 dumping 'client hello, random bytes' (32 bytes)
mbedtls: ssl_cli.c:777 0000:  ee e9 9b ad be e6 4e ec 27 6d f6 2a 47 58 6b b9  ......N.'m.*GXk.
mbedtls: ssl_cli.c:777 0010:  30 14 7e 9d 34 6a f0 c6 a9 30 23 26 b2 b0 a2 fc  0.~.4j...0#&....
mbedtls: ssl_cli.c:830 client hello, session id len.: 0
mbedtls: ssl_cli.c:831 dumping 'client hello, session id' (0 bytes)
mbedtls: ssl_cli.c:898 client hello, add ciphersuite: 003d
mbedtls: ssl_cli.c:898 client hello, add ciphersuite: 0035
mbedtls: ssl_cli.c:898 client hello, add ciphersuite: 003c
mbedtls: ssl_cli.c:898 client hello, add ciphersuite: 002f
mbedtls: ssl_cli.c:905 client hello, got 4 ciphersuites (excluding SCSVs)
mbedtls: ssl_cli.c:914 adding EMPTY_RENEGOTIATION_INFO_SCSV
mbedtls: ssl_cli.c:963 client hello, compress len.: 1
mbedtls: ssl_cli.c:965 client hello, compress alg.: 0
mbedtls: ssl_cli.c:72 client hello, adding server name extension: 149.129.44.205
mbedtls: ssl_cli.c:189 client hello, adding signature_algorithms extension
mbedtls: ssl_cli.c:555 client hello, adding extended_master_secret extension
mbedtls: ssl_cli.c:1039 client hello, total extension length: 43
mbedtls: ssl_tls.c:2749 => write record
mbedtls: ssl_tls.c:2895 output record: msgtype = 22, version = [3:1], msglen = 98
mbedtls: ssl_tls.c:2898 dumping 'output record sent to network' (103 bytes)
mbedtls: ssl_tls.c:2898 0000:  16 03 01 00 62 01 00 00 5e 03 03 ee e9 9b ad be  ....b...^.......
mbedtls: ssl_tls.c:2898 0010:  e6 4e ec 27 6d f6 2a 47 58 6b b9 30 14 7e 9d 34  .N.'m.*GXk.0.~.4
mbedtls: ssl_tls.c:2898 0020:  6a f0 c6 a9 30 23 26 b2 b0 a2 fc 00 00 0a 00 3d  j...0#&........=
mbedtls: ssl_tls.c:2898 0030:  00 35 00 3c 00 2f 00 ff 01 00 00 2b 00 00 00 13  .5.<./.....+....
mbedtls: ssl_tls.c:2898 0040:  00 11 00 00 0e 31 34 39 2e 31 32 39 2e 34 34 2e  .....149.129.44.
mbedtls: ssl_tls.c:2898 0050:  32 30 35 00 0d 00 0c 00 0a 06 01 05 01 04 01 03  205.............
mbedtls: ssl_tls.c:2898 0060:  01 02 01 00 17 00 00                             .......
mbedtls: ssl_tls.c:2464 => flush output
mbedtls: ssl_tls.c:2483 message length: 103, out_left: 103
mbedtls: ssl_tls.c:2489 ssl->f_send() returned 103 (-0xffffff99)
mbedtls: ssl_tls.c:2508 <= flush output
mbedtls: ssl_tls.c:2907 <= write record
mbedtls: ssl_cli.c:1065 <= write client hello
mbedtls: ssl_cli.c:3336 client state: 2
mbedtls: ssl_tls.c:2464 => flush output
mbedtls: ssl_tls.c:2476 <= flush output
mbedtls: ssl_cli.c:1458 => parse server hello
mbedtls: ssl_tls.c:3785 => read record
mbedtls: ssl_tls.c:2256 => fetch input
mbedtls: ssl_tls.c:2414 in_left: 0, nb_want: 5
mbedtls: ssl_tls.c:2438 in_left: 0, nb_want: 5
mbedtls: ssl_tls.c:2439 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)
mbedtls: ssl_tls.c:2451 <= fetch input
mbedtls: ssl_tls.c:3535 dumping 'input record header' (5 bytes)
mbedtls: ssl_tls.c:3535 0000:  16 03 03 00 51                                   ....Q
mbedtls: ssl_tls.c:3544 input record: msgtype = 22, version = [3:3], msglen = 81
mbedtls: ssl_tls.c:2256 => fetch input
[... 142 lines cut ...]
mbedtls: ssl_tls.c:2508 <= flush output
mbedtls: ssl_tls.c:2907 <= write record
mbedtls: ssl_tls.c:4201 <= send alert message
mbedtls: ssl_tls.c:4739 ! Certificate verification flags 4
mbedtls: ssl_tls.c:4748 <= parse certificate
mbedtls: ssl_tls.c:6684 <= handshake
E (131856) esp-tls: mbedtls_ssl_handshake returned -0x2700
I (131866) esp-tls: Failed to verify peer certificate!
I (131876) esp-tls: verification info:   ! The certificate Common Name (CN) does not match with the expected CN

E (131926) esp-tls: Failed to open new connection
E (131926) TRANS_SSL: Failed to open a new connection
E (131926) MQTT_CLIENT: Error transport connect
I (131936) MQTT_CLIENT: Reconnect after 10000 ms
I (131936) fish_tank: MQTT_EVENT_DISCONNECTED
````

### Code to reproduce this issue

use demo only replace pem

### use mqtt.fx client with pem
[image]
[image]
[image]

clibing, 2020-04-30:
I have solved it myself. The reason is the problem of CN

closed 2020-04-30 by clibing

heidariarash, 2020-04-30:
Hey @clibing. I encountered the same issue. How did you fix it? I don't know how to fix CN!
Thanks.

clibing, 2020-04-30:
When openssl generates the root ca, it does not specify CN, just specify /C/ST/L/O/OU, and specify /CN when signing, which is also the link when you access MQTT Broker

For example, /CN=clibing.com, MQTT access when the URL is mqtts://clibing.com:port

example: 

```
docker run -it --name centos7 -v `pwd`/conf:/conf centos:centos7 /bin/sh

cd /conf/

cp /etc/pki/tls/openssl.cnf ./

rm -rf /etc/pki/CA/*.old

touch /etc/pki/CA/index.txt

echo 01 > /etc/pki/CA/serial

rm -rf certs;mkdir certs
```

### Run CA env
```
docker run -it --name centos7 -v `pwd`/conf:/conf centos:centos7 /bin/sh

cd /conf/

cp /etc/pki/tls/openssl.cnf ./

rm -rf /etc/pki/CA/*.old

touch /etc/pki/CA/index.txt

echo 01 > /etc/pki/CA/serial

rm -rf certs;mkdir certs
```

### create base ca

```
openssl genrsa -out certs/root-ca.key 2048

openssl req -new -x509 -days 36500 -config ./openssl.cnf -key certs/root-ca.key -out certs/root-cacert.pem -subj "/C=CN/ST=Beijing/O=clibing/OU=clibing.com"

```

/C=CN/ST=Beijing/O=clibing/OU=`clibing.com`, the `OU` not `CN`

### check ca 

```
openssl x509 -in certs/root-cacert.pem -noout -text
```

### create server pem

```
openssl genrsa -out certs/server.key 2048

openssl req -new -days 3650 -key certs/server.key -out certs/server-cert.csr -subj "/C=CN/ST=Beijing/O=clibing/OU=clibing.com/CN=clibing.com"

openssl ca -config ./openssl.cnf -extensions v3_req -days 3650 -in certs/server-cert.csr -out certs/server-cert.pem -cert certs/root-cacert.pem -keyfile certs/root-ca.key
```

### verfiy

```
openssl verify -CAfile certs/root-cacert.pem certs/server-cert.pem
```

copy server-cert.pem server-key.

### emqx config 

```
listener.ssl.external.keyfile = etc/certs/server.key
listener.ssl.external.certfile = etc/certs/server-cert.pem 

```

### last replace ESP8266_RTOS_SDK/example/.../ssl demo pem is test ok

clibing, 2020-05-03:
> 嘿@clibing。我遇到了同样的问题。你怎么修好它的？我不知道如何修复CN！
> 谢谢。

> Hey @clibing. I encountered the same issue. How did you fix it? I don't know how to fix CN!
> Thanks.

View my detailed steps