espressif/ESP8266_RTOS_SDK #892: Compiler warning from option -Wno-frame-address

withattribution, 2020-05-09:
## Environment

- Development Kit: N/A
- IDF version: ee0c4459
- Compiler version: xtensa-lx106-elf-gcc (crosstool-NG crosstool-ng-1.22.0-100-ge567ec7) 5.2.0
- Development Env: Make
- Operating System: MacOS
- Power Supply: N/A

## Problem Description

When compiling any app if there are any errors or warnings (such as an unused variable) I also see the added warning: `cc1: warning: unrecognized command line option '-Wno-frame-address'`
If there are no warnings then the compilation output is clean. 

I have not found any reference to the `-Wno-frame-address` flag on either
[gcc Warning Options](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html)
or 
[gcc Xtensa Options](https://gcc.gnu.org/onlinedocs/gcc-5.2.0/gcc/Xtensa-Options.html#Xtensa-Options)

There are references to `-Wframe-larger-than=byte-size` and `-Wframe-address` which I'm wondering could have been intended? Otherwise removing the offending flag from the make/project.mk `COMMON_FLAGS` does remove the extra noise from the output log, but since I cannot find a reference to what that flag is used for I'm not sure if there would be unintended side effects etc.

### Expected Behavior

Application compilation output should only show warnings that are relevant to the app level code, not warnings being raised by potentially unsupported compiler flags.

### Actual Behavior

Each time time an error or warning is generated, an un-related unsupported compiler flag warning is logged as well.

### Steps to reproduce

1. navigate to installed ESP8266_RTOS_SDK directory
2. edit examples/get-started/hello_world/hello_world_main.c
3. Initialize any variable such as `int unused_int;`
4. make

### Code to reproduce this issue

in hello_world_main.c add any unused variable

```cpp

#include "esp_log.h"

void app_main()
{
    int unused_int;
}

```

## Debug Logs

for each compiled file with any warning or error the output I see is:

```
/ESP8266_RTOS_SDK/examples/get-started/hello_world/main/hello_world_main.c: In function 'app_main':
/ESP8266_RTOS_SDK/examples/get-started/hello_world/main/hello_world_main.c:19:9: warning: unused variable 'unused_int' [-Wunused-variable]
     int unused_int;
         ^
/ESP8266_RTOS_SDK/examples/get-started/hello_world/main/hello_world_main.c: At top level:
cc1: warning: unrecognized command line option '-Wno-frame-address'
```

dsptech, 2020-05-17:
Hi,
I already noticed that. It is mentioned here:
https://gcc.gnu.org/onlinedocs/gcc-8.2.0/gcc/Option-Index.html
(and in v8.5.0) but the related description is missing.

The setting is not mentioned in the v5.2 compiler instead. 

Anyway, the v8.2 is the current compiler of the ESP32 and the actual ESP8266 RTOS framework is a migration from the ESP32 framework
So, I suspect that some not required ESP32 setting is still here.
Regards.

rustyx, 2021-07-08:
The flag is still there in [`make/project.mk`](https://github.com/espressif/ESP8266_RTOS_SDK/blob/master/make/project.mk#L387):

```
COMMON_FLAGS = \
	-Wno-frame-address \
```

It's quite annoying, prints a warning on each source file.