espressif/ESP8266_RTOS_SDK #895: ESP8266_RTOS_SDK Pre-release v3.3-rc1 monitor不支持默认波特率74880

XiaoSenLuo, 2020-05-13:
**按照doc设置的, 环境是Windows 10 64BIT WSL Ubuntu, 终端是Windows terminal, 编译下载没有问题, 经过测试, 开发板是能正常输出启动信息的, 但是启用IDF monitor显示乱码, menuconfig关于monitor波特率设置是74880. 输出如下:**
```
Leaving...
Hard resetting via RTS pin...
MONITOR
--- idf_monitor on /dev/ttyS3 74880 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/ubuntu/esp/esp8266/ESP8266_RTOS_SDK/tools/idf_monitor.py", line 136, in _run_outer
    self.run()
  File "/home/ubuntu/esp/esp8266/ESP8266_RTOS_SDK/tools/idf_monitor.py", line 225, in run
    self.serial.open()
  File "/home/ubuntu/.local/lib/python2.7/site-packages/serial/serialposix.py", line 272, in open
    self._reconfigure_port(force_update=True)
  File "/home/ubuntu/.local/lib/python2.7/site-packages/serial/serialposix.py", line 442, in _reconfigure_port
    self._set_special_baudrate(custom_baud)
  File "/home/ubuntu/.local/lib/python2.7/site-packages/serial/serialposix.py", line 130, in _set_special_baudrate
    raise ValueError('Failed to set custom baud rate ({}): {}'.format(baudrate, e))
ValueError: Failed to set custom baud rate (74880): [Errno 25] Inappropriate ioctl for device
```

nsfilho, 2020-05-22:
Are you using a valid `port` in windows? I'm asking that, because we see /dev/ttyS3, this is valid for linux, unix, MacOS, but I don't know if Windows support this kind of name. I'm not use windows at least 10 years or more. But in my "memory", windows named the COM in another way.

XiaoSenLuo, 2020-05-29:
> > Are you using a valid `port` in windows? I'm asking that, because we see /dev/ttyS3, this is valid for linux, unix, MacOS, but I don't know if Windows support this kind of name. I'm not use windows at least 10 years or more. But in my "memory", windows named the COM in another way.
> 
> The serial port "ttyS3" is legal，and
> the Port is free, because I use WSL system, in Windows, the port is "COM3" , in wsl is "ttyS3", if in windows is "COM2", then in wsl is "ttyS2"

Now, I change the boadrate to 115200, and it word done.

pigno5e, 2020-07-12:
Hi, I just had the same issue. For ESP8266 if baud is changed to 115200, you won't be able to read content send in ASCII format in a serial port monitor. Could anyone help with this? Thanks.

nsfilho, 2020-07-18:
You can disable the formatting  message in ascii mode in `make menuconfig`. But you can use properly only changing your "windows shell", as `windows cygwin`. This can understanding  properly ascii colors chars, and show you like a charm.

But if you wouldn't  change to cygwin, you can disable this `pretty formatting` to see only clear text messages.

Masterjeef117, 2022-12-27:
I'm running into this same issue while trying to monitor the `hello_world` example.  After changing the monitor baud rate to 115200 I encounter the same problem as pigno5e.  
> You can disable the formatting message in ascii mode in `make menuconfig`. But you can use properly only changing your "windows shell", as `windows cygwin`. This can understanding properly ascii colors chars, and show you like a charm.

Even after disabling ASCII formatting in `make menuconfig` I still encounter bad data formatting in the monitor leaving me unable to decipher output. 

74880 is the default ESP8266 bootloader baud rate: can anyone help me find a workaround to monitor this baud rate or change the default?

closed 2023-04-08 by XiaoSenLuo

Mathe13, 2023-04-11:
i got the same problem as @Masterjeef117 changed the baudrate and now i got garbage on the monitor and after some seconds it give me a error

Masterjeef117, 2023-04-11:
@Mathe13 seems that the devs aren't keen on fixing this issue, to get around this I dual-booted Ubuntu on my windows machine and developed exclusively in Linux. I've had better luck with the ESP32 lineup on windows. Best of luck