espressif/ESP8266_RTOS_SDK #900: PWM causes the software to watch the door trigger

SmartArduino, 2020-05-26:
I have port
https://github.com/espressif/ESP8266_RTOS_SDK/blob/126c5cf1f5ac285dfbfe3a140b1257e7938612b2/components/esp8266/driver/pwm.c
to esp-open-rtos.
It work well. However,
Every once in a while
PWM causes the software to watch the door trigger.
Is there Function feed soft watch dog?

nsfilho, 2020-05-26:
I can be wrong, but I believe of who feeds watchdog task is the priority (FreeRTOS). If watchdog task stalled, will cause a don't feed for watchdog, resulting in restart.

Most than the way you implement your driver, it could be how your final application is wrote (about the priority perspective). In example, if you use priority MAX or Critical, can cause that if it is running for a long time (around some milleseconds).

Again, I'm not an expert, but I'm sharing my discoveries with you.

dsptech, 2020-05-26:
Hi, 
the watchdog clear is performed by the the esp_internal_idle_hook() function (called from the idle task) .
(see: https://github.com/espressif/ESP8266_RTOS_SDK/blob/master/components/freertos/port/esp8266/port.c)

The pwm driver is software but fully NMI driven instead, so, it should not cause any problems (at output frequencies that it can manage). 

A possible problem could be that an higher priority task does not go in wait, so the idle task is not longer able to clear the watchdog.
If you could share your code, we may be able to help you.
Regards.

SmartArduino, 2020-06-02:
This issue has been resolved because flash SPI speed

closed 2020-06-03 by xiongyumail