espressif/ESP8266_RTOS_SDK #906: WPA2 Enterprise example

Vbansal1996, 2020-06-05:
## Environment

- Development Kit: NodeMCU 12E
- IDF version  - release/v3.3
- Development Env:      Make
- Operating System:     Ubuntu
- Power Supply:         USB

## Problem Description

I am trying to connect to WPA2 enterprise network but getting failed and device is getting reset.

Issue coming with passphrase error.
This issue was not there with previous commits, but latest release/v3.3 I am facing this issue.

### Expected Behavior

### Actual Behavior

### Steps to repropduce

1. step1
2. ...

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
}

## Debug Logs
```
D (929) system_event: SYSTEM_EVENT_STA_START
D (931) event: running task for loop 0x40108410
D (934) event: running post WIFI_EVENT:2 with handler 0x4028dfd0 on loop 0x40108410
0x4028dfd0: handle_sta_start at /home/keshav/esp/ESP8266_RTOS_SDK/components/tcpip_adapter/event_handlers.c:70

D (939) event: running post SYSTEM_EVENT:2 with handler 0x40267fa0 on loop 0x40108410
0x40267fa0: esp_event_post_to_user at /home/keshav/esp/ESP8266_RTOS_SDK/components/esp_event/event_loop_legacy.c:44

assertion "phlen >= 6 && phlen < 64" failed: file "wpa.c", line 2100, function: wpa_set_passphrase
abort() was called at PC 0x40235f2c on core 0
0x40235f2c: __assert_func at /home/dongheng/workspace/espressif/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-lx106-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4023bb3e  PS      : 0x00000030  A0      : 0x4023bb3c  A1      : 0x3fff4230  
0x4023bb3e: abort at /home/keshav/esp/ESP8266_RTOS_SDK/components/newlib/newlib/port/syscall.c:168 (discriminator 1)

0x4023bb3c: abort at /home/keshav/esp/ESP8266_RTOS_SDK/components/newlib/newlib/port/syscall.c:168 (discriminator 1)

A2      : 0x00000000  A3      : 0x00000001  A4      : 0xffffffdb  A5      : 0x00000001  
A6      : 0x00000000  A7      : 0x402a8db5  A8      : 0x3fff4179  A9      : 0x000000a0  
A10     : 0x0000000f  A11     : 0x00000002  A12     : 0x402ab518  A13     : 0x402ab4fc  
A14     : 0x402ab4f4  A15     : 0x00000834  SAR     : 0x0000001e  EXCCAUSE: 0x0000001d  

Backtrace: 0x4023bb3e:0x3fff4230 0x40235f2f:0x3fff4240 0x40257171:0x3fff4270 0x40257310:0x3fff4280 0x40247fb1:0x3fff4330 0x40247ffc:0x3fff4370 0x4021ee70:0x3fff4380 0x4021ea79:0x3fff43d0 0x4021eaec:0x3fff43e0 0x4021f0c4:0x3fff43f0 0x4021f2fb:0x3fff4420 0x4021bdd3:0x3fff4490 0x4021c1f1:0x3fff44f0 0x4021e899:0x3fff4510 0x4021e8b4:0x3fff4520 0x40213609:0x3fff4530 0x4029ee32:0x3fff4540 
0x4023bb3e: abort at /home/keshav/esp/ESP8266_RTOS_SDK/components/newlib/newlib/port/syscall.c:168 (discriminator 1)

0x40235f2f: __assert_func at ??:?

0x40257171: wpa_set_passphrase at /home/keshav/esp/ESP8266_RTOS_SDK/components/wpa_supplicant/src/rsn_supp/wpa.c:2103

0x40257310: wpa_set_bss at /home/keshav/esp/ESP8266_RTOS_SDK/components/wpa_supplicant/src/rsn_supp/wpa.c:2079

0x40247fb1: wpa_config_bss at /home/keshav/esp/ESP8266_RTOS_SDK/components/wpa_supplicant/port/esp_supplicant/esp_wpa_main.c:129

0x40247ffc: wpa_sta_connect at /home/keshav/esp/ESP8266_RTOS_SDK/components/wpa_supplicant/port/esp_supplicant/esp_wpa_main.c:151

0x4021ee70: dhcp_bind_check at ??:?

0x4021ea79: chm_init at ??:?

0x4021eaec: chm_start_op at ??:?

0x4021f0c4: dhcp_bind_check at ??:?

0x4021f2fb: cnx_start_handoff_cb at ??:?

0x4021bdd3: scan_build_chan_list at ??:?

0x4021c1f1: scan_start at ??:?

0x4021e899: chm_end_op at ??:?

0x4021e8b4: chm_end_op at ??:?

0x40213609: ets_timer_proc at ??:?

0x4029ee32: ppTask at ??:?

```

nsfilho, 2020-07-18:
Hi @Vbansal1996 , I don't know if you solved the problem, but appears you are using a small phrase to WPA2. Please try change the phrase, or show the code as throwing this error for us and we will try to help you :)

Best regards, NS.

Vbansal1996, 2020-09-21:
Hi 

After cloning the latest changes still I am not able to connect with the WPA network.
Device is throwing error 
`
11:46:02.368 -> [0mD (173468) system_event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:tcwifi, ssid_len:6, bssid:98:da:c4:69:2b:ab, reason:23 (802.1x auth failed)
11:46:02.368 -> D (173472) event: running post WIFI_EVENT:5 with handler 0x4026c1bc on loop 0x40106da0
11:46:02.368 -> D (173480) tcpip_adapter: if0 start ip lost tmr: enter
11:46:02.368 -> D (173485) tcpip_adapter: if0 start ip lost tmr: no need start because netif=0x401071c8 interval=120 ip=0
11:46:02.403 -> D (173494) event: running post WIFI_EVENT:5 with handler 0x40224340 on loop 0x40106da0
11:46:02.947 -> [0;32mI (174049) example: ~~~~~~~~~~~[0m
11:46:02.947 -> [0;32mI (174050) example: IP:0.0.0.0[0m
11:46:02.947 -> [0;32mI (174051) example: MASK:0.0.0.0[0m
11:46:02.947 -> [0;32mI (174052) example: GW:0.0.0.0[0m
11:46:02.947 -> [0;32mI (174054) example: ~~~~~~~~~~~[0m
11:46:02.947 -> [0;32mI (174058) example: free heap:71652
11:46:02.947 -> [0m
11:46:04.442 -> D (175559) system_event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:tcwifi, ssid_len:6, bssid:00:00:00:00:00:00, reason:205 (other reason)
11:46:04.477 -> D (175563) event: running post WIFI_EVENT:5 with handler 0x4026c1bc on loop 0x40106da0
11:46:04.477 -> D (175570) tcpip_adapter: if0 start ip lost tmr: enter
11:46:04.477 -> D (175575) tcpip_adapter: if0 start ip lost tmr: no need start because netif=0x401071c8 interval=120 ip=0
11:46:04.477 -> D (175584) event: running post WIFI_EVENT:5 with handler 0x40224340 on loop 0x40106da0
11:46:04.957 -> [0;32mI (176064) example: ~~~~~~~~~~~[0m
11:46:04.957 -> [0;32mI (176065) example: IP:0.0.0.0[0m
11:46:04.957 -> [0;32mI (176066) example: MASK:0.0.0.0[0m
11:46:04.957 -> [0;32mI (176067) example: GW:0.0.0.0[0m
11:46:04.957 -> [0;32mI (176069) example: ~~~~~~~~~~~[0m
11:46:04.957 -> [0;32mI (176073) example: free heap:71672
11:46:04.957 -> [0m
11:46:05.092 -> [0;32mI (176199) wifi:state: 0 -> 2 (b0)
11:46:05.092 -> [0m[0;32mI (176203) wifi:state: 2 -> 3 (0)
11:46:05.092 -> [0m[0;32mI (176212) wifi:state: 3 -> 5 (10)
11:46:05.092 -> [0m[0;32mI (176297) wifi:state: 5 -> 0 (17a0)
11:46:05.194 -> [0mD (176297) system_event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:tcwifi, ssid_len:6, bssid:98:da:c4:69:2b:ab, reason:23 (802.1x auth failed)
11:46:05.194 -> D (176302) event: running post WIFI_EVENT:5 with handler 0x4026c1bc on loop 0x40106da0
11:46:05.194 -> D (176309) tcpip_adapter: if0 start ip lost tmr: enter
11:46:05.194 -> D (176314) tcpip_adapter: if0 start ip lost tmr: no need start because netif=0x401071c8 interval=120 ip=0
11:46:05.228 -> D (176324) event: running post WIFI_EVENT:5 with handler 0x40224340 on loop 0x40106da0
11:46:06.967 -> [0;32mI (178075) example: ~~~~~~~~~~~[0m
11:46:06.967 -> [0;32mI (178076) example: IP:0.0.0.0[0m
11:46:06.967 -> [0;32mI (178077) example: MASK:0.0.0.0[0m
11:46:06.967 -> [0;32mI (178078) example: GW:0.0.0.0[0m
11:46:06.967 -> [0;32mI (178080) example: ~~~~~~~~~~~[0m
11:46:06.967 -> [0;32mI (178084) example: free heap:71652
`
But using same credentials I am able to connect with network using PC and Mobile Phones.

Can you suggest what are the possible reasons.
I am using wpa2_enterprise examples for testing.

Additionally with latest changes in release/v3.3 - using same example I am getting watchdog reset when device is trying to get the IP.

Vbansal1996, 2020-09-21:
Adding additional logs

1:59:24.501 -> D (553) nvs: nvs_get sta.phybw 1
11:59:24.501 -> D (556) nvs: nvs_get_str_or_blob sta.apsw
11:59:24.501 -> D (560) nvs: nvs_get_str_or_blob sta.apinfo
11:59:24.501 -> D (564) nvs: nvs_get sta.scan_method 1
11:59:24.501 -> D (567) nvs: nvs_get sta.sort_method 1
11:59:24.501 -> D (571) nvs: nvs_get sta.minrssi 1
11:59:24.501 -> D (574) nvs: nvs_get sta.minauth 1
11:59:24.501 -> D (577) nvs: nvs_get_str_or_blob ap.ssid
11:59:24.526 -> D (581) nvs: nvs_get_str_or_blob ap.mac
11:59:24.526 -> D (584) nvs: nvs_get_str_or_blob ap.passwd
11:59:24.526 -> D (588) nvs: nvs_get_str_or_blob ap.pmk
11:59:24.526 -> D (592) nvs: nvs_get ap.chan 1
11:59:24.526 -> D (594) nvs: nvs_get ap.authmode 1
11:59:24.526 -> D (597) nvs: nvs_get ap.hidden 1
11:59:24.526 -> D (600) nvs: nvs_get ap.max.conn 1
11:59:24.526 -> D (603) nvs: nvs_get bcn.interval 2
11:59:24.526 -> D (607) nvs: nvs_get ap.phym 1
11:59:24.561 -> D (609) nvs: nvs_get ap.phybw 1
11:59:24.561 -> D (612) nvs: nvs_get ap.sndchan 1
11:59:24.561 -> D (615) nvs: nvs_get lorate 1
11:59:24.561 -> D (618) nvs: nvs_get sta.pmf_e 1
11:59:24.561 -> D (621) nvs: nvs_get sta.pmf_r 1
11:59:24.561 -> D (624) nvs: nvs_get ap.pmf_e 1
11:59:24.561 -> D (627) nvs: nvs_get ap.pmf_r 1
11:59:24.561 -> D (630) nvs: nvs_get_str_or_blob country
11:59:24.561 -> D (633) nvs: nvs_get sta.rm_e 1
11:59:24.561 -> D (636) nvs: nvs_get sta.btm_e 1
11:59:24.561 -> [0;32mI (640) example: Setting WiFi configuration SSID tcwifi...[0m
11:59:24.596 -> [0;32mI (646) wpa: WPA2 ENTERPRISE VERSION: [v2.0] enable
11:59:24.596 -> [0m
11:59:24.596 -> D (651) wpa: WPA2 ENTERPRISE CRYPTO INIT.

11:59:24.596 -> 
11:59:24.596 -> 15, 15, 11 

11:59:24.596 ->  D (656) system_event: SYSTEM_EVENT_STA_START
11:59:24.596 -> D (660) event: running post WIFI_EVENT:2 with handler 0x4026fa94 on loop 0x40106ddc
11:59:24.596 -> D (667) event: running post WIFI_EVENT:2 with handler 0x40225084 on loop 0x40106ddc
11:59:25.211 -> D (1284) wpa: RSN: PMKSA cache search - network_ctx=0 try_opportunistic=0
11:59:25.211 -> D (1287) wpa: RSN: Search for BSSID 98:da:c4:69:2b:ab
11:59:25.211 -> D (1289) wpa: RSN: No PMKSA cache entry found
11:59:26.158 -> [0;32mI (2247) wifi:state: 0 -> 2 (b0)
11:59:26.158 -> [0m[0;32mI (2253) wifi:state: 2 -> 3 (0)
11:59:26.192 -> [0m[0;32mI (2259) wifi:state: 3 -> 5 (10)
11:59:26.192 -> [0m[0;32mI (2260) wpa: wpa2_task prio:2, stack:6656
11:59:26.192 -> [0m
11:59:26.192 -> D (2263) wpa: WPA2: wifi->wpa2 api completed sig(0)
11:59:26.192 -> D (2264) wpa: WPA2: wpa2 api return, sm->state(1)
11:59:26.192 -> D (2267) wpa: IEEE 802.1X RX: version=2 type=0 length=5
11:59:26.192 -> 
11:59:26.192 -> V (2271) wpa: WPA2: RX EAPOL-EAP PACKET - hexdump(len=9):
11:59:26.227 -> V (2276) wpa: 02 00 00 05 01 dd 00 05 01 
11:59:26.227 -> D (2280) wpa: WPA2: wifi->wpa2 api completed sig(1)
11:59:26.227 -> D (2284) wpa: WPA2: wpa2 api return, sm->state(1)
11:59:26.227 -> D (2299) wpa: IEEE 802.1X RX: version=2 type=0 length=5
11:59:26.227 -> 
11:59:26.227 -> V (2301) wpa: WPA2: RX EAPOL-EAP PACKET - hexdump(len=9):
11:59:26.227 -> V (2304) wpa: 02 00 00 05 01 49 00 05 01 
11:59:26.227 -> D (2306) wpa: WPA2: wifi->wpa2 api completed sig(1)
11:59:26.261 -> D (2309) wpa: WPA2: wpa2 api return, sm->state(1)
11:59:26.261 -> D (2327) wpa: IEEE 802.1X RX: version=2 type=0 length=6
11:59:26.261 -> 
11:59:26.261 -> V (2329) wpa: WPA2: RX EAPOL-EAP PACKET - hexdump(len=10):
11:59:26.261 -> V (2331) wpa: 02 00 00 06 01 4a 00 06 19 20 
11:59:26.261 -> D (2333) wpa: TLS: Phase2 EAP types - hexdump(len=8):
11:59:26.261 -> D (2337) wpa: 00 00 00 00 1a 00 00 00 
11:59:26.261 -> D (2340) wpa: TLS: using phase1 config options
11:59:26.295 -> D (2349) wpa: SSL: Received packet(len=6) - Flags 0x20
[... 157 lines cut ...]
11:59:27.142 -> V (3197) wpa: WPA2: RX EAPOL-EAP PACKET - hexdump(len=8):
11:59:27.142 -> V (3199) wpa: 02 00 00 04 04 4c 00 04 
11:59:27.142 -> [0;32mI (3202) wpa: >>>>>wpa2 FAILED
11:59:27.142 -> [0m
11:59:27.142 -> D (3206) wpa: WPA2: wifi->wpa2 api completed sig(1)
11:59:27.142 -> D (3210) wpa: WPA2: wpa2 api return, sm->state(3)
11:59:27.142 -> D (3216) wpa: WPA2: queue deleted
11:59:27.142 -> D (3218) wpa: WPA2: task deleted
11:59:27.142 -> D (3221) wpa: WPA2: wifi->wpa2 api completed sig(2)
11:59:27.176 -> D (3225) wpa: WPA2: wpa2 api return, sm->state(3)
11:59:27.176 -> D (3230) wpa: wpa2 eap_peer_sm_deinit: free data lock
11:59:27.176 -> [0;32mI (3235) wifi:state: 5 -> 0 (17a0)
11:59:27.176 -> [0mD (3239) system_event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:tcwifi, ssid_len:6, bssid:98:da:c4:69:2b:ab, reason:23 (802.1x auth failed)
11:59:27.176 -> D (3251) event: running post WIFI_EVENT:5 with handler 0x402727c8 on loop 0x40106ddc
11:59:27.211 -> D (3258) event: running post WIFI_EVENT:5 with handler 0x4026fadc on loop 0x40106ddc
11:59:27.211 -> D (3265) tcpip_adapter: if0 start ip lost tmr: enter
11:59:27.211 -> D (3270) tcpip_adapter: if0 start ip lost tmr: no need start because netif=0x40107254 interval=120 ip=0
11:59:27.211 -> D (3279) event: running post WIFI_EVENT:5 with handler 0x40225084 on loop 0x40106ddc
11:59:28.588 -> [0;32mI (4673) example: ~~~~~~~~~~~[0m
11:59:28.588 -> [0;32mI (4674) example: IP:0.0.0.0[0m
11:59:28.622 -> [0;32mI (4676) example: MASK:0.0.0.0[0m
11:59:28.622 -> [0;32mI (4677) example: GW:0.0.0.0[0m
11:59:28.622 -> [0;32mI (4679) example: ~~~~~~~~~~~[0m
11:59:28.622 -> [0;32mI (4681) example: free heap:64072
11:59:28.622 -> [0m

Vbansal1996, 2020-09-21:
FYI, it working fine till this commit: 655f9340