espressif/ESP8266_RTOS_SDK #920: Failed to verify partition table E (90) boot: load partition table error!

arjo129, 2020-06-17:
INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](http://esp-idf.readthedocs.io/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit: NodeMCU
- IDF version: 8189b9a4
- Development Env:      Make
- Oprating System:    Ubuntu
- Power Supply:         USB

## Problem Description

Unable to run programs via make flash monitor. Invalid magic number.

### Expected Behavior

Able to run program

### Actual Behavior

### Steps to repropduce

1. `make flash monitor`
````
load 0x40100000, len 7168, room 16 
tail 0
chksum 0xe0
load 0x3ffe8408, len 24, room 8 
tail 0
chksum 0x09
load 0x3ffe8420, len 3556, room 8 
tail 12
chksum 0x1e
I (45) boot: ESP-IDF v3.3-rc1-104-g8189b9a4-dirty 2nd stage bootloader
I (45) boot: compile time 10:43:59
I (46) qio_mode: Enabling default flash chip QIO
I (54) boot: SPI Speed      : 80MHz
I (61) boot: SPI Mode       : QIO
I (67) boot: SPI Flash Size : 4MB
E (73) flash_parts: partition 0 invalid magic number 0x3333
E (82) boot: Failed to verify partition table
E (90) boot: load partition table error!

````
// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

fdroidsaidwtf, 2020-06-18:
info@idf.gov

nsfilho, 2020-07-18:
Hi! I belive this is the exactly same case as #929 . If you could, please do the same things as described in this refer issue and we do your best to help you.

OK? Best regards, NS.