espressif/ESP8266_RTOS_SDK #925: Not able to download complete OTA file using https

swetha-paladugu-oyo, 2020-06-29:
## Environment

- Development Kit:
- Development Env:      Eclipse
- Operating System:     Windows
- Power Supply:            external 3.3V

I am using release/v3.2 version ESP8266_RTOS_SDK

## Problem Description

I am opening a tcp socket in the code for getting wifi credentials and also mqtt for data transmission, but because i understood ota over https is not working since there is insufficient memory i have removed all my other code and tried with the basic api calls used in simple_ota_example.c file. But still the OTA file doesn't download completely. My OTA file size is 450k, but only 250k file gets downloaded, sometimes even less only 10k, 20k gets downloaded.  

### Code to reproduce this issue

void main()
{
	initialise_wifi();

	xEventGroupWaitBits(wifi_event_group, CONNECTED_BIT,
	                        false, true, portMAX_DELAY);

	ota_task();

	while(1);
}

void ota_task()
{
    ESP_LOGI(TAG, "Testing https connection...");

    ESP_LOGI(TAG, "URL:%s",TEST_FIRMWARE_UPGRADE_URL);

    esp_http_client_config_t config = {
        .url = TEST_FIRMWARE_UPGRADE_URL,
        .cert_pem = (char *)server_cert_pem_start,
        .event_handler = _http_event_handler,
    };
    esp_err_t ret = esp_https_ota(&config);
    if (ret == ESP_OK) {
        esp_restart();
    } else {
        ESP_LOGE(TAG, "Firmware Upgrades Failed");
        task_fatal_error(OTA_MGR_ERR);
    }
}

## Debug Logs

##only 13K got downloaded

E (71070) ota: HTTP_EVENT_ON_DATA, len=436
E (71200) esp_https_ota: Written image length 13140
E (71340) HTTP_CLIENT: is_data_remain=1, is_chunked=0
E (71490) HTTP_CLIENT: need_read=1460, byte_to_read=512, rlen=351, ridx=0
E (71690) HTTP_CLIENT: http_on_body 351
E (71810) ota: HTTP_EVENT_ON_DATA, len=351
E (71940) HTTP_CLIENT: is_data_remain=1, is_chunked=0
E (77090) esp-tls: read error :-76:
E (77090) TRANS_SSL: esp_tls_conn_read error, errno=No more processes
E (77090) HTTP_CLIENT: need_read=1109, byte_to_read=512, rlen=-1, ridx=351
E (77300) esp_https_ota: Written image length 13491
E (77440) HTTP_CLIENT: is_data_remain=1, is_chunked=0
E (82590) HTTP_CLIENT: need_read=1460, byte_to_read=512, rlen=0, ridx=0
I (82590) esp_https_ota: Connection closed,all data received
E (82660) ota: HTTP_EVENT_DISCONNECTED
E (82800) ota: HTTP_EVENT_DISCONNECTED
E (82890) esp_https_ota: Total binary data length writen: 13491
I (83060) esp_image: segment 0: paddr=0x00110010 vaddr=0x40210010 size=0x6f3ec (455660) map
E (83410) esp_image: unaligned segment length 0xffffffff
E (83460) esp_https_ota: Error: esp_ota_end failed! err=0x5379. Image is invalid
E (83670) ota: Firmware Upgrades Failed

complete log is attached in ota_files.zip "ota_log_without_timers_1460.txt"

## only 283k got downloaded

E (441082) ota: HTTP_EVENT_ON_DATA, len=512
E (441212) HTTP_CLIENT: is_data_remain=1, is_chunked=0
E (441362) HTTP_CLIENT: need_read=948, byte_to_read=512, rlen=143, ridx=512
E (441562) HTTP_CLIENT: http_on_body 143
E (441682) ota: HTTP_EVENT_ON_DATA, len=143
E (441812) HTTP_CLIENT: is_data_remain=1, is_chunked=0
E (446972) esp-tls: read error :-76:
E (446972) TRANS_SSL: esp_tls_conn_read error, errno=No more processes
E (446982) HTTP_CLIENT: need_read=805, byte_to_read=512, rlen=-1, ridx=655
E (447182) esp_https_ota: Written image length 283895
E (447332) HTTP_CLIENT: is_data_remain=1, is_chunked=0
E (452482) HTTP_CLIENT: need_read=1460, byte_to_read=512, rlen=0, ridx=0
I (452482) esp_https_ota: Connection closed,all data received
E (452552) ota: HTTP_EVENT_DISCONNECTED
E (452702) ota: HTTP_EVENT_DISCONNECTED
E (452792) esp_https_ota: Total binary data length writen: 283895
I (452972) esp_image: segment 0: paddr=0x00110010 vaddr=0x40210010 size=0x6f3ec (455660) map
E (453322) esp_image: unaligned segment length 0xffffffff
E (453372) esp_https_ota: Error: esp_ota_end failed! err=0x5379. Image is invalid
E (453582) ota: Firmware Upgrades Failed

Attaching sdk_config file for reference.
[ota_files.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/4844629/ota_files.zip)

Please reply fast..

nsfilho, 2020-07-17:
I faced the same problem many and many times. You can see my `sdkconfig.defaults` -- I really belive will help you to solve the problem at https://github.com/nsfilho/E12AIO3,  because you need to change some mbed parameters.

Best regards,

swetha-paladugu-oyo, 2020-07-17:
@nsfilho 
Thnaks for your response, that issue is solved... however i didnt enable these 2 macros CONFIG_MBEDTLS_DEFAULT_MEM_ALLOC=y
CONFIG_MBEDTLS_ASYMMETRIC_CONTENT_LEN=y. 
Now i have another problem... there is memory issue... My mqtt connection fails when CONFIG_MBEDTLS_SSL_OUT_CONTENT_LEN is made 16384... Application is short of memory... What can be done? Any suggestion?

swetha-paladugu-oyo, 2020-07-17:
@nsfilho 
Even if mqtt connects... upon mqtt disconnect, it wont connect again.. memory issue..

nsfilho, 2020-07-18:
Hi my friend, how are you? First of all, I'm happy of the solution works and again, I faced the exactly same problem as  you! heheheheh -- Let me ask you something, what do you think do help me to develop a opensource framework? Because you are facing the same things, and I solved many of them until now.

What solution I have done to this? Let me explain (isn't  short):

1. I create a  flag in app_main to see if it's a  normal boot, or a boot for update (I used spiffs, but you can use ROM too).
2. If it's a  firmware boot, not start the things and only do a update -- write the result in spiffs (in my case, but you can use ROM too).
3. If it's a  normal boot, I see if have a updated flag marked as done to show the results and continue boot (loading all other services).

When the user ask to update firmware, I do:

1. Mark in spiffs to update
2. Save the url for the firmware (in spiffs too and this could be a loooonnggg url -- because of that I used a chunk way in httpd to solve that).
3. restart the ESP.

In this way, you have enough memory.

You can see all theses steps in my firmware. And again, what should you thing to help on that? https://github.com/nsfilho/e12aio3

swetha-paladugu-oyo, 2020-07-18:
Hi, @nsfilho, 
i was also doing exactly the same, but since we are increasing CONFIG_MBEDTLS_SSL_OUT_CONTENT_LEN to 16384, the tls buffers which are allocated with large size and when i do mqtt connection, it fails some times saying allocation memory failed.
I re-visiting the features in code to get some memory but still i couldn't.. 
Is there any possibility that this CONFIG_MBEDTLS_SSL_OUT_CONTENT_LEN can be enabled only for ota in application during run time?

nsfilho, 2020-07-18:
I understanding that, but you only use MBEDTLS when are you doing the OTA, correctly? If you see my code (and my sdkconfig.defaults) you can see that. Ahh, an important point, my normal code, load many services (mqtt, httpd, and ...).

When you are in ota mode (if we can call as that) you don't load any other modules, ... correctly? What are you doing to receive the user asked `please do an ota update`?

> Important thing: mqtt stop or disconnect always crash our code. MQTT stop or disconnect always freeze, don't you that hehehe.

If  you could share a  little of your code to  help you in what context you are using, I will do my best to try to help you.

swetha-paladugu-oyo, 2020-07-19:
Hi,
OTA flow on boot up, i'm also checking with a flag if it is needed and doing only ota with out mqtt or any other...
But the normal flow(with out) ota, also uses configurations which are defined in sdkconfig.h, where we defined CONFIG_MBEDTLS_SSL_OUT_CONTENT_LEN to 16384.. Which is effecting mqtt connection... the sdkconfig.h will be same for ota and without ota correct? If we can undo that configuration in non ota flow during run time, this would solve our problem..

nsfilho, 2020-07-19:
hmmm... you are usign mqtt over ssl, correctly? that's the problem. If you do that, you will have memory problems. I don't know how to solve that (if you really need mqtt over ssl).

I always use clear mqtt protocol, because it's a short footprint and is a little quickly. If you can, try not use mqtt over ssl....

swetha-paladugu-oyo, 2020-07-20:
@nsfilho 
yes we need mqtt over ssl...

nsfilho, 2020-07-20:
@swetha-paladugu-oyo after think a lot about your use case, I see 3 possible solutions:

a) Use 2 images -- in OTA0, normal image -- and when you need OTA, switch to OTA1 partition start, update OTA0 and change partition to OTA0 again. In this case you can use 2 totally different images (including with some flags as MBEDTLS different).

b) Switch to ESP32, because in this case you really need a more powerful platform (a lot more memory). In this case it is super-easy (because you are doing your firmware in ESP8266_RTOS_SDK version 3.0+).

c) Rethink the design of your application, because ESP, after boot, without MBED use, have around 50k of memory free. If you use MBED, you have around 34k (it is enough for many use cases). What I do, in most of my use cases is start small stacks (and close), processing the things for a moment and "die". In this way, you preserve your heap memory.

In my case, I have no option and used the -C- solution, because all my devices are based on ESP8266. In your case, if you can, change to ESP32 because in a long term, it is a best platform to support your customers.

swetha-paladugu-oyo, 2020-07-21:
@nsfilho thanks for your quick response

In First case how do we do OTA is the question.
Second case is not possible, we cannot move to esp32.
Third case Revisiting logic, yes i'm doing.

nsfilho, 2020-07-21:
When `user select` for OTA, you change the partition to boot as OTA1 and reboot.
In `OTA Mode` (on OTA1), when you receive your firmware, you overwrite OTA0 partition. You always overwrite only OTA0.
After you download the entire firmware, checks if its OK (by checksum), you change partition boot to OTA0 again.
Restart ESP and ready!

I believe your next question will be about where you can find the sample to do the this think. We have a nice sample (using all command of you will need), directly in https://github.com/espressif/ESP8266_RTOS_SDK/blob/2fd6ea138c069963425a88144683cb1d58709e68/components/esp_https_ota/src/esp_https_ota.c#L32

Anyways, if you have any doubt and if I can help, please let me know 😄 

> You select where you will download the firmware by update_partition in line 79: `esp_ota_begin(update_partition`