espressif/ESP8266_RTOS_SDK #927: simple_wifi.c example of ESP8266_RTOS_SDK Release v3.3 crashed

someyan, 2020-06-30:
## Environment

- Development Kit:
- IDF version ESP8266_RTOS_SDK Release v3.3
- Development Env:      [Make]
- Operating System:     [Windows]
- Power Supply:         [USB]

## Problem Description
create softap example in simple_wifi.c will crash after running 30 to120 seconds.

### Expected Behavior
The softap works fine without any core panic or reset

### Actual Behavior
System reboot, reset_reason: RTC reset 4 wakeup 0 store 0, reason is 7
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
### Steps to repropduce

1. 
#define EXAMPLE_ESP_WIFI_MODE_AP   1 //TRUE:AP FALSE:STA
#define EXAMPLE_ESP_WIFI_SSID      "CMCC555"
#define EXAMPLE_ESP_WIFI_PASS      ""
#define EXAMPLE_MAX_STA_CONN       4
2. make 
   make flash 

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

simple_wifi.c example in ESP8266_RTOS_SDK Release v3.3

## Debug Logs

[20:55:48.916]收←◆?[0;32mI (330) system_api: Base MAC address is not set, read default base MAC address from EFUSE[0m
[0;32mI (340) system_api: Base MAC address is not set, read default base MAC address from EFUSE[0m

[20:55:49.021]收←◆phy_version: 1159.0, 85b471e, Apr 21 2020, 17:03:08, RTOS new
[0;32mI (405) phy_init: phy ver: 1159_0[0m
[0;32mI (410) reset_reason: RTC reset 4 wakeup 0 store 0, reason is 7[0m
[0;32mI (417) simple wifi: ESP_WIFI_MODE_AP[0m

[20:55:49.067]收←◆[0;32mI (460) simple wifi: wifi_init_softap finished.SSID:CMCC555 password:[0m
SYSTEM_EVENT_AP_START

[20:56:22.311]收←◆Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Core 0 register dump:
PC      : 0x4022541b  PS      : 0x00000033  A0      : 0x40215400  A1      : 0x3fff1e80  
A2      : 0x00000000  A3      : 0x00022ecb  A4      : 0x4024aade  A5      : 0x000689d0  
A6      : 0x0000066c  A7      : 0xffffffff  A8      : 0x402153c4  A9      : 0x000000c0  
A10     : 0x00000091  A11     : 0x00000002  A12     : 0x3ffe8a2c  A13     : 0x3ffe8a1c  
A14     : 0x00000000  A15     : 0x3ffeac28  SAR     : 0x00000019  EXCCAUSE: 0x00000000

nsfilho, 2020-07-18:
I will flow that problem, because I'm facing a lot of problems in this **release** as in `esp_wifi_set_config` with `ESP_IF_WIFI_AP`.

nsfilho, 2020-07-19:
@someyan I solved my problem disabling LOOPBACK interface. After this, all around ESP_IF_WIFI_AP, as APSTA mode, it is now fully functional.

nsfilho, 2020-07-23:
Reading again your post, I see something important: when password is empty (strlen == 0), you need to use `auth open`. I don't remember if simple_wifi.c have a test for this....

Anyways @someyan are you solved your problem? If yes, please share with us how and the steps 😄 . Probably this could help someone in future.

someyan, 2020-07-25:
@nsfilho 
When password is empty,the authmode is WIFI_AUTH_OPEN.
The crash will appear on the products of manufacturer number 20（flash_id Manufacturer: 20）
When change to Development board of flash_id Manufacturer: c8，it works fine.
I suspect the problem is caused by the device itself.
I will do more tests.