espressif/ESP8266_RTOS_SDK #929: Failed to verify partition table

GarMingLi, 2020-07-04:
## Environment
- IDF version : a215fe3b43d10cef2475437861d30dfbc3a74e0f
- Development Env:      Make
- Operating System:     win10 wsl（Ubuntu18.04）
- toolchain: xtensa-lx106-elf-linux32-1.22.0-100-ge567ec7-5.2.0.tar
- Board: NodeMCU

## Problem Description
After I execute the command 'make Flash 'to burn the firmware, I execute the' Make Monitor 'to run the program.The following error occurs:
```
I (66) boot: compile time 12:28:27
I (75) qio_mode: Enabling default flash chip QIO
I (75) boot: SPI Speed      : 40MHz
I (76) boot: SPI Mode       : QIO
I (80) boot: SPI Flash Size : 4MB
E (84) flash_parts: partition 0 invalid magic number 0x3333
E (90) boot: Failed to verify partition table
E (95) boot: load partition table error!
user code done
```

nsfilho, 2020-07-18:
What partition table are you using? You can see in `make menuconfig` or if it's a custom, please send us the partition.csv and we try to help you.

best regards, ns.

nsfilho, 2020-07-23:
@GarMingLi are you solved your problem? If yes, please share with us how and the steps 😄 . Probably this could help someone in future.

GarMingLi, 2020-08-17:
@nsfilho When I changed the QIO of SPI Mode to DIO, the problem was solved.

MY201314MY, 2021-02-01:
> @nsfilho When I changed the QIO of SPI Mode to DIO, the problem was solved.

Well Done!

alexpalade, 2021-04-08:
Thanks @GarMingLi, your solution fixed this for my NodeMCU (set SPI mode to DIO).

captainzkx, 2021-04-16:
Thanks,the problem was solved for my NodeMCU.

HHyeon, 2021-04-22:
> @nsfilho When I changed the QIO of SPI Mode to DIO, the problem was solved.

I was middle in figure it out those error that you mentioned. 
in nodeMCU 0.9, to using rtos sdk , that need to change SPI Mode QIO to DIO Apparently.
thank you. I solved my that error thanks to you.

pzarzycki, 2023-08-27:
I had similar issue on `8266` platform after disabling option `Enable ‘nano’ formatting options for printf/scanf family` option configuration for `components` / `newlib`.
Clean build did not help.

Error message:
```
flash_parts: partition 0 invalid magic number 0x6f5c
```

Pz1c, 2023-09-12:
Hi guys
I got same error "flash_parts: partition 0 invalid magic number 0x867a"
with my esp32-wroom-32 chip when try to configure Secure Boot v1
only one change for default settings was enlarge bootloader partition size
buil/partition-table-flash_args

--flash_mode dio --flash_freq 40m --flash_size 4MB
0xd000 partition_table/partition-table.bin
build/flash_args.in

--flash_mode dio --flash_freq 40m --flash_size 4MB
0x20000 $<TARGET_PROPERTY:__idf_build_target,EXECUTABLE_NAME>.bin
0xd000 partition_table/partition-table.bin
appreciate any help)